stefanbode/Sonoff-Tasmota #3: Pulsetimers

mrmin24, 2017-04-23:
Hi,

There are still only 4 pulsetimers available for PCF8574 setup. This is not ideal especially for irrigation.

Thanks

stefanbode, 2017-04-24:
Hi, this can be changed easily in the standard already in sonoff.ino. OK?

#define MAX_PULSETIMERS        4            // Max number of supported pulse timers

mrmin24, 2017-04-24:
Yes, I will test this too. I suppose it is worth having a" IFDEF USE_PCF8574" to adjust these to 8 or more.

stefanbode, 2017-04-24:
There is an additional hardcoded line in sonoff.ino: line 1373 somewhere here:

This is hardcoded &3 (Binary 11) and needs to be flexible. 

`    pulse_timer[(device -1)&(1<<(MAX_PULSETIMERS-1)-1)] = (power & mask) ? sysCfg.pulsetime[(device -1)&(1<<(MAX_PULSETIMERS-1)-1)] : 0;`

mrmin24, 2017-05-03:
I tested your latest baseline and the timers are working, but it seems that if you activate the nth timer and the n + 4th time (e.g.1 and 5 or 2 and 6) at the same time, the nth relay never switches off. If you activate any other timer with nth timer, it works fine

stefanbode, 2017-06-06:
Fixed with 5.1.4 update

closed 2017-06-06 by stefanbode