stefanbode/Sonoff-Tasmota #4: Power on state

mrmin24, 2017-04-23:
I know the MQTT configuration is not working, but this means that the power on state cannot be set. maybe worth defaulting to off?

stefanbode, 2017-04-24:
The MQQT IS working. I have it running in my configuration. The only issue is the STATE message that you get every 300 seconds. This can handle maximum 8 relays because of length.

/setting/power7 on

for example should work.

mrmin24, 2017-04-24:
No, you are correct the MQTT works perfectly. I was talking about the PowerOnState command, but this does not seem to work even on the original software. If I set PowerOnState to 0, the relays should switch off on restart?

stefanbode, 2017-04-24:
It seems to be that this part is a bit tricky. The poweronstate is ONLY running, if the reset reason is "Power On".
sonoff.ini line 2235ff
f (ESP.getResetReason() == "Power on") {

It looks like this special reason is not send by all systems. My Wemos Mini send on Power on the message "External System". Therefore this can never work. I assume, if you change the string it might work.

mrmin24, 2017-04-24:
Yes it seems you are correct. I changed the code to look for "External System" and now it works perfectly. Maybe worth highlighting to arendst

mrmin24, 2017-05-07:
fixed in arendst/tasmota

closed 2017-05-07 by mrmin24