stefanbode/Sonoff-Tasmota #5: PCF8574 with Sonoff SV relay switching

mrmin24, 2017-05-07:
When running the PCF8574 on the SV, the relay 1 is obviously the onboard relay. 

The 8 other relays work just fine as well. 

The problem is that when the relay one is switched off, the relay 5 and sometimes 8 is switched on by itself. 

Then also if relay 5 is switched off in turn, relay 8 also switches off.

mrmin24, 2017-05-12:
Hi,

Any chance you could look at this?

thanks

stefanbode, 2017-05-13:
Hmm, quite strange. You use the pulsetimers? I do the same on wemos with all 8 relays. I already found a bug that interfere with GPIO0. Please can you send me your complete setup to reproduce. I assume you use the latest version

mrmin24, 2017-05-13:
This is on the Sonoff SV. Yes using pulsetimers. There seems to be some sort of bitmask or shifting problem. This seems to be related to #3

[image]
[image]

[image]

mrmin24, 2017-05-25:
any chance you could look at this?

thanks

stefanbode, 2017-05-26:
I'm currently on holiday. Be back next week.

stefanbode, 2017-06-06:
I changed some behavior that may cause the interference with the I2C relays and the GPIO relays. Additional upgrade to the latest arendst branch. Can you check, if this now works for you?

mrmin24, 2017-06-07:
Update to version 5.1.5. Still same behaviour when switching relay1, relay 5 switches everytime relay1 switches off.

stefanbode, 2017-10-09:
@mrmin24 : To enable the usage of the onboard relays 1-4 there where major code adjustments required to get rid of several side effects. I did test it with 2 PCF8574 and up to 4 relays and I was able to reproduce your error before. Now all 20 relays behave as they should. Also, the bouncing is away and I have removed the library to save code.

closed 2017-10-09 by stefanbode

mrmin24, 2017-10-15:
@stefanbode Hi, nice work thanks. I am busy testing. So var so good. I have 9 relays including the on-board relay on the SV.