stefanbode/Sonoff-Tasmota #9: ADS1115 Support

fdsailer, 2017-11-04:
Hallo Stefan
Habe versucht zu compilieren, leider habe ich nichts lauffähiges zustande gebracht. Sonoff ist nach 15s "abgestürzt" und war danach nicht mehr erreichbar. Ich verwende Sonoff Basic und wollte eine Füllstandsensor 4-20mA mittels des ADS1115 einbinden. 
Kommt deine Unerstütung für ADS115 in absehbarer Zeit in das Sonoff-Tasmota Projekt ? oder könntest du mir eine lauffähiges bin file für den Sonoff Basic zukommen lassen ? 

Besten Dank

Michael

stefanbode, 2017-11-07:
Hi Michael, ich lade mal ein BIN hoch. Also wenn ich das richtig lese, dann hat das compilieren funktioniert aber das laufen nicht mehr wirklich. Was Du machen kannst, weil das Problem habe ich schon gesehen, wenn schon mal das oginale software drauf war. Ändere mal den CFG_HOLDER in der user_config_override.h Die kannst Du Dir aus dem orignalen Projekt holen. Ist bei mir nicht drin. ggf. dann mal SERIALLOG 5 und die letzten Zeilen vor dem Sterben hier posten. Der Support für ADS1115 ist wie alles andere ein generelles Problem mit den RAM und FLASH Speicher des ESP. Das ist halt alles schon extrem eng und auf Kante programmiert. Abstürze können auch passieren wenn Out-Of-Memory.

stefanbode, 2017-11-07:
So, aktualisiert auf 5.8.0q und die beiden /BIN/ Dateien abgelegt. Dann mal viel Spaß.

fdsailer, 2017-11-07:
Danke für die promte Antwort. Werde leider erst am WE Zeit haben um zu testen.

stefanbode, 2017-11-16:
Support also merged into the main branch. No extra code here

closed 2017-11-16 by stefanbode