stefanbode/Sonoff-Tasmota #10: bin file with pcf8574

Karl-opec, 2017-11-13:
Could you be so nice and put a compiled bin file with pcf8574 support
in your bin folder? 

I would like to test it with a 8 relay board.

Thank you very much.

stefanbode, 2017-11-15:
Do you also need MQTT TLS support. If yes you should be aware that using the webserver required do disable MQTT first.

Karl-opec, 2017-11-15:
No I don't need TLS support, I will use it in my Intranet only.

stefanbode, 2017-11-16:
New BIN shared

closed 2017-11-16 by stefanbode

Karl-opec, 2017-11-16:
Thanks you very much for providing the bin file.

I use it with the electrodragon 2 Relay board and a chinese pcf8574 and an 8 relay board.
The relays connected to the pcf8574 are working, the relays onboard don't switch anymore but they show up in the weggui.
Any Idea?