stefanbode/Sonoff-Tasmota #11: bin file with pcf8574 support

Karl-opec, 2017-11-17:
I commented this already in the closed issue. But perhaps it might be better to open a new issue.

I use your last bin file  with the electrodragon 2 Relay board and a chinese pcf8574 and an 8 relay board.
The relays connected to the pcf8574 are working, the relays onboard don't switch anymore but they show up in the weggui.
Any Idea?

stefanbode, 2017-11-17:
Thanks for informing me. I will check on the weekend. arendst changed a lot during the last weeks on relays. Not very easy to keep the rest working. I had fo change a major portion of the code already. You will get feedback asap

stefanbode, 2017-11-19:
I found the bug and fixed it. Tested today only with ONE board, but hopefully also working with more than one. BIN files updated

closed 2017-11-19 by stefanbode

Karl-opec, 2017-11-22:
Thanks for fixing it. Yes all 10 Relays are switchable now by remote commands.
The two onboard switches of the electrodragon board don't act as in arendst version.

Thanks

stefanbode, 2017-11-22:
I checked with two relays on the WEMOS and 8 relays on the pcf. For me this worked perfectly though mqtt and webinterface . I have no eletrodragon to verify.

Karl-opec, 2017-11-23:
Relay switching does also work for me by web and mqtt. 

On the electrodragon board there are two mini switches which can also switch the two relays on the board with the standard arendst software. with yours variant they don't. But it is ok for me. I wanted only mention it. 

Never mind.  Thanks for fixeing the main bug.

stefanbode, 2017-11-23:
ok, now I understand. I assume 2 GPIO are assigned as SWITCH1 and SWITCH2. They should switch RELAY1 and RELAY2. let me see. should be fixable.

Karl-opec, 2017-11-23:
[Details of the electrodragon board](http://www.electrodragon.com/w/ESP_Relay_Board_Hardware#Specific_for_Relay_Board) you find here. the switches are on gpio0 and gpio2

stefanbode, 2017-11-23:
I checked it with GPIO0 on a WEMOS board and everything is running fine. Depending on my switch ON/OFF the relay goes on or off. If you need a toggle mode you need to set SWITCHMODE 3 or 4

[image]
[image]

Karl-opec, 2017-12-03:
Could you be so nice and update the bin files to version 5.10.0?

Thanks in advance.

stefanbode, 2017-12-04:
Yes, done.