stefanbode/Sonoff-Tasmota #12: ULTRADEEPSLEEP - haw to set

roblad, 2017-11-21:
Hi

I am looking setting for deep sleep of tasmota

Thanks.

stefanbode, 2017-11-21:
Just set ‚deepsleep‘ in the console or as a Mqtt command. There is no webinterface, but you can use the CONSOLE in the webinterface to do. The Status Information shows the current setting

roblad, 2017-11-21:
deepsleep in standard tasmota does not exist, could you do a pull request to author for adding it - it is very useful. Additionaly Iam interesting to use pcf8574 and disply results from Analog input or Digital on www and send values through MQTT, could you prepare some wiki with your enhancement of tasmota it is really  good job, but without some description nobody will now haw to use it.

stefanbode, 2017-11-21:
Arendst already declined to add deepsleep, because the sonoff devices are not battery driven. Regarding the pcf8574 I currently only support the relay usage. Running it as input is not working. Will be an enhancement for the future. If you need deepsleep you can take my version here. It is almost same as the original one. Try to keep the versions in sync with 1-2 weeks of delay.

stefanbode, 2017-11-21:
for analog input you can now use the ads1115. Arendst added my extension two weeks ago to the standard. The PCf can only do digital IO

roblad, 2017-11-21:
Ok,

I need to add according to pcf8574  to Sonoff POW 3 additional DIO for checking tank level but it is good to have see status of DIO on tasmota WWW, in your mode is it possible ?

stefanbode, 2017-11-21:
Digital input signals are only reported through MQTT and/or the STATUS message that you will see through TELEPERIOD. There is no reporting of the SWITCHES in the webinterface. Only if you have a SWITCH1 and a RELAY1 you can see it again through the webinterface. As mentioned currently I do not support input signals on the PCF8574.

roblad, 2017-11-22:
Do you see ?

https://github.com/arendst/Sonoff-Tasmota/issues/1230#issuecomment-346292079

roblad, 2017-11-22:
Hi,

1) Could you give me your variables for your changes in language pack, I will translate to Polish
2) I have a question regarding to deepsleep, I am paning to use MOSFET with control by one GPIO, what is a power output when dipsleep starts, I would like to switch off the MOSFET and when module will wake up, I need to change i.e. to high for open MOSFET. 
3) do you think if I will change max curent in POW HLW8012 from 16001 to 30001 I will get extent of max current to ~30A see:

http://tinkerman.cat/hlw8012-ic-new-sonoff-pow/

  else if (CMND_CURRENTLOW == command_code) {
    if ((payload >= 0) && (payload < 16001)) {
      Settings.hlw_imin = payload;
    }
    nvalue = Settings.hlw_imin;
    unit = UNIT_MILLIAMPERE;
  }
  else if (CMND_CURRENTHIGH == command_code) {
    if ((payload >= 0) && (payload < 16001)) {
      Settings.hlw_imax = payload;
    }
    nvalue = Settings.hlw_imax;
    unit = UNIT_MILLIAMPERE;

stefanbode, 2017-11-22:
Let’s start from the beginning. Deepsleep 65535 is just an error if you came from normal tasmoto. I ignore this number to avoid unwanted behavior. Just send itvto something useful in seconds and it will work. If you want to sleep for a day use deepsleep just 24x60x60. I need to check if the size of the variable is big enough. Maybe 65000 is Max.

All and really all changes are tagged with ‚//STB mod‘ in the code. Therefore really easy to find the addition in the language pack.

The MOSFET should work, I have the same configuration. Just be aware to have a resistor between GPIO and the base of the mosfet. Otherwise the port go to low during reboot. I think an error but Arendst did not find it. Anyhow works great if the GPIO is protected with some resistor.

For deepsleep support D 0 (gpio 16) must be connected through a wire with RST. The wakeup will be initiated through this GPIO and create a reboot. There are many examples at google how to do it.

stefanbode, 2017-11-22:
Extra post because very important : 
Never ever use more than 16A AC on the device to switch. Even this is in mind a bit to optimistic. The complete layout is not made for your proposed current.

stefanbode, 2017-11-22:
Let me know how long your battery last, because I assume that the eBay item mill draw much more current than required because of the usb interface. Reallylooking forward to you first measurement of current usage in deepsleep.

roblad, 2017-11-22:
>>>Extra post because very important :
Never ever use more than 16A AC on the device to switch. Even this is in mind a bit to optimistic. The complete layout is not made for your proposed current.

Why ? I will do extra wire soldered :-) I need to know if it will measure properly - max pic I have 20 ~25 A, I do not need to use relays POW is for POW in my plan :-)

>>>>Let me know how long your battery last, because I assume that the eBay item mill draw much more current than required because of the usb interface. Reallylooking forward to you first measurement of current usage in deepsleep.

I will cut one line and add smal switch for RS/UART chip - the question is haw long  it can work on 1885 battery 2400 mAh when measure will be every 15 minutes 

>>>>> The MOSFET should work, I have the same configuration. Just be aware to have a resistor between GPIO and the base of the mosfet. Otherwise the port go to low during reboot. I think an error but Arendst did not find it. Anyhow works great if the GPIO is protected with some resistor.

Could you past me your solution schema of that I did not catch corectly , schema will give me some example :-)

stefanbode, 2017-11-23:
Regarding the "how long last a 18650 battery" is more related to how man current is drawn during deepsleep. The ESP8266 is quite optimized for battery, but the voltage regulator and for sure the USB interface in many devices draw 100x more energy because the design was never optimized on deepsleep. My battery driven devices, therefore, use a NATIVE esp8266 with a voltage regulator: HT7333-A. No USB. This go down to ESP = 10µA + HT7333-A = 8µA == > 18µA in total. (I do measure 21µA). Total idle time = 3400mAh / 21µA ~18 years. Typical uptime for my fork is 8 seconds and current is 100mA. Depending on the wakeups der day (for example 24x4 = 100) the average current through operating is:
100 x 8s / 24*60*60 s = 0,925%. Thes compareable current is: 0,925% * 100mA = 1000µA. You see this is much more than the 18µA deepsleep. Therefore this battery should last < 4month.

stefanbode, 2017-11-23:
Measuring 30A will not work because this is not a software issue, this is a hardware design problem. The shunt-resistor will see double load maybe getting too hot. It has 0.001ohm (create 7.5W !!!! at 250V and 30A). Secondly, the voltage drop you need to calculate current will be double. You cannot just put double input voltage to the HLW8012. Take a look at the datasheet what other resistors need to be changed.

roblad, 2017-11-23:
Ok,

I will see haw it will work for POW, what about MOSFET schema, do you have an example ?

stefanbode, 2017-11-23:
Take this awesome tutorial to connect your MOSFET depending on the requirement in the right way. 
If you have an iPAD I also can suggest to buy iCircuit to test and prove your design before burning the house. 

Regarding the mV discussion. The HLW8012 can take up to 495mV. Therefore agree that this hopefully will work. Maybe the problem is on the side detecting the frequency correctly with the ESP. 

http://www.electronics-tutorials.ws/transistor/tran_7.html

roblad, 2017-11-23:
Ok,

But question is:

I need start deep sleep and switch off power for sensors through MOSFET during this time - according to the schema where I need to plug Vin in ESP GPIO to get switch off MOSFET Vout, and next when deep sleep will finish and will restart ESP haw to switch on MOSFET for sensors power and again and again ?
To which GPIO I have to plug GATE-G for haveing - when module is up I have MOSFET switched on - when module have gone to deep sleep , MOSFET is switched off :-) 

https://pl.pinterest.com/pin/464785624029177998/

Haw it was prepared by you on your module ?

Do not wory about fire - node will be outside house :-) 

The HLW standard measure will be max 10A but when is some water electrical pot and iron and something else with high power consumption on the same phase, it can peak to 20 MAX  25A (I hope not) - not often but can happened for some time  

When the POWs will come to me I will start tests - right now there is only theoretical discussion, I can also  change shunt-resistor for better power i.e. 5W I found some.

stefanbode, 2017-11-23:
Yes right, but this is an easy build. For example, just connect your MOSFET to GPIO 12. Then define a normal RELAY1 on GPIO12 and switch it ON just once. The default configuration of TASMOTO is to retain the last status on boot. This means, when you go into deepsleep, GPIO12 will go LOW and your MOSFET is OFF. And with the reboot after deepsleep (RST and GPIO16 always connected) the GPIO12 goes high again , because this was the last state, that is also saved to FLASH. It also works, if you completely remove the ESP from power.

roblad, 2017-11-23:
OK Thanks for advice - nice idea :-)

look on that:
https://bitbucket.org/xoseperez/hlw8012/src/cc3baec18f87?at=master

I se that interputs also are avaliable, what do you think is it possible to wake up as in AVR through external interrupt. In that example author usage the library "power.h" as well - what do you think, I do not know what GPIO on any ESP has external interputs build in, becuse deep sleep based on RTC wake up and gives reset module. In my plan I am looking such this:

Module is on deep sleep mode, some external inpulse hapened (counter Gas sensor pulse measure) and wake it up and counting. Gas is not usage and after some time module is going to sleep to next waking up by the external interput. From time to time makes a report per 1h i.e. that module exists ond the world.
Could you tell me, how well do you know ESP, in the context of external interrupts, is this at all possible on that?

roblad, 2017-12-01:
Hi,

Success i have 100 uA during sleep with the chip CP2104 ther is 1.2 mA but I found solution. But for the switches there is black magic for me - I do not know haw to set POWER1 POWER2 "switchTopic1" "SwitchTopic2" etc. And additionaly 2 dummy devicec RELAY0 and RELAY1  for operating for 2 topicks.

stefanbode, 2017-12-01:
Hi @roblad. I found an issue in the deepsleep. it was limited to 65000 seconds due to some limitation in the used types. Is fixed now. everything up to 4 billion should be fine now.

roblad, 2017-12-01:
Hi,

What is diferenties betwen ultradeepsleep and deepsleep ?

stefanbode, 2017-12-01:
I already changed the documentation in readme.md. You just need to use deepsleep in seconds. If it is more than 1h than automatically the deepsleep is chunked in to 1h packages until total deepsleep is reached. the normal deepsleep just supports ~71 minutes in maximum, then there is a type overflow and it starts from 0 again. To stop just disconnect from power, because the remaining time is stored in RTC memory.

roblad, 2017-12-01:
Hi,

You forgot about that script in your repository

esp8266.flash.16m.ld

it have to be created and inside given it:

/* Flash Split for 16M chips */
/* sketch 999KB */
/* eeprom 20KB */
/* spiffs 15MB */

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xfeff0
}

PROVIDE ( _SPIFFS_start = 0x402FB000 );
PROVIDE ( _SPIFFS_end = 0x402FB000 );
PROVIDE ( _SPIFFS_page = 0x100 );
PROVIDE ( _SPIFFS_block = 0x2000 );

INCLUDE "esp8266.flash.common.ld"

stefanbode, 2017-12-01:
done

roblad, 2017-12-01:
Hi,

I set Switch1 and Switch2 and both of them show the same result

22:08:49 CMD: SwitchTopic 0
22:08:49 MQT: stat/wemeos/REZULTAT = {"SwitchTopic":""}
22:09:03 MQT: tele/wemeos/STAN = {"Czas":"2017-12-01T22:09:03","Uptime":0, "Wifi":{"AP":1, "SSID":"robert_private", "RSSI":86, "APMac":"84:16:F9:C8:85:E6"}, "DeepSleep":0, "Stos":25504}
22:09:03 MQT: tele/wemeos/SENSOR = {"Czas":"2017-12-01T22:09:03","Przelacznik1":"OFF","Przelacznik2":"OFF","Wej_analogowe0":3}

Switch1

22:09:06 MQT: stat/wemeos/REZULTAT = {"POWER":"ON"}
22:09:06 MQT: stat/wemeos/POWER = ON
22:09:07 MQT: stat/wemeos/REZULTAT = {"POWER":"OFF"}
22:09:07 MQT: stat/wemeos/POWER = OFF

Switch2

22:09:08 MQT: stat/wemeos/REZULTAT = {"POWER":"ON"}
22:09:08 MQT: stat/wemeos/POWER = ON
22:09:08 MQT: stat/wemeos/REZULTAT = {"POWER":"OFF"}
22:09:08 MQT: stat/wemeos/POWER = OFF
22:09:10 MQT: stat/wemeos/REZULTAT = {"POWER":"ON"}
22:09:10 MQT: stat/wemeos/POWER = ON
22:09:10 MQT: stat/wemeos/REZULTAT = {"POWER":"OFF"}
22:09:10 MQT: stat/wemeos/POWER = OFF

closed 2018-07-08 by stefanbode