stefanbode/Sonoff-Tasmota #13: Update language files

roblad, 2017-11-23:
Hi,

Could you update language files, I will translate , I do not know which variables I need to add

roblad, 2017-11-29:
You can add

//STB mod
#define D_MOISTURE "Wilgotnosc"
#define D_DISTANCE "Odleglosc"
//end

stefanbode, 2017-12-01:
Thanks for your input. Added the PL language info. Yes, the CHIRP is the one you mentioned. It works really well as a moisture sensor. Also added your configuration to the platform.ini. I do no know how and where to add the change in the board configuration.

GPIO_SEN_TRIG,
GPIO_SEN_ECHO, are for the HR-SR04 ultrasonic distance measure.

The other 4 are not used currently, They are planned for motors that need inverse voltage and therefore need two operating relays. Valid input is then -1 | 0 | 1 for backward, stop, forward.

closed 2017-12-01 by stefanbode

roblad, 2017-12-01:
Hi,

Do you have any place where can I buy it (chip ofcource :-)) CHIRP - it will solve my one configuration in the garden :-)

stefanbode, 2017-12-01:
I got mine from tindie. I found no cheaper place to by the ESP version

https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/

roblad, 2017-12-01:
What is a price to send to you ? Maybe you have some chiper site :-), we do not have EU salaries :-)

stefanbode, 2017-12-01:
Moisture measurement is quite difficult AND expensive if you want to make it reliable and get some values, that makes sense. The CHIRP is presented to me at 13USD+shipping. It has really good values, that really matches the real situation. There is also analog version: https://de.aliexpress.com/item/DFRobot-Capacitive-Analog-Soil-Moisture-Sensor-3-3-5-5V-Corrosion-Resistant-with-Gravity-3-Pin/32574020064.htm This is 8USD+shipping. It is also not bad in the beginning, but have problems after 9 months outdoor. Even using plastic spray and shrink tubes to protect the circuit board. If you search for professional compensated moisture sensors they start at 80USD. Everything that is not capacitive, just do not order. This is just for playing.

roblad, 2017-12-01:
Hi,

I compiled your new version and uploaded through www, right now I did not noticed any issue :-)