stefanbode/Sonoff-Tasmota #14: Haw to switch on for operate when deepsleep is runing ?

roblad, 2017-11-30:
Hi,

Is it possible to add deepsleep came back, when deepsleep starting there is no possibility to do any changes or update. What do you think about add button function to break deep sleep after pressing of button a couple of times ? It will give possibility to change something, after that (some period time) deep sleep will continue.

Rgds,
RL

stefanbode, 2017-11-30:
I use many of them in the garden, therefore, I have the same use-case. Currently, I send through MQTT a retained message on cmd/device/deepsleep 0. In most cases, the device with the next wake-up will read the values and stay online. In some cases it takes a second wakeup. Important is to store the message in the MQTT stack. The button you can easily solder between RST and 3.3V. This cause a wakeup, whenever you close the button. Be sure that there are no other messages stored in cmd/device because the pubsubclient can only execute ONE stored message on restore.

roblad, 2017-11-30:
Hi,

Haw to do it as retained ?

" send through MQTT a retained message on cmd/device/deepsleep 0"

PS.

Do you know haw to do button get worked without relay assigment ?

stefanbode, 2017-11-30:
The first one is the correct one because in the second one there is a space missing between password and user and it could be that default is cmnd and not cmd. Retaining a message requires -r. This is correct. IF you like it more convenient I would suggest using MQTT spy. https://kamilfb.github.io/mqtt-spy/

the POWER message could be from the device itself. It has some nasty behavior to post something in the cmnd. I don't like it, but this is the way arendst has programmed it. Having a NULL here is a side-effect when there is no RELAY. You can assign a button and it works. The multitouch gestures on the button are working. You just get no MQTT message on pressing the button e.g. 3 times. This is not implemented. You can define "switchtopic switch", then you get an MQTT message. the Format is fixed

roblad, 2017-11-30:
Hi,

What do you mean "You can assign a button and it works. The multitouch gestures on the button are working."

I do not catch that.

roblad, 2017-11-30:
I got creasy, I defined 2 switches and

19:56:09 CMD: switchtopic switch
19:56:09 MQT: stat/wemeos/SWITCHTOPIC = {"SwitchTopic":"switch"}
19:56:22 MQT: cmnd/switch/POWER = ON
19:56:23 MQT: cmnd/switch/POWER = OFF
19:56:26 MQT: cmnd/switch/POWER = ON
19:56:27 MQT: cmnd/switch/POWER = OFF
19:56:36 MQT: cmnd/switch/POWER = TOGGLE
19:56:36 MQT: cmnd/switch/POWER = TOGGLE

I do not get such efect:

https://github.com/arendst/Sonoff-Tasmota/issues/88

roblad, 2017-12-01:
Hi,

Success i have 100 uA during sleep with the chip CP2104 ther is 1.2 mA but I found solution. But for the switches there is black magic for me - I do not know haw to set POWER1 POWER2 "switchTopic1" "SwitchTopic2" etc. And additionaly 2 dummy devicec RELAY0 and RELAY1 for operating for 2 topicks.