stefanbode/Sonoff-Tasmota #15: Modyfication for pulse counter

roblad, 2017-12-03:
Hi,

What do you think (or if you can help me) to modyfi counter of pulse.

See that:

https://github.com/arendst/Sonoff-Tasmota/issues/1300

stefanbode, 2017-12-04:
In general no big deal to store the counter in RTC memory and maybe also in Flash. But we have to take into consideration how many pulses we expect in which timeframe. Additionally, it is interesting how you want to react on a reboot (accidentally or intended). Is there a requirement to remember the last state. What about power off? is it ok if we remember on 400 pulses and it was 415 when it was shut down on power?
Questions to anser:
1.) How many pulses in which timeframe you expect?
2.) Need the pulsescount survive a reboot? (Using RTC memory)
2.) Need the pulsecount survive power off (Using Flash)

stefanbode, 2017-12-04:
I do see, that the behaviour is somewhere already in the code. Normally the count will only be reported by the TELEPERIOD (afaik). This will not give you exactly the state when 1L is reached, but you will get an update on the flow, for example, every 10sek.

roblad, 2017-12-04:
Hi,

1) Pulses not in timeframe - it could be 10, 100 in the same period of time - it is flow and when tap is open fully or many taps are opened the flow will be different.
2) Currently I have AVR controller where I save value for every 100 L to EPROM. When there is power cut the latest value from EEPROM is taken - correction I send to the AFR through RMF69 and I receive through RMF69. The attache interput have 2 counters 1 is for getting 4100  for each 10 L of water and one is for counting each pulse.
3) I need to keep some last value in eeprom i.e. or RTC memory, but main is a possibility to multiply counter for i.e. 410 pulses for  1 L and result it on MQTT or/and www. The best is to set somewhere the multiplayer in some variable.

stefanbode, 2017-12-04:
Hi, yes stefan_bode at web de. Please see the newest version. introduced a new command counterdevider

closed 2017-12-04 by stefanbode

roblad, 2017-12-04:
Hi,

translation for polish for #define D_CMND_COUNTERDEVIDER "CounterDevider"

#define D_CMND_COUNTERDEVIDER "Dzielnik licznika".

I am just starting test it

stefanbode, 2017-12-07:
it should be
#define D_CMND_COUNTERDEVIDER "CounterDevider"
there is a command, not text to translate, it has CounterDevider1 or CounterDevider2 or CounterDevider3 or CounterDevider4 depends on max_counters
I suggested to change it - but I updated the response to #15

yep, there is a reason that command normally should stay in English. Changed back and will be updated with the next sync.

roblad, 2017-12-07:
Hi,

Could you try to add other things, and I sent you a mail, could you answer ?

Thanks,
RL

stefanbode, 2017-12-07:
I saw just one mail from you where you said: hello from monday. no requests

roblad, 2017-12-07:
Hi,

I have not received , could yoy try again ?