stefanbode/Sonoff-Tasmota #17: HC-SR04 feedback

boverby, 2017-12-10:
Thank you for adding support for HC-SR04. 

I have used it this morning on a christmas tree water level sensor. The only change I needed to make was upping the MAX_DEVIATION_BETWEEN_MEASURES from 0.1 to 75.00.  The original value left me wondering if the sensor was fried, but enabling the "HS-SR04 In Duration:" debug showed that it was indeed changing with new depths. The 0.1 value for MAX_DEVIATION_BETWEEN_MEASURES  was acting as an intended damping factor , but was so slow that I got confused.  I know that the sensor is "noisy" so I was thinking about taking the 10 samples and getting the median value instead.

No changes requested, just feedback that might help another user.  
Thanks again

stefanbode, 2017-12-11:
Hi @boverby,

yes the noise is quite a pain with the SR04. I already do some compensation in the code. I found out that better than the median/average is to work on the maximum. To don't wast much time I just do 10 loops each measurement. Anyhow also this method does not create good sensor out of a bad one. Therefore introduced the damping factor. this is the maximum change between two measurements between to probes.

junk-barkass, 2017-12-27:
Hi.
I'm trying to use your code with a regular son off basic. no need for deep sleep how do i disable it? I just need it to talk to my SR04. I red the above and maybe this link will help you to get better reading for the SR04. https://www.youtube.com/watch?v=6F1B_N6LuKw
Best Regards Magnus

stefanbode, 2017-12-27:
Thanks Magnus, I will give it a try. Maybe it is because of the 3.7V power and startup from Deepsleep that produce the headache. If you do not set deepsleep it should be 0 BY DEFAULT , that means OFF.

junk-barkass, 2017-12-27:
Thanks for quick reply. I'm not a programmer myself but i think it was a very good video.
I ran your code on nodemcu and it worked fine no deepsleep. But on Sonoff it deepsleep even if it's 0. 
My problem is not the SR04 sensor yet...

stefanbode, 2017-12-27:
Really strange. I’m on holiday. Therefore a bit difficult to test. There should be no difference. Anyhow. What deepsleep ist reporter in the STATUS Message? You can easily override by sending a deepsleep 0. I also have several sonoff without active deepsleep running fine.

junk-barkass, 2017-12-27:
Hello.
Enjoy your holiday!

I’ve set deep sleep to 0 but it seems not to listen to it. Yes it’s very strange indeed.
When are you back home? We can talk more then. Or just send me an email. We can discuss more then.
//Magnus

stefanbode, 2017-12-28:
Can you send the log messages during startup? Seriallog 4 should be sufficient. If it uses deepsleep you should see the corresponding messages here.

junk-barkass, 2017-12-28:
Hello.
It writes as you can see deep sleep 0, but it is so sleepy... and after a while it restart by itself.

21:20:46 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Boot Count 72
00:00:00 HS-SR04: Device configured: Trig: 14 Echo: 4
00:00:00 Project sonoff Belysning vid spisen (Topic 8, Fallback 8, GroupTopic 8) Version 5.10.0b
00:00:00 CFG: Saved to flash at F8, Count 93, Bytes 1676
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 lj3m in mode 11N as s8.larssons.us.to...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on s8.larssons.us.to with IP address 192.168.200.208
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/8/LWT = Online (retained)
00:00:05 MQT: cmnd/8/POWER = 
00:00:05 MQT: Subscribe to cmnd/8/#
00:00:05 MQT: Subscribe to cmnd/8/#
00:00:05 MQT: Subscribe to cmnd/8/#
00:00:05 MQT: tele/8/INFO1 = {"Module":"Sonoff Basic","Version":"5.10.0b","FallbackTopic":"8","GroupTopic":"8"}
00:00:05 MQT: tele/8/INFO2 = {"WebServerMode":"Admin","Hostname":"s8.larssons.us.to","IPAddress":"192.168.200.208"}
00:00:05 MQT: tele/8/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: knapp/8/RESULT = {"POWER":"OFF"}
00:00:06 MQT: knapp/8/POWER = OFF
00:00:12 HS-SR04: Device configured: Trig: 14 Echo: 4
00:00:13 MQT: tele/8/STATE = {"Time":"1970-01-01T00:00:13","Uptime":0,"Vcc":3.108,"POWER":"OFF", "Wifi":{"AP":1, "SSId":"lj3m", "RSSI":74, "APMac":"10:7B:44:CC:B7:C8"}, "DeepSleep":0, "Heap":29144}
00:00:13 HS-SR04: Start measurement: Trig: 14 Echo: 4
00:00:16 APP: (UTC) Thu Dec 28 20:21:03 2017
00:00:16 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:16 APP: (STD) Sun Oct 29 03:00:00 2017
21:21:20 HS-SR04: Avg Distance measured: 0.00
21:21:20 HS-SR04: New Distance: 0.00
21:21:20 MQT: tele/8/SENSOR = {"Time":"1970-01-01T00:00:13","Distance":}
21:21:21 MQT: tele/8/STATE = {"Time":"2017-12-28T21:21:21","Uptime":0,"Vcc":3.109,"POWER":"OFF", "Wifi":{"AP":1, "SSId":"lj3m", "RSSI":74, "APMac":"10:7B:44:CC:B7:C8"}, "DeepSleep":0, "Heap":28752}
21:21:21 HS-SR04: Start measurement: Trig: 14 Echo: 4
21:21:41 HS-SR04: Avg Distance measured: 0.00
21:21:41 HS-SR04: New Distance: 0.00
21:21:41 MQT: tele/8/SENSOR = {"Time":"2017-12-28T21:21:21","Distance":}
21:21:50 HS-SR04: Device configured: Trig: 14 Echo: 4
21:21:52 MQT: tele/8/STATE = {"Time":"2017-12-28T21:21:51","Uptime":0,"Vcc":3.108,"POWER":"OFF", "Wifi":{"AP":1, "SSId":"lj3m", "RSSI":66, "APMac":"10:7B:44:CC:B7:C8"}, "DeepSleep":0, "Heap":28472}
21:21:52 HS-SR04: Start measurement: Trig: 14 Echo: 4
21:22:12 HS-SR04: Avg Distance measured: 0.00
21:22:12 HS-SR04: New Distance: 0.00
21:22:12 MQT: tele/8/SENSOR = {"Time":"2017-12-28T21:21:52","Distance":}
21:22:12 WIF: Checking connection...
21:22:12 WIF: Connected
21:22:21 HS-SR04: Device configured: Trig: 14 Echo: 4
21:22:22 MQT: tele/8/STATE = {"Time":"2017-12-28T21:22:22","Uptime":0,"Vcc":3.108,"POWER":"OFF", "Wifi":{"AP":1, "SSId":"lj3m", "RSSI":64, "APMac":"10:7B:44:CC:B7:C8"}, "DeepSleep":0, "Heap":28384}
21:22:22 HS-SR04: Start measurement: Trig: 14 Echo: 4
21:22:42 HS-SR04: Avg Distance measured: 0.00
21:22:42 HS-SR04: New Distance: 0.00
[... 129 lines cut ...]
64 bytes from 192.168.200.208: icmp_seq=949 ttl=128 time=714.138 ms
Request timeout for icmp_seq 950
Request timeout for icmp_seq 951
Request timeout for icmp_seq 952
64 bytes from 192.168.200.208: icmp_seq=952 ttl=128 time=1712.311 ms
Request timeout for icmp_seq 954
64 bytes from 192.168.200.208: icmp_seq=955 ttl=128 time=713.365 ms
64 bytes from 192.168.200.208: icmp_seq=956 ttl=128 time=708.622 ms
Request timeout for icmp_seq 957
64 bytes from 192.168.200.208: icmp_seq=958 ttl=128 time=712.570 ms
64 bytes from 192.168.200.208: icmp_seq=959 ttl=128 time=713.835 ms
64 bytes from 192.168.200.208: icmp_seq=960 ttl=128 time=712.101 ms
Request timeout for icmp_seq 961
64 bytes from 192.168.200.208: icmp_seq=962 ttl=128 time=715.999 ms
64 bytes from 192.168.200.208: icmp_seq=963 ttl=128 time=715.304 ms
Request timeout for icmp_seq 964
Request timeout for icmp_seq 965
Request timeout for icmp_seq 966
Request timeout for icmp_seq 967
Request timeout for icmp_seq 968
Request timeout for icmp_seq 969
Request timeout for icmp_seq 970
Request timeout for icmp_seq 971
Request timeout for icmp_seq 972
64 bytes from 192.168.200.208: icmp_seq=973 ttl=128 time=215.139 ms

stefanbode, 2017-12-29:
This has nothing to do with deepsleep. This is just the normal behavior after getting a restart command. But I assume you did not type it in. It just appear. Do you have a saved command in your mqtt server? One idea. I can say for sure that from somewhere there is. ‚Restart 1‘ coming.

stefanbode, 2017-12-29:
Have you switched your trigger an echo pin? I just see the timeout and a measure of 0. this only happens if not connected or wires are wrong

junk-barkass, 2017-12-29:
Hi.
Oh.. Yes you are so right. My server checks connection with all my son off devices and if it’s not responding to a ping then it will send restart. 
So YES you are so right on.

Bu why isn’t it responding to ping as all the other? check the ping time very long time for roundtrip.

//Magnus

junk-barkass, 2017-12-29:
At the moment when the logg was taken the sensor were disconnected.

junk-barkass, 2017-12-29:
Is it possible that if no sensor/something not correct will take all time from 8266 and therefore not responding to all my ping:s?

stefanbode, 2017-12-29:
Yes right. If the sensor does not answer, it freezes the ESP for other things. I can reduce the issue a bit , but not general avoid. As soon as there is a device connected everything should be fine.

junk-barkass, 2017-12-29:
Aha!!! That’s a fully answer to my problem. Then I know that and can focus on the sensor communication.
Thank’s a lot.
//Magnus

junk-barkass, 2017-12-30:
To minimize the code which one is needed for my sonoff basic and the SR-04 sensor?
//Magnus

stefanbode, 2018-01-15:
Added check and support, if SR04 is not connected to avoid the freeze. Will be updated with 5.11 version

junk-barkass, 2018-01-16:
Great!!!

stefanbode, 2018-02-20:
The sr04 reading is now quite stable and values fit most of the time. No more changes planned for the distance sensor.

closed 2018-02-20 by stefanbode