stefanbode/Sonoff-Tasmota #19: Full AP mode

cyclemat, 2018-01-09:
Can you please add a full AP mode so you can use it in a caravan without wlan router ?

stefanbode, 2018-01-10:
The ESP does not allow these kind of complex router functionality. You have to buy a small router e.g. from. Huawei

cyclemat, 2018-01-13:
nope i mean only one Phone or Tablet can Controll The Device Without Wifi router on Client is possible or not ? i test to stop the reboot counter in AP mode and set it to the mainpage but it will not work !

stefanbode, 2018-01-16:
You can change the behavior a bit when you change the timeout:
support.ino
#define WIFI_MANAGER_SEC  180  // seconds before restart

but because the underlying variable is a uint_8 the max value is 255. Therefore you have to trick again and avoid timer decrease when it has a specific value. e.g. 255. change in WifiCheck in support.ini either.
  ```
  //STB mod
      if (wifi_config_counter < 255) {
        wifi_config_counter--;
      }
```

cyclemat, 2018-01-16:
Will test it

stefanbode, 2018-01-18:
I‘m in discussion with the main branch of this project because in AP mode you only get a fraction of the Webserver UI . This makes no sense and I voted for a change having the same UI also in AP mode. I assume in intermediate I will change this on my side, because this is also asecurity issue