stefanbode/Sonoff-Tasmota #20: Add multi-PZEM support - 3-phase support

stefanbode, 2018-01-15:
In Germany all houses are connected through a three phase power connection. To measure the complete energy usage it is required to measure all three phases. Instead of using 3 ESP8266 and PZEM-004T there is some convenience to connect all 3 PZEM to one ESP.

The trick is to use 3 TX lines communicating all over ONE RX line, because the ESP does not support more.

On Hardware wire you have to remove resistor R15 at TWO of the three devices. Please see: https://github.com/apreb/eNode/issues/1#issuecomment-354369777

roblad, 2018-01-17:
Hi,

Do you plan to came back with that ?

stefanbode, 2018-01-18:
Yes,  I have the devices already on my table and made some initial changes to support multiple devices already. My plan is to finalize and test at the weekend. I will use them productive also personally.

roblad, 2018-01-18:
Hi,
I am still waiting from China

stefanbode, 2018-01-21:
Really nasty. I got one working smoothly but still not successful getting values from the second one. As soon as there are TWO devices connected no more responses on the RX. Still debugging.

roblad, 2018-01-21:
Hi,

Probably you need to plug 3 of them

stefanbode, 2018-03-25:
Hi rob, due to toooooo many problems I will not include this into my branch. To many side effects I do not understand.

roblad, 2018-03-26:
Hi,

Any progress ?

stefanbode, 2018-03-26:
No, currently I would classify it with too complex for me. I will cancel this and measure the e phases with three wemos minis.

roblad, 2018-03-26:
Hi,

Then me to :-)

roblad, 2018-04-26:
Hi,

Have seen that ?

https://github.com/arendst/Sonoff-Tasmota/issues/2315#issuecomment-384643624

May be there is an solution n ?

roblad, 2018-05-03:
Work is going :-)

https://github.com/arendst/Sonoff-Tasmota/issues/2315#issuecomment-385266151

stefanbode, 2019-09-01:
Fully included in master firmware

closed 2019-09-01 by stefanbode