stefanbode/Sonoff-Tasmota #24: compile error

tarbax, 2018-03-03:
i get a compile error when i uncomment pcf8574

exit status 1
'pcf8574_switchrelay' was not declared in this scope

stefanbode, 2018-03-03:
Let me recheck.  Based in the frequent changes, this cold happen

tarbax, 2018-03-03:
idk if the full log helps but here it is:

/Users/tarbax/Desktop/Sonoff-Tasmota-stefan_patch1/sonoff/sonoff.ino: In function 'void setRelay(uint64_t)':
sonoff:353: error: 'pcf8574_switchrelay' was not declared in this scope
       pcf8574_switchrelay(i, state);
                                   ^
/Users/tarbax/Desktop/Sonoff-Tasmota-stefan_patch1/sonoff/sonoff.ino: In function 'void every_second()':
sonoff:1652: error: 'pcf8574_detect' was not declared in this scope
         pcf8574_detect();
                        ^
/Users/tarbax/Desktop/Sonoff-Tasmota-stefan_patch1/sonoff/sonoff.ino: In function 'void GPIO_init()':
sonoff:2103: error: 'pcf8574_Init' was not declared in this scope
     pcf8574_Init();
                  ^
/Users/tarbax/Desktop/Sonoff-Tasmota-stefan_patch1/sonoff/webserver.ino: In function 'void startWebserver(int, IPAddress)':
webserver:329: error: 'handleI2C' was not declared in this scope
       webServer->on("/i2c", handleI2C);
                             ^
/Users/tarbax/Desktop/Sonoff-Tasmota-stefan_patch1/sonoff/webserver.ino: In function 'void handleSave()':
webserver:929: error: 'pcf8574_saveSettings' was not declared in this scope
       pcf8574_saveSettings();
                            ^
exit status 1
'pcf8574_switchrelay' was not declared in this scope

stefanbode, 2018-03-04:
Hi did you compile with or without the pcf module. What have you defined in you configuration.

tarbax, 2018-03-04:
i only defined my wifi, mqtt and with this :
  #define USE_I2C                                  
  #define USE_BH1750                         
  #define USE_PCF8574                            
  //#define USE_ADS1115                          
  #define USE_BMP                             
  //#define USE_HTU                            
  //#define USE_SHT  

even when i compile without any changes except pcf8574 enabled it gives the error.

stefanbode, 2018-03-04:
If you use pcf you must enable i2c. I always use this configuration. Will check tomorrow. Looks like a side effect.

tarbax, 2018-03-15:
@stefanbode how can i fix this issue ?

stefanbode, 2018-03-16:
It works for me. I assume i2c is not enabled

tarbax, 2018-03-18:
@stefanbode 
it is enabled  if i set it in user config
#define USE_I2C 
there are no // in front of it
or do i need to configure it in another file?

stefanbode, 2018-03-25:
Any change should be made in the user—configuration—override file. The user configuration normally should not be changed. This is the intend. The version you‘re using is correct. I made some changes yesterday and have some time to play today. I will double check where this comes from. Normally when you get this message at the end the use_pcf... is NOT set. This is the only reason, that the variable is not included during compile.

stefanbode, 2018-03-25:
So, I double checked that the compile and the device are running with the current version. I changed to include the PCF now in the xdrv startup procedure, which makes it more in sync with the other modules. Anyhow I still have no clear idea why it did not work at your side. Can you double check that you have the PCF ino file in your source folder. It is now named: xdrv_20_pcf8574.ino

tarbax, 2018-03-25:
@stefanbode 
i can't find the "xdrv_20_pcf8574.ino"file and i also cannot seem to find it onto your fork.

stefanbode, 2018-03-26:
missed a sync with the commit. my fault. fixed

tarbax, 2018-03-26:
ok now it compiles sucsesfully but cant seem to get I2C to working and pcf8574 is not under configuration.
webgui console:

CMD: I2Cscan
RSL: hm/status/sonoff/RESULT = {"Command":"Unknown"}

my user configuration override(without wifi and mqtt that works):

#define USE_DS18x20                              // Optional using OneWire library for multiple DS18B20 and/or DS18S20 (+2k code)

#ifdef USE_I2C
#undef USE_I2C                    // I2C using library wire (+10k code, 0.2k mem) - Disable by //
#endif

#undef USE_BH1750                             // Add I2C code for BH1750 sensor
#define USE_BMP                                // Add I2C code for BMP/BME280 sensor
#undef USE_HTU                                // Add I2C code for HTU21/SI7013/SI7020/SI7021 sensor
#undef USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+3k code, 0.3k mem)
#undef USE_IR_RECEIVE

#ifdef USE_WS2812
#undef USE_WS2812                   // 176byte memc
#endif

#ifdef USE_PMS5003
#undef USE_PMS5003                   // 32byte memc
#endif

#ifdef USE_PZEM004T
#undef USE_PZEM004T                   // 48byte memc
#endif

//#ifdef USE_ADS1115
//#undef USE_ADS1115                   // 16byte memc
//#endif

//#undef USE_CHIRP
#define USE_PCF8574
#define USE_CHIRP

#ifdef USE_SHT3X
#undef USE_SHT3X                   // 0byte memc
#endif

#ifdef USE_SHT
#undef USE_SHT                   // 0byte memc
#endif

stefanbode, 2018-03-26:
Please correct the i2c section. you undef the variable. This is the reason your command is not working. Comment out the 3 lines and put #define USE_I2C

tarbax, 2018-03-26:
same result

```
#define USE_DS18x20                              // Optional using OneWire library for multiple DS18B20 and/or DS18S20 (+2k code)

#define USE_I2C
//#ifdef USE_I2C
//#undef USE_I2C                    // I2C using library wire (+10k code, 0.2k mem) - Disable by //
//#endif

#undef USE_BH1750                             // Add I2C code for BH1750 sensor
#define USE_BMP                                // Add I2C code for BMP/BME280 sensor
#undef USE_HTU                                // Add I2C code for HTU21/SI7013/SI7020/SI7021 sensor
#undef USE_IR_REMOTE                            // Send IR remote commands using library IRremoteESP8266 and ArduinoJson (+3k code, 0.3k mem)
#undef USE_IR_RECEIVE
```

stefanbode, 2018-03-27:
have you defined the I2C pins in the COnfiguration? The SDA and SCL pin? There is a check for I2C that the scan for sure can only run if these pins are assigned.

tarbax, 2018-03-27:
i did that. i tested this with another board and now this is working.
how do i configure my gpio pins of the pcf8574?

stefanbode, 2018-03-27:
if you have defined SDA and SCL pin and the i2scan is showing your board there is an additional Option in the webserver in configuration to define input / output pins of the pcf

tarbax, 2018-03-27:
i have no extra options in there...

tarbax, 2018-03-28:
also connected an bh1750 sensor and it gives two readings:

BH1750 Illuminance 543 lx (correct)
BH1750 Illuminance 1073676336 lx (not correct)

stefanbode, 2018-03-28:
Please report this to arenst in the original code. This sensor is not maintained by me.

tarbax, 2018-03-29:
ok and how about the pcf8574 options i have no extra menu in configuration ?

tarbax, 2018-03-29:
hmmmm not sure what is happening:

```
00:00:09 RSLT: pcf8574 0 boards
00:00:09 RSLT: Final max devices: 0, PCF8574 devices 0
23:31:34 MQT: hm/status/sonoff/STATE = {"Time":"2018-03-29T23:31:34","Uptime":"0T00:00:17", "Wifi":{"AP":1, "SSId":"Black", "RSSI":82, "APMac":"78:8A:20:4C:95:63"}, "DeepSleep":0, "Heap":24112}
23:31:34 MQT: hm/status/sonoff/SENSOR = {"Time":"2018-03-29T23:31:34","Counter1":0,"Analog0":1}
23:31:35 CMD: I2Cscan
23:31:35 MQT: hm/status/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x26"}
```

It is wired up correctly but wont detect it, tried 2 different pcf8574 and 2 wemos d1 mini pro's
:-/

stefanbode, 2018-03-30:
In your configuration is something wrong because it cannot connect to the board. I can see this in the log file  

00:00:09 RSLT: pcf8574 0 boards

Maybe SCL and SDA mixed.

tarbax, 2018-03-30:
i already tried that, also tried on other gpio pins :-/

stefanbode, 2018-03-30:
Could it be that the module is 5V only?

tarbax, 2018-03-30:
i also already tried the whole range of input voltage.

tarbax, 2018-04-02:
i ordered two more pcf8574 to be sure mine are broken or not.

tarbax, 2018-04-02:
i just confirmed that my modules are working.
when i connect only 1 pcf8574 module i get these reading on the webgui:

Analog0     1
BH1750 Illuminance       1073676304 lx
CHIRP Moisture      0.0%
CHIRP Temperature      128.5°C

stefanbode, 2018-04-02:
Please take a look above. If a Bord is detected on the I2C Bus you must see this in the log

tarbax, 2018-04-02:
```
00:00:00 RSLT: pcf8574 0 boards
00:00:00 RSLT: Final max devices: 0, PCF8574 devices 0
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_096BC4, GroupTopic sonoffs) Version 5.12.0i-2_3_0
00:00:00 WIF: Connecting to AP1 Black_Ubiquiti_2.4Ghz in mode 11N as sonoff-3012...
00:00:10 WIF: Connected
00:00:10 HTP: Web server active on sonoff-3012 with IP address 192.168.1.245
00:00:12 MQT: Attempting connection...
00:00:12 MQT: Connected
00:00:12 MQT: hm/status/sonoff/LWT = Online (retained)
00:00:12 MQT: hm/setting/sonoff/POWER = 
00:00:12 MQT: hm/status/sonoff/INFO1 = {"Module":"Generic","Version":"5.12.0i","FallbackTopic":"DVES_096BC4","GroupTopic":"sonoffs"}
00:00:12 MQT: hm/status/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-3012","IPAddress":"192.168.1.245"}
00:00:12 MQT: hm/status/sonoff/INFO3 = {"RestartReason":"External System"}
00:00:12 RSLT: pcf8574 0 boards
00:00:12 RSLT: Final max devices: 0, PCF8574 devices 0
21:00:30 MQT: hm/status/sonoff/STATE = {"Time":"2018-04-02T21:00:30","Uptime":"0T00:00:20", "Wifi":{"AP":1, "SSId":"Black_Ubiquiti_2.4Ghz", "RSSI":42, "APMac":"80:2A:A8:C7:A1:C3"}, "DeepSleep":0, "Heap":24192}
21:00:30 MQT: hm/status/sonoff/SENSOR = {"Time":"2018-04-02T21:00:30","Analog0":1,"CHIRP":{"Light":1028,"Moisture":0.0,"Temperature":128.5},"TempUnit":"C"}
```

stefanbode, 2018-04-02:
Can you check on your Board wich I2C adddress is used? Maybe just remove a jumper. I have tested it with 8 boards in parallel. The starting address is hardcore in the Pcf ino file. The detect is a simple write check on the bus. If this does not work there is a hardware issue. Can you check, if there are 10k pull-up or Pulldown resistors required for the I2C?

stefanbode, 2018-04-03:
Here is a seriallog how it should look like during startup with Loglevel 5 and above:

```
00:00:00 I2C: PFC8574 found at 0x38
00:00:00 RSLT: pcf8574 1 boards
00:00:00 RSLT: I2C Config: 255
00:00:00 RSLT: I2C shift i 0: 1. Powerstate: 0, devices_present: 0
00:00:00 RSLT: I2C shift i 1: 1. Powerstate: 0, devices_present: 1
00:00:00 RSLT: I2C shift i 2: 1. Powerstate: 0, devices_present: 2
00:00:00 RSLT: I2C shift i 3: 1. Powerstate: 0, devices_present: 3

00:00:00 RSLT: I2C shift i 4: 1. Powerstate: 0, devices_present: 4
00:00:00 RSLT: I2C shift i 5: 1. Powerstate: 0, devices_present: 5
00:00:00 RSLT: I2C shift i 6: 1. Powerstate: 0, devices_present: 6
00:00:00 RSLT: I2C shift i 7: 1. Powerstate: 0, devices_present: 7
00:00:00 RSLT: Final max devices: 8, PCF8574 devices 8
```
You can see that the device reports on address x38. This is the first possible address when ALL three jumpers are set to the right side. SLC and SDA are D1 and D2 on a Wemos board.

Please check this website for wireing the native ESP: at least SDA must have a 3.3K pull-up
http://www.gunook.com/wie-benutzt-man-die-esp8266-01-stifte/

tarbax, 2018-04-15:
so i can only dedect the first pcf ig it has adress x38 ? i can only reach as high as x27

stefanbode, 2018-04-19:
Ok in this case just habe the start address permanent in the pcf *.ino file. My devices all start at x38 an above. I can’t test on all 127 addresses because other I2C devices will also send an ok. Therefore normally the devices have no overlapping IDs

tarbax, 2018-04-22:
@stefanbode thanks! now it is detected and i have the option in my configuration menu. but there is still the blank page bug

stefanbode, 2018-04-23:
The was in older versions a bug. The current one at least with chrome and on Apple devices run fine . These are my two testing devices.

tarbax, 2018-04-23:
i gave osx and tried safari and chrome. latest version. still blank page

closed 2018-06-22 by stefanbode