stefanbode/Sonoff-Tasmota #25: Device stays active for 155 seconds before go back to sleep

taste66, 2018-03-13:
Thanks a lot for this nice fork of my favorite ESP8266 firmware. I use it on the ESP01 module which I want to power with 2 penlights. Going into and recovering from Deepsleep works fine. However it takes 155 seconds before it goes to sleep again. It also send a number of MQTT messages which would not be needed:
{"WebServerMode":"Admin","Hostname":"ESP01-01","IPAddress":"192.168.2.111"}
{"Module":"Generic","Version":"5.12.0c","FallbackTopic":"DVES_360491","GroupTopic":"sonoffs"}
qos : 0, retain : false, cmd : publish, dup : false, topic : cmnd/ESP01/01/POWER, messageId : , length : 21
{"Time":"2018-03-13T22:36:41","Uptime":"0T00:02:35","Vcc":3.422, "Wifi":{"AP":1, "SSId":"xxxx", "RSSI":48, "APMac":"F8:D1:11:xx:xx:xx"}, "DeepSleep":900, "Heap":23184}

Any idea why this happens?

stefanbode, 2018-03-14:
Yes. please set the TELEPERIOD = 10 to get minimum wakeup time. This ensures all measurement is done. Normally Teleperiod is 5 minutes. Therefore DEEPSLEEP might keep the device up and running for 5 minutes.

taste66, 2018-03-14:
Thanks a lot, this did the trick now only 15s active so I can start testing with real batteries :-)

closed 2018-07-08 by stefanbode