stefanbode/Sonoff-Tasmota #26: Device hangs, when HC-SR04 sensor is connected

helmar74, 2018-03-20:
Hi, 

I installed this tasmota port on a 8266 Nodemcu. I connected a HC-SR04 ultrasonic sensor on 2 ports. The distance is measured well, but after a few seconds the Nodemcu get's inaccessible. The webinterface isn't reached any longer and i get ping timeouts.

After resetting the device it works again for a few seconds, before it get's down again.
This also occurs if I only select that a HC-SR04 is connected to 2 GPIOs. It doesn't matter if a sensor is connected physically or not....

Thanks a lot 

Helmar

stefanbode, 2018-03-20:
This issue is new. Can you please recheck, if the behavior is reproduce able with a different SR04. Currently I assume a hardware defect, that causes the nodemcu wait to long for a sound reply. The behavior is known, if you disconnected the device.

helmar74, 2018-03-20:
Hi, 
Nodemcu is fine, I can run a lot of different sketches on it, and also "normal" tasmota firmware

It doesn't matter if a SR04 is connected or not. I only selected as Device "18-generic) and for 
D2 GPIO4 - 68 SNS_TRIG
D1 GPIO5 - 69 SNS_ECHO
[image]

the SR04 isn't connected. After saving the settings the device isn't reachable any longer or really slow:
[image]

stefanbode, 2018-03-20:
As mentioned a disconnected device will always behave this way. The esp is a single core CPU. Is it wait for the signal nothing else will work. The sketch will only work with a connected and well working SR04

helmar74, 2018-03-20:
OK understood. I also figured out, that the distance is measured correctly. A 70cm Distance is show  35cm:
[image]

whhsn, 2018-03-24:
Hi Helmar,

Do you still connect the sensor as:
   D2 GPIO4 - 68 SNS_TRIG
   D1 GPIO5 - 69 SNS_ECHO

I am trying this fork but I am getting strange behavior from the sensor - distance always > 1000 cm!!!

I am connecting the sensor as:
   D1 GPIO5   - 68 SNS_TRIG
   D5 GPIO14 - 69 SNS_TRIG

Did you manage to figured it out?

Thanks,

WH

stefanbode, 2018-03-24:
I haven’t seen this behavior on my SR04. Some of the GPIO on the wemos have 10k pull-up or Pulldown resistors. But I  do not believe that this causes the issue. Do you have connected the SR04 to the 5V. This is important because the module does not work with 3.3V. My configuration also uses D5 as echo pin.

stefanbode, 2018-03-24:
The reported value is the timeout value, if there is no response detected

whhsn, 2018-03-25:
The module is connected to the 5V & I tried with & without a capacitor on the Vcc but same results.
I noticed that the large values are obtained when  the sensor is covered or facing a distant wall (I think more than 2 meters or so)

I turned on all logging & I got this:

15:57:24 HS-SR04: **_Avg Distance measured: 8.80_**, loops: 20
15:57:24 HS-SR04: New Distance: 3445.62
15:57:27 HS-SR04: **_Avg Distance measured: 9.12,_** loops: 20
15:57:27 HS-SR04: New Distance: 3445.52
15:57:29 HS-SR04: **_Avg Distance measured: 17.69_**, loops: 20
15:57:29 HS-SR04: New Distance: 3445.42
15:57:31 HS-SR04: **_Avg Distance measured: 22.23_**, loops: 20
15:57:31 HS-SR04: New Distance: 3445.32
15:57:34 HS-SR04: **_Avg Distance measured: 31.06_**, loops: 20
15:57:34 HS-SR04: New Distance: 3445.22
15:57:36 HS-SR04: **_Avg Distance measured: 18.63_**, loops: 20
15:57:36 HS-SR04: New Distance: 3445.12

You see that Avg distance measured is the right distance.
I'll check the code within a day or two to see what went wrong.

stefanbode, 2018-03-25:
As posted in the other issue there is a limit defined about the maximum change between two values. Please take a look in the wiki here.

helmar74, 2018-03-26:
Hi @stefanbode @whhsn, 
connected that way:
[image]
Always get round about 22cm Distance, no matter how long it is in real ...
[image]

stefanbode, 2018-03-26:
Take a look at the WIKI here. There is a 0.1mm max deviation between two measurements. If you don't want to have this change the constant: MAX_DEVIATION_BETWEEN_MEASURES . You can override the value in user_config_override.ino

Otherwise, you should see on the MAIN Web page that the value will constantly change by 0.1mm. Please feedback

stefanbode, 2018-05-02:
Changed CODE to main project because SR04 now supported in original project

closed 2018-05-02 by stefanbode