stefanbode/Sonoff-Tasmota #27: Problem with distance measurement

whhsn, 2018-03-24:
Hi Stefan,

I am trying your firmware on a WeMos D1 mini but I can't get it to work.

I am connecting the Trig to D1& Echo to D5 (Sensor as stated in the label)

I am getting wrong distance .... always > 1000 cm!!!

Any suggestions?

Thanks

WH

whhsn, 2018-03-24:
I turned on all logging & I got this:

15:57:24 HS-SR04: _**Avg Distance measured: 8.80**_, loops: 20
15:57:24 HS-SR04: New Distance: 3445.62
15:57:27 HS-SR04: **_Avg Distance measured: 9.12_**, loops: 20
15:57:27 HS-SR04: New Distance: 3445.52
15:57:29 HS-SR04: **_Avg Distance measured: 17.69_**, loops: 20
15:57:29 HS-SR04: New Distance: 3445.42
15:57:31 HS-SR04: **_Avg Distance measured: 22.23_**, loops: 20
15:57:31 HS-SR04: New Distance: 3445.32
15:57:34 HS-SR04: **_Avg Distance measured: 31.06_**, loops: 20
15:57:34 HS-SR04: New Distance: 3445.22
15:57:36 HS-SR04: **_Avg Distance measured: 18.63_**, loops: 20
15:57:36 HS-SR04: New Distance: 3445.12

You see that Avg distance measured is the right distance.
I'll check the code within a day or two to see what went wrong.

WHT

stefanbode, 2018-03-25:
Ahh,now I see clearly. There is damping factor that ensures values do not vary that much from one measure tot he other. Is described in the Wiki. Your first measurement was 3m. Now any additional measurement will change the value maximum 0.1mm. If you repower or change the limit from e.g. 0.1 to 1000 you should see the measured values. Anyhow the deviation is really high

whhsn, 2018-03-26:
It is working :)
Thanks for your great work

helmar74, 2018-03-26:
I'm measuring distances which can varies about 10-20cm between two measurings.
Usage:
  Device is in deepsleep for 1hour, than measuring

Example
12:00 distance is  180cm
13:00 distance is 181cm
14:00 distance is 200cm
15:00 distance is 182cm
16:00 distance is 183cm

Which settings should I do?
Have set deepsleep to 3600 and teleperiod to 10 (seconds)
How should I set MAX_DEVIATION_BETWEEN_MEASURES, and how must i define it in user_conf_override.h?

stefanbode, 2018-03-26:
For the override. Please put the right spelling just posting from my iPad the concept

#ifdef MAX_DEVIA......
#undef MAX_DEVI.....
#endif
#define MAX_DEVIATIO...... = 100

helmar74, 2018-03-28:
Hi again,

 tried these setting in user_conf_override.h:
#ifdef MAX_DEVIATION_BETWEEN_MEASURES
#undef MAX_DEVIATION_BETWEEN_MEASURES
#endif
#define MAX_DEVIATION_BETWEEN_MEASURES 100

The following compile error occurs:
Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 26 MHz, 40MHz, DOUT, 115200, 1M (no SPIFFS), nodemcu, Disabled, None"

In file included from C:\xxxxxxxxxxxxxxxxxx\sonoffAkku.ino:34:0:
user_config_override.h:38: error: expected unqualified-id before numeric constant

 #define MAX_DEVIATION_BETWEEN_MEASURES 100
                                        ^
C:\xxxxxxxxxxxxxxxxxx\xsns_19_HC-SR04.ino:23:7: note: in expansion of macro 'MAX_DEVIATION_BETWEEN_MEASURES'

 float MAX_DEVIATION_BETWEEN_MEASURES = 0.1;

       ^

exit status 1
expected unqualified-id before numeric constant

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

If i comment out the line "#define MAX_DEVIATION_BETWEEN_MEASURES 100" I can compile without error. 
I also tried the followin lines:
#define MAX_DEVIATION_BETWEEN_MEASURES=100
#define MAX_DEVIATION_BETWEEN_MEASURES=100;

All lines bring me compile errors.

Can you please help?

whhsn, 2018-03-28:
Hi helmar74,
MAX_DEVIATION_BETWEEN_MEASURES is a float variable - you cannot override it in user_config_override.h.

In the file **_xsns_19_HC-SR04.ino_** you just change the number from 0.1 to 100 - like that:

float MAX_DEVIATION_BETWEEN_MEASURES = 100;

stefanbode, 2018-03-29:
I made the value dynamically by getting an idea how strong your values change. the measured value is sometimes a little behind the current value if this changes at a constant rate. Anyhow better than before. Also removed some timing issues, if there is no echo detected.

stefanbode, 2018-05-06:
This not w part of the main project. Behavior is different now

closed 2018-05-06 by stefanbode