stefanbode/Sonoff-Tasmota #28: Issue with deepsleep

helmar74, 2018-03-26:
Hi, I've 

got a ESP8266 Nodemcu . I want to realize the follwoing scenario.
Wakeup of Nodemcu, measure Distance, deepsleep for 1 hour.
I tried to send the following command:

18:31:13 CMD: deepsleep 3600
18:31:13 MQT: stat/sonoffAkku/RESULT = {"DeepSleep":"3600 (3600)"}

But the NodeMCU doesn't fall to deepsleep. I also connected D0 to RST, this doesn't help..

Any idea, what to do?

Thanks a lot!

stefanbode, 2018-03-26:
set teleperiod 10
This causes a direct measure after wakeup. deepsleep will only happen after successfull measurment.

keep me posted.

helmar74, 2018-03-26:
OK, tried that:
11:45:40 MQT: stat/sonoffAkku/RESULT = {"TelePeriod":"10"}
11:45:41 MQT: tele/sonoffAkku/STATE = {"Time":"2018-03-26T11:45:41","Uptime":"0T00:00:17","Vcc":2.984, "Wifi":{"AP":1, "SSId":"hs.net_EG", "RSSI":58, "APMac":"9C:C7:A6:F7:72:A2"}, "DeepSleep":10, "Heap":22672}
11:45:41 MQT: tele/sonoffAkku/SENSOR = {"Time":"2018-03-26T11:45:41","Distanz":70.21}
11:45:51 MQT: tele/sonoffAkku/STATE = {"Time":"2018-03-26T11:45:51","Uptime":"0T00:00:27","Vcc":2.983, "Wifi":{"AP":1, "SSId":"hs.net_EG", "RSSI":60, "APMac":"9C:C7:A6:F7:72:A2"}, "DeepSleep":10, "Heap":22744}
11:45:52 MQT: tele/sonoffAkku/SENSOR = {"Time":"2018-03-26T11:45:51","Distanz":70.22}
11:46:02 MQT: tele/sonoffAkku/STATE = {"Time":"2018-03-26T11:46:02","Uptime":"0T00:00:38","Vcc":2.982, "Wifi":{"AP":1, "SSId":"hs.net_EG", "RSSI":60, "APMac":"9C:C7:A6:F7:72:A2"}, "DeepSleep":10, "Heap":22744}
11:46:02 MQT: tele/sonoffAkku/SENSOR = {"Time":"2018-03-26T11:46:02","Distanz":70.32}
11:46:05 CMD: deepsleep 15
11:46:05 MQT: stat/sonoffAkku/RESULT = {"DeepSleep":"15 (15)"}

Device has gone to sleep after measurement but didn't wake up...

stefanbode, 2018-03-26:
You need to wire the RST pin with the D0 pin . Take a look at google to get more info

stefanbode, 2018-03-27:
Please close the issue, if fixed.

helmar74, 2018-03-27:
It's working with RST to D0 wire

closed 2018-03-27 by helmar74