stefanbode/Sonoff-Tasmota #30: MQ-2 library

whhsn, 2018-03-29:
Good code for MQ-2 sensor:
http://www.mybotic.com.my/products/MQ-2-General-Combustable-Gas-Sensor-Module/308

stefanbode, 2018-03-29:
For all of this MQ sensors, you do not need any code. There is a pure analog signal from 0..5V that cames out and must execute through an A/D chip. The ESP has one the A0. You can use this with a voltage divider before. Please take a look here in the Wiki. If you need more to connect use the ADS1115 board. This is natively supported by TASMOTO.

Finally: There is nothing to add :-)

stefanbode, 2018-03-29:
Just saw the log and pow functions used in the calculations. We already had some discussions in the energy metering. To use this the math() library needs to be included and this will jeopardize the RAM. Sorry. This is too much for the ESP

stefanbode, 2018-03-29:
Ok, I have added the whole sensor suite. Just now need to add some documentation how to define the sensors and do the calibration. the wiki need some update. Sensors are 5v and need a 165kOhm resistor before A0, if using the WEMOS D1 mini.

roblad, 2018-03-30:
Hi,

There is similar code in my ino 

https://github.com/stefanbode/Sonoff-Tasmota/blob/master/sonoff/xsns_94_MQ_SENSOR.ino

i did not observed any issues hangs during work because of pow() and log() functions. I have one issue in Sonoff 4ch pro. hangs when wifi disapeared, but I do not know if there is a MQ code problem, I have to test it , but I asked Stefan for some bin compilation because my platformIO is not updated yet to latest version.

but in config file there is some additional declaration needed 

#define USE_MQ_SENSOR
and for voltage measure

#define USE_VOLTAGE

but both of them could not work simultaneously.

stefanbode, 2018-03-30:
The bin is updated

roblad, 2018-03-31:
Hi,

I tryied but:

Sonoff 4CH Pro Module
Sonoff
Upload Failed

Magic byte is not 0xE9

stefanbode, 2018-03-31:
How much flash has this device. If 512k, then we have the issue. The image is larger than it can be.

Sent from remote phone

stefanbode, 2018-03-31:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro

Maybe this helps

whhsn, 2018-03-31:
@stefanbode - 
I am using the sensor without any resistor with the WeMos!
I have no way to check if the collected ppm data is right or not - but it seems ok - increasing with smoke & lighter gas - this is good for my usage.

roblad, 2018-03-31:
Hi,

I upgraded and work perfectly with my mq5 sensor no any hangs as in previous releases sonoff 4ch pro with gas sensor mq5 and DHT 23kb free memory

roblad, 2018-03-31:
@whhsn 

Hi,

You have to use resistor the voltage  value you get when ppm is small is approx. below 3.3 V maximum voltage from sensor for CH4 concentration could be 4.5 V

closed 2018-07-18 by stefanbode