stefanbode/Sonoff-Tasmota #31: No distance in the sonoff-adapter witch HC-SR04

RamsesIoBroker, 2018-04-02:
Hi and thank you for this project.

I use your software on a nodeMCU with ESP8266. On the web page I see the correct value and the ioBroker also get the distance. But I have the problem that there is no line with a distance value under the object sonoff.0

I use iobroker.admin 3.3.5 and sonoff-adapter 2.0.2

[image]

[image]

[image]

stefanbode, 2018-04-02:
Is this democitz? As you can see, I report the value inside the teleperiod as json value distance. If you give me an idea where you need it, i can make a change

RamsesIoBroker, 2018-04-02:
No domoticz. 
I thought that the value will be shown like temperature, humidity or the vcc and RSSI in our case. Like the normal behaviour of the sonoffs. 
If I look in the log I see that every report of a value will be followed by a stateChange command but there is nothing behind the distance message. Is this normal?

stefanbode, 2018-05-02:
SR04 is now replaced by the official version form main branch.

closed 2018-05-02 by stefanbode