stefanbode/Sonoff-Tasmota #32: Will the fork be merged back to the original Sonoff-Tasmota including the PCF8574

knopserl, 2018-04-02:
Hi Stefan, I like your form, because I use it with ESP01 and PCF8574 modules and a 8 port relay-bank - see the attached picture. It works fine.
My question is, if your enhancements mainly the PCF8574 will be merged back to the original Sonoff-Tasmota project from arendst? I have seen, that you are in tight contact with hom.

[image]

stefanbode, 2018-04-02:
Great to see. Maybe arendst will add the fork if fix also the input functionality and the trigger. Both are currently not working. Ok the input might work but the trigger for sure not

knopserl, 2018-04-03:
Is there a PR at the original project open to merge it? I have not seen it.

stefanbode, 2018-04-19:
No, in this case I had to separate this change from all the others. Too much work.

stefanbode, 2018-05-02:
The SR04 support is now in the main project. From time to time changes will get it to the main.

closed 2018-05-02 by stefanbode