stefanbode/Sonoff-Tasmota #33: Temporary disable deep sleep for maintenance

gheesung, 2018-04-09:
Hi 

Thanks for the good extension to the fine Tasmota firmware. 

Occasionally, I may need to temporary disable the deep sleep to reconfigure some parameters. Is there a way disable it like in ESP-EASY firmware [SleepMode](https://www.letscontrolit.com/wiki/index.php/SleepMode) via a jumper setting?

Thanks
Ghee Sung

stefanbode, 2018-04-09:
Hi Ghee, interesting idea. Currently, I use MQTT and retain the message "deepsleep 0". Mostly this helps after a reboot that the device keeps awake. Make a suggestion for the switch how it should work. Interested to implement.

gheesung, 2018-04-09:
Hi @stefanbode thanks for your suggestion, using MQTT retain message can do too. 

For the ESP Easy firmware, as suggested in their wiki, I make a small 3 pin slide switch to enable/disable deep sleep on my battery powered wemos d1 mini.

stefanbode, 2018-04-09:
Hi Ghee, please see my latest commit. You can now define a GPIO as "Deepsleep Switch". If open or not connected the device will operate normally. If connected to ground deepsleep will be disabled. I checked it only with D7 gpio on a wemos. Seems to work

stefanbode, 2018-04-10:
[image]