stefanbode/Sonoff-Tasmota #34: mqtt tele/sensor data

kodibrain, 2018-04-12:
hi, after the deep sleep mqtt does not send my sensor data. But why?
I have changed the teleperiod to 30 but I get no sensor data

stefanbode, 2018-04-13:
For deepsleep normally teleperiod =10 should be ok. Do you get your data without deepsleep? Please can you send the log seriallog=5 and what sensor you want to gather

kodibrain, 2018-04-13:
How do I find the serial log?
the AM2301 sensor sends if deep sleep is set to 0

ajanulis, 2018-05-11:
Hi @stefanbode, first of all - huge thanks for this fork - it's so useful to have tasmota with deep sleep (in my opinion it's a shame that Theo Arends doesn't think the same). Anyway, you did a great job. But, I am struggling with the same issue as @kodibrain - mqtt messages with BME280 are sent if DEEPSLEEP = 0 or even DEEPSLEEP = 10, but stops if I go above (using ESP12, 5.13.1a compiled with Arduino core 2.4.1).  Would be graceful for any hint how to figure out what cause it. (TELE_PERIOD            30, but I guess it doesn't make any difference). By the way, can you point me to the file holding "core" program running Main loop (I am new to Tasmota, but very excited that I found it).

Thank you in advance,
Aidas

m-anish, 2018-06-13:
I am observing this issue as well. My setup is the following:

NodeMCU board ESP12E connected to BME280 over pins D1 (SDA) and D2 (SCL).

Teleperiod seems to control what the delay is between each mqtt publish. 

Now. When I set:

A.  Teleperiod=N, Deepsleep=0
I get a publish every N seconds

B. Teleperiod=10, Deepsleep=(I tried 30, 60, 120)
The board boots after the prescribed duration, but the sensor data does not get published

C. Teleperiod=300, Deepsleep=30
Sensor data gets published sometimes

Here are the loglevel=5 logs for

A. Teleperiod=10, Deepsleep=0
`
00:00:00 APP: Boot Count 61
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_05FB6A, GroupTopic sonoffs) Version 5.13.1a-2_4_0
00:00:00 CFG: Saved to flash at F8, Count 92, Bytes 2788
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 cygnus x-1 in mode 11N as sonoff-7018...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff-7018.local with IP address 192.168.225.26
00:00:06 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff/POWER = 
00:00:07 MQT: Subscribe to cmnd/sonoff/#
00:00:07 MQT: Subscribe to cmnd/sonoffs/#
00:00:07 MQT: Subscribe to cmnd/DVES_05FB6A/#
00:00:07 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"5.13.1a","FallbackTopic":"DVES_05FB6A","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-7018","IPAddress":"192.168.225.26"}
00:00:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"External System"}
00:00:07 APP: (UTC) Wed Jun 13 10:17:21 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
11:17:22 MQT: tele/sonoff/STATE = {"Time":"2018-06-13T11:17:22","Uptime":"0T00:00:08","Vcc":3.050, "Wifi":{"AP":1, "SSId":"cygnus x-1", "RSSI":60, "APMac":"40:C8:CB:11:50:BA"}, "DeepSleep":0, "Heap":16712}
11:17:31 SHT: Sensor did not ACK command
11:17:31 I2C: BME280 found at 0x76
11:17:32 MQT: tele/sonoff/STATE = {"Time":"2018-06-13T11:17:32","Uptime":"0T00:00:18","Vcc":3.050, "Wifi":{"AP":1, "SSId":"cygnus x-1", "RSSI":62, "APMac":"40:C8:CB:11:50:BA"}, "DeepSleep":0, "Heap":16360}
11:17:32 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-13T11:17:32","BME280":{"Temperature":23.6,"Humidity":52.7,"Pressure":803.0},"TempUnit":"C"}
11:17:40 WIF: Checking connection...
11:17:40 WIF: Connected
11:17:41 SHT: Sensor did not ACK command
11:17:42 MQT: tele/sonoff/STATE = {"Time":"2018-06-13T11:17:42","Uptime":"0T00:00:28","Vcc":3.050, "Wifi":{"AP":1, "SSId":"cygnus x-1", "RSSI":62, "APMac":"40:C8:CB:11:50:BA"}, "DeepSleep":0, "Heap":16360}
11:17:42 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-13T11:17:42","BME280":{"Temperature":23.6,"Humidity":53.7,"Pressure":803.0},"TempUnit":"C"}
11:17:51 SHT: Sensor did not ACK command
11:17:52 MQT: tele/sonoff/STATE = {"Time":"2018-06-13T11:17:52","Uptime":"0T00:00:38","Vcc":3.050, "Wifi":{"AP":1, "SSId":"cygnus x-1", "RSSI":62, "APMac":"40:C8:CB:11:50:BA"}, "DeepSleep":0, "Heap":16360}
11:17:52 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-13T11:17:52","BME280":{"Temperature":23.6,"Humidity":54.2,"Pressure":803.0},"TempUnit":"C"}
`

B. Teleperiod=10, Deepsleep=60
`
00:00:00 APP: Boot Count 63
[... 62 lines cut ...]
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-7018.local with IP address 192.168.225.26
00:00:06 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff/POWER = 
00:00:07 MQT: Subscribe to cmnd/sonoff/#
00:00:07 MQT: Subscribe to cmnd/sonoffs/#
00:00:07 MQT: Subscribe to cmnd/DVES_05FB6A/#
00:00:07 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"5.13.1a","FallbackTopic":"DVES_05FB6A","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-7018","IPAddress":"192.168.225.26"}
00:00:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Deep-Sleep Wake"}
00:00:07 APP: (UTC) Wed Jun 13 10:29:21 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
11:29:28 SHT: Sensor did not ACK command
11:29:28 I2C: BME280 found at 0x76
11:29:29 MQT: tele/sonoff/STATE = {"Time":"2018-06-13T11:29:29","Uptime":"0T00:00:15","Vcc":3.054, "Wifi":{"AP":1, "SSId":"cygnus x-1", "RSSI":72, "APMac":"40:C8:CB:11:50:BA"}, "DeepSleep":60, "Heap":16144}
11:29:29 MQT: tele/sonoff/SENSOR = {"Time":"2018-06-13T11:29:29","BME280":{"Temperature":24.3,"Humidity":52.0,"Pressure":802.9},"TempUnit":"C"}
11:29:29 MQT: tele/sonoff/LWT = Offline
11:29:29 MQT: tele/sonoff/UPTIME_S = {"Time":"2018-06-13T11:29:29", "Uptime_s":15}
`

m-anish, 2018-06-13:
So in C. above, the board doesnt actually wait 5 minutes before publishing data, and it does get published, but B. (teleperiod=10, deepsleep=60) does not work.

Note, I could try and get it to work with rules, but it would be ideal if this worked as documented (or maybe I'm doing it wrong)

Thanks for your amazing work btw! Great project!

stefanbode, 2018-06-13:
ok, gents. agree. The proposed behavior should be that on teleperiod=10 the board detects al sensors do one measurement and then go into deepsleep. From the logs you provided it looks like that the recognition of the BME sensor does not work in time and therefore it goes into deepsleep without publishing a value. 

I2C: BME280 found at 0x76

this is the important line that must be logged before going into deepsleep. Otherwise, it is clear that there is no measurement because the program assumes there is no measuring device. Also in A you can see the wrong behavior because the first state is without the BME and then there is the detect of the board and the next measurement is correct. This should be "fixable". I have some time today evening in the hotel to sync to the newest version but have no ESP with me to test. Willl upload it anyway and do the testing at Friday.

stefanbode, 2018-06-15:
Ok, the root cause is now clear. Normally the timer gets executed every second and ONE second before it initate the I2C scan. But on the boot process the gettime function that calls the timeserver takes so long, that it misses the second before the measurement.

closed 2018-07-08 by stefanbode