stefanbode/Sonoff-Tasmota #37: Update Tasmota?

kskenyon, 2018-05-02:
Tasmota 5.13 has been released.  Time for an update?

Deepsleep is very useful with the new Wemos D1 with 18650 but this build doesn't seem to be working correctly with it.  Web interface saves are not persistent.  Regular Tasmota works.

Kevin

stefanbode, 2018-05-02:
Hi Kevin, Webinferface savings ONLY work if the device is NOT in deepsleep. Should be clear. You can use the "wakeup pin" as described in the wiki to keep the device online. Then the web interface will work. After finishing you can bring it back to deepsleep.

kskenyon, 2018-05-02:
Thanks, I knew that.  This Wemos D1+18650 was not in deepsleep at the time. It has a esp8266-wroom-02 board.  Not sure what is different about this particular board but it works OK with regular Tasmota.  Sonoff-Tasmota-Deepsleep works OK with other D1 mini boards. One has been running for weeks on two 18650’s in parallel.  

Will try the console to see if it will take commands that way.

Kevin

roblad, 2018-05-02:
Hi,

There is a lot of changes witch 2.4.1, new area for config is used and esp8266-wroom-02 module should be chosen, I will test it but I thing that should not be any problems with Wemos D1 with 18650.

stefanbode, 2018-05-02:
Updated to latest master release. @roblad: Will implement your changes on the Counterdevider asap. Maybe tomorrow. For the long term it should also work with 2.4.1. Let me know what I can change.

closed 2018-05-02 by stefanbode

stefanbode, 2018-05-03:
If you have more than 1 sensor the parasite power might not work because the current is too high to support all of them. You need to connect them to a stable 3.3V and Ground connection. 8x°C is a typical reply if the measurement was not successfully. I have similar problems with the LiIon 3.7V powered devices. If the battery is "end of life" you will also see this in wrong measures of the temperature.

roblad, 2018-05-03:
Hi,

Even I use 1,5 k resistor betwen data and "+" now is better, but I think that conversion takes more time, but I changed resolution and it did not help, may be 5 V will solve the problem ? but I do not know if esp will not crash on the gpio. Do uou know any devider from 5 to 3.3 for DS ?

stefanbode, 2018-05-03:
Should be no problem:

http://forum.arduino.cc/index.php?topic=233644.0

roblad, 2018-05-03:
Hi,

I did 5v vdd and pull-up to 3.3v through 1k, looks better, but sometimes after reboot there is 1 lost, I will monitor