stefanbode/Sonoff-Tasmota #38: PCF8574 support

Karl-opec, 2018-05-06:
as i understand pcf8574 support is disabled. I enabled it and compiled it on latest git(148fdc3).

1) to get it compiled i needed to disable USE_TIMERS and USE_TIMERS_WEB otherwise
 it showed an error because of a duplicate label case7 in webserver.ino

```
sonoff-sb-20180506/sonoff/webserver.ino: In function 'void HandleSaveSettings()':
webserver:1140: error: duplicate case value
   case 7:
   ^
webserver:1108: error: previously used here
   case 7:
   ^
exit status 1
duplicate case value
```
2) there are two many Relays shown. the elektrodragen board has 2 Relays on Board the PCF8574 adds 8. In Total it should be 10.

[image]

As the picture shows, there are 18 (10 plus 8) the Relays 3 to 10 do nothing the Relays 1,2 and 11-18
can be switched by the webgui.

How can i help to fix this?.

Best Regards

stefanbode, 2018-05-06:
The Webserver one seems to be a big I can correct easily tomorrow. Regarding the ElektroDragon this is more complicated because for the PCF I scan the I2C bus and any of the valid channels count as one PCF with 8 relays. I assume my algorithm detects the electrodragon as a PCF board. Can you share in debug log the start procedure. Maybe it helps to change the I2C address. Further investigation is required

Karl-opec, 2018-05-06:
Hello these are messages i get after reboot:
```
00:00:00 RSLT: pcf8574 1 boards
00:00:00 RSLT: Final max devices: 10, PCF8574 devices 8
00:00:00 Project sonoff Sonoff (Topic sonoff_ed1, Fallback DVES_338C76, GroupTopic sonoffs) Version 5.13.1a-2_4_1
00:00:00 WIF: Connecting to AP1 xxxxx in mode 11N as sonoff_ed1-3190...
16:49:35 WIF: Connected
16:49:35 DNS: Initialized
16:49:35 HTP: Web server active on sonoff_ed1-3190.local with IP address 192.168.40.50
16:49:36 MQT: Attempting connection...
16:49:36 MQT: Connected
16:49:36 MQT: tele/sonoff_ed1/LWT = Online (retained)
16:49:36 MQT: cmnd/sonoff_ed1/POWER = 
16:49:36 MQT: tele/sonoff_ed1/INFO1 = {"Module":"ElectroDragon","Version":"5.13.1a","FallbackTopic":"DVES_338C76","GroupTopic":"sonoffs"}
16:49:36 MQT: tele/sonoff_ed1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_ed1-3190","IPAddress":"192.168.40.50"}
16:49:36 MQT: tele/sonoff_ed1/INFO3 = {"RestartReason":"Software/System restart"}
16:49:36 RSLT: Final max devices: 18, PCF8574 devices 16
```
It seams that it detects first the correct situation and later it things there is one additional pcf8547.

Best regards

stefanbode, 2018-05-06:
Oh yes, this is interesting and for sure an unwanted behavior. I will try to fix this asap

stefanbode, 2018-06-19:
The behavior is fixed and tested with the 6.0.0 upgrade

Karl-opec, 2018-06-19:
Where do i need now enable PCF8574 support?

stefanbode, 2018-06-19:
No change to the old behavior. SCL and SDA must be defined and use_pcf8574 must be enabled. if at least one board is found during scan the webinterface will show a configuration page. Please see wiki.

#define USE_PCF8574

Karl-opec, 2018-06-21:
Thanks for fixing the bug. It works now.
But i recognized two  curious MQTT messages. Switching relay 1 to 8  gives messages like this
```
Client mosqsub|4233-vdr received PUBLISH (d0, q0, r0, m0, 'stat/sonoff_ed1/RESULT', ... (16 bytes))
stat/sonoff_ed1/RESULT {"POWER3":"OFF"}
Client mosqsub|4233-vdr received PUBLISH (d0, q0, r0, m0, 'stat/sonoff_ed1/POWER3', ... (3 bytes))
stat/sonoff_ed1/POWER3 OFF
```

Relays 9 to 10 gives additional a domoticz message as follows:

```
Client mosqsub|4233-vdr received PUBLISH (d0, q0, r0, m0, 'domoticz/in', ... (58 bytes))
domoticz/in {"idx":1950,"nvalue":1,"svalue":"","Battery":89,"RSSI":10}
Client mosqsub|4233-vdr received PUBLISH (d0, q0, r0, m0, 'stat/sonoff_ed1/RESULT', ... (16 bytes))
stat/sonoff_ed1/RESULT {"POWER10":"ON"}
Client mosqsub|4233-vdr received PUBLISH (d0, q0, r0, m0, 'stat/sonoff_ed1/POWER10', ... (2 bytes))
stat/sonoff_ed1/POWER10 ON
```
Is this a feature or a bug?

stefanbode, 2018-06-21:
Hm, sound not like a cool feature. therefore i would vote for a bug. 
Let me guess you have defined 2 RELAYS and the 8 RELAYS on the PCF. Therefore 1 and 2 should be normal and 3-10 should be executed by the PCF functions.

stefanbode, 2018-06-22:
Ok, updated some functionality. The new behavior is now that the GPIO relays are first in the list. the the PCF relays show up. This has not changed so far. In the Standard configuration, 4 devices are defined for domoticz. The first 4 Relays are assignable. Changing this setting should allow to enhance it to more relays:

`#define MAX_DOMOTICZ_IDX       4            // Max number of Domoticz device, key and switch indices`

Karl-opec, 2018-06-22:
There is still somezhing strange:
```
00:00:00 RSLT: pcf8574 1 boards
00:00:00 RSLT: Final max devices: 2, PCF8574 devices 0
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_338C76, GroupTopic sonoffs) Version 6.0.0a-2_4_1
00:00:00 WIF: Connecting to AP1 giot12 in mode 11N as sonoff-3190...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff-3190.local with IP address 192.168.40.50
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff/POWER = 
00:00:07 MQT: tele/sonoff/INFO1 = {"Module":"ElectroDragon","Version":"6.0.0a","FallbackTopic":"DVES_338C76","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-3190","IPAddress":"192.168.40.50"}
00:00:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
00:00:08 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
00:00:08 MQT: stat/sonoff/POWER1 = OFF
00:00:08 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
00:00:08 MQT: stat/sonoff/POWER2 = OFF
```
As you can see PCF is detected but the 8 relays don`t show in the webui.

stefanbode, 2018-06-22:
please goto CONFIGURATION -> Configure PCF8574 and set the Relays to OUTPUT. Done

Karl-opec, 2018-06-24:
Thanks it works now. It was my fault. I forgot that I had to do the configuration from scratch,
because of compilation for 4M flash to be able to do OTA agin.