stefanbode/Sonoff-Tasmota #39: deepsleep command not exists

roblad, 2018-06-19:
Hi,

After upgrade deepsleep is lost

22:26:29 MQT: stat/zewnetrzne/RESULT = {"Command":"Unknown"}
22:26:40 CMD: deepsleep 1200
22:26:40 MQT: stat/zewnetrzne/RESULT = {"Command":"Unknown"}

stefanbode, 2018-06-20:
oh yes, found already yesterday too. I have no idea why, but now fixed it in the way it should be with all the other commands. See my latest checkin from this morning. Also take your chirp code. Like!!

roblad, 2018-06-20:
Hi, is it possible to add config parameter to set deepsleep on or off for i.e. switching it off permamently?

stefanbode, 2018-06-20:
Hi Rob, one way is to use one of the GPIO ports for this. I copied the behavior of EPSEASY here. On the other side, you can set it to 0 which is equal to set it off.

closed 2018-07-04 by stefanbode