stefanbode/Sonoff-Tasmota #40: ESP-01: No D0 available, D4 (GPIO2) not working?

Drachenkaetzchen, 2018-06-21:
Hi,

I tried to use this fork and I struggled with getting DeepSleep on an ESP-01 to work.

Things I've done:
- Flashed the firmware with the usual setup
- Connected RST to D4 (GPIO2) and configured it in the Web Interface
- Set `deepsleep 60`

The device enters deep sleep fine. However, when attempting to wakeup, I see the blue LED blinking (AFAIR it's Serial TX) in very short intervals. It looks like the device tries to wakeup and gets to sleep immediately.

Can you give me a bit of info on how the DeepSleep pin is supposed to work? Is it floating during non-deepsleep operation and does it get pulled up or down for waking up?

Drachenkaetzchen, 2018-06-21:
I believe I have figured out the root cause. GPIO16 has a pull down resistor, which GPIO2 does not. So it seems that it's required to hard-wire GPIO16 to the reset pin directly on the board, unless we find some other way with an external circuit.

Drachenkaetzchen, 2018-06-21:
As I don't want to solder stuff to my ESP-01 module: I assume that the default state of GPIO16 is floating to allow the module to pull RESET high.

Once wakeup is triggered, the device should trigger the RESET by pulling GPIO16 low, which triggers the reset and GPIO16 is in floating again. Is that assumption correct so far?

stefanbode, 2018-06-21:
The deepsleep on the esp8266 is hardwired to send the wakeup signal on D0 GPIO16. You cannot trick around this. Why do you have no access to it ? Just connect an additional wire with a small 10k resistor to it. But I thin RST and GPIO16 can also directly wired without a resistor

stefanbode, 2018-06-21:
In my Firmware you can define a deepsleep GPIO, but this is just for enable/disable with a switch that the device go into deepsleep. It has nothing to do with the wake up.

Drachenkaetzchen, 2018-06-21:
> Why do you have no access to it ?

The ESP-01 modules do not break out GPIO16. If this was an one-off project I'd solder the wires, but I plan on deploying 10 ESP-01 modules (which I actually have in stock, so I don't intend to buy other ones) so modding each of those is quite some task.

I think this issue can be closed, but I'd appreciate a link from the repo's README to this issue to help others having a similar issue.

I think I'll play around with it, maybe an external 555 software triggered reset circuit would work. Just prior the regular deepsleep, the firmware could send a signal on one of the regular GPIOs to start the timer, which then triggers the reset. It's a workaround, but I feel it's better than to solder wires to 10 ESP-01 modules ;)

ajanulis, 2018-06-21:
http://www.instructables.com/id/Enable-DeepSleep-on-an-ESP8266-01/

Drachenkaetzchen, 2018-06-21:
I added an article on hackaday to the initial comment with a tl;dr regarding deep sleep on ESP-01 modules.

closed 2018-06-21 by Drachenkaetzchen