stefanbode/Sonoff-Tasmota #42: MCP23008/MCP23017 Support

andrethomas, 2018-06-22:
Hi

Are you planning on adding support for MCP23008/MCP23017 to this?

Thanks

stefanbode, 2018-06-22:
Hi, I took a small look on this chip and for me it looks like there is no big deal to support this similar to the PCF one. I still have the open task to support also the digital input use-case. Do you really need this? If yes I would order some chips to get them tested.

andrethomas, 2018-06-22:
It's not urgent - I have the chips and can probably use your pcf code as a template to add support for the MCP once you're done with handling the input side.

stefanbode, 2018-06-23:
Because the devices nearly do the same it could be much more straight forward to integrate instead of duplicate the code. I will order some devices and work on the input side. I have still the requirements to detect the valve end position.

stefanbode, 2018-07-08:
I have seen you pull request at the original code. As soon as my hardware arrives I will give it a try we’ll done and interesting chipset.

andrethomas, 2018-07-08:
Cool, I found the MCP's to be more useful for inputs than the PCF's most specifically because of the interrupt polling that may be implemented through I2C whereas the PCF-8574 relies on using an external interrupt pin which will in some way need to be connected to the ESP8266 and I cannot see that happening in the main build for now since most of the Sonoff devices have limited input pins.

Either way, the interrupts are a cool way of triggering MQTT messages which I found useful for my application :)

andrethomas, 2018-07-11:
Doesn't look like its gonna get committed into the main development - Guess I'll have to do what you're doing... just a pain to keep the fork updated all the time though.. which method do you use?

stefanbode, 2018-07-11:
I can offer you to integrate this into my branch, then you don't have the additional effort. But I have to take a look into it first. My hardware still not arrived for testing. 

Merging is a pain. That's the reason why I tag all my changes with //STB MOD  //END. This makes is simple for me to check if I just merge the main branch or have to take a closer look at how the changes affect me. Anyhow it is still quite work. Especially when something general changes.

I do the merge with "beyond compare" simple easy and I still own a license. :-)

andrethomas, 2018-07-11:
beyond compare looks pretty cool for $60

andrethomas, 2018-07-18:
MCP23008/MCP23017 for input was merged to arendst:development

stefanbode, 2018-07-21:
This is good news. As soon as arend enabled something inter master like the HR04 I remove it immediate from my branch. Want to keep the differences as minimal as possible. It should be a mod of the original firmware and not a competitive side development.

closed 2018-07-21 by andrethomas