stefanbode/Sonoff-Tasmota #44: Module Configuration

Karl-opec, 2018-07-04:
Despite having selected:

```
#define USE_SERIAL_BRIDGE                        // Add support for software Serial Bridge (+0k8 code)
#define USE_SDM120                               // Add support for Eastron SDM120-Modbus energy meter (+1k7 code)
  #define SDM120_SPEED         9600              // SDM120-Modbus RS485 serial speed (default: 2400 baud)
#define USE_SDM630                               // Add support for Eastron SDM630-Modbus energy meter (+2k code)
  #define SDM630_SPEED         9600              // SDM630-Modbus RS485 serial speed (default: 9600 baud)
```

these module config options don't show up in the config menu.  It show Option Nr. 72,73,74,75 without
description

stefanbode, 2018-07-04:
Hi Karl, I did not make a change hopefully in this section. Therefore it should behave like the original software. Please, can you confirm that it works with the original software and then attach two pictures to show the differences? I will try to fix it asap.

Karl-opec, 2018-07-04:
Sorry the drop down menu disappears when i try to make a screenshot.
But original software shows:

71 SerBr TX
72 SerBr Rx
75 SDM120 TX
76 SDM120 RX
77 SDM630 TX
78 SDM630 RX

Yours shows:

71 DeepSleep Switch
72 
73
74
75

Hope you can find the reason!

It might get mixed up by your introduced entry "DeepSleep Switch"?

stefanbode, 2018-07-04:
Upgraded to c version. Please be aware that I had to change (hopefully finally) the setoption parameters for interlock and the shutter. New ones are 81 and 80

closed 2018-07-04 by stefanbode

Karl-opec, 2018-07-04:
That fine, i will be patient. But what are above mentioned
defines for in your software?

stefanbode, 2018-07-04:
Please see the wiki. [](https://github.com/stefanbode/Sonoff-Tasmota/wiki/Shutter-support-for-ONE-shutter-blinds-working-with-two-relays)  There is now support for blinds and shutters. This is a bit tricky because you need to deal with two relays in a smart way and need to do all calculation inside the firmware to ensure proper operation.