stefanbode/Sonoff-Tasmota #45: Compile Errors

B1G1, 2018-07-04:
Hi Stefan,
I'm trying to compile with Arduino-IDE but I get this error:

> In file included from /Users/marcobucari/Downloads/Sonoff-Tasmota-master/sonoff/sonoff.ino:33:0:
> sketch/user_config.h:354:34: fatal error: user_config_override.h: No such file or directory
>  #include <user_config_override.h>
>                                   ^
> compilation terminated.
> exit status 1

if I comment this line #include <user_config_override.h> at the end of user_config.h
then I get this errors:

> sonoff:95: error: expected ',' or ';' before 'D_CMND_INTERLOCKMASK'
>    D_CMND_GPIO "|" D_CMND_GPIOS "|" D_CMND_PWM "|" D_CMND_PWMFREQUENCY "|" D_CMND_PWMRANGE "|" D_CMND_COUNTER "|"  D_CMND_COUNTERTYPE "|"   D_CMND_COUNTERDEVIDER "|" D_CMND_MQTTENABLE "|" D_CMND_DEEPSLEEP "|" D_CMND_INTERLOCKMASK "|"
>                                                                                                                                                                                                                  ^
> xdrv_02_webserver:187: error: expected ',' or ';' before 'D_CLOSE'
>    "<div><span class='p'>" D_CLOSE "</span><span class='q'>" D_OPEN "</span></div>"
>                            ^
> In file included from /Users/marcobucari/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/Arduino.h:241:0,
>                  from sketch/sonoff.ino.cpp:1:
> /Users/marcobucari/Downloads/Sonoff-Tasmota-master/sonoff/xdrv_02_webserver.ino: In function 'void HandleAjaxStatusRefresh()':
> xdrv_02_webserver:642: error: 'D_CMND_POSITION' was not declared in this scope
>      snprintf_P(svalue, sizeof(svalue), PSTR(D_CMND_POSITION " %s"), tmp);
>                                              ^
> /Users/marcobucari/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/pgmspace.h:16:68: note: in definition of macro 'PSTR'
>  #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
>                                                                     ^
> xdrv_02_webserver:642: error: expected ')' before string constant
>      snprintf_P(svalue, sizeof(svalue), PSTR(D_CMND_POSITION " %s"), tmp);
>                                                              ^
> /Users/marcobucari/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/pgmspace.h:16:68: note: in definition of macro 'PSTR'
>  #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
>                                                                     ^
> /Users/marcobucari/Downloads/Sonoff-Tasmota-master/sonoff/xdrv_97_shutter.ino: At global scope:
> xdrv_97_shutter:23: error: 'D_CMND_OPEN' was not declared in this scope
>    D_CMND_OPEN "|" D_CMND_CLOSE "|" D_CMND_STOP "|" D_CMND_POSITION  "|" D_CMND_OPENTIME "|" D_CMND_CLOSETIME "|" D_CMND_SHUTTERRELAY;
>    ^
> /Users/marcobucari/Downloads/Sonoff-Tasmota-master/sonoff/xdrv_97_shutter.ino: In function 'boolean Xdrv97(byte)':
> xdrv_97_shutter:165: error: 'D_SHUTTER' was not declared in this scope
>          snprintf_P(mqtt_data, sizeof(mqtt_data), JSON_SHUTTER_POS, mqtt_data, D_SHUTTER, Shutter_Real_Position * 100 / Shutter_Open_Max);
>                                                                                ^
> Più di una libreria trovata per "TasmotaSerial.h"
> Usata: /Users/marcobucari/Documents/Arduino/libraries/TasmotaSerial-2.0.0
> Non usata: /Users/marcobucari/Documents/Arduino/libraries/TasmotaSerial-1.1.0
> exit status 1
> expected ',' or ';' before 'D_CMND_INTERLOCKMASK'

what am I doing wrong?

stefanbode, 2018-07-04:
Can you please try to compile it with the EN settings or use the newest build. I assume that you use italy or spain and that there are some definitions missing in this language. I fixed this a few hours ago.

B1G1, 2018-07-04:
Yes, I can compile with the EN language, but I still need to comment #include <user_config_override.h> in user_config.h

stefanbode, 2018-07-04:
Yes this is typical. Normally you put in the override all your setting like WLAN password ans so on. Therefore I was not expecting this to miss.

B1G1, 2018-07-04:
ok, i'v falshed a sonoff dual r2 but I can't set it up as for roller shutter use

> 18:44:37 CMD: setoption14 1
> 18:44:37 SRC: WebConsole from 192.168.1.23
> 18:44:37 RSL: Group 0, Index 14, Command SETOPTION, Data 1
> 18:44:37 MQT: stat/sonoff/RESULT = {"SetOption14":"ON"}
> 18:45:01 CMD: setoption81 1
> 18:45:01 SRC: WebConsole from 192.168.1.23
> 18:45:01 RSL: Group 0, Index 81, Command SETOPTION, Data 1
> 18:45:01 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}

what I miss?

stefanbode, 2018-07-04:
Setoption81 is introduced with the newest release 6.0.0c. Before it was 30 and 31. depending on your version. I would suggest to download the latest one because I had to release 30 because this is now used in the master branch of the software. Hope,this helps.

B1G1, 2018-07-05:
But your master is 6.0.0a... sorry i’m a little confused

stefanbode, 2018-07-05:
Hm Right. Looks like a sync issue. Only some files updated but for sure not all required. Need to do this as soon as I get access to AC power again. Maybe some hours

stefanbode, 2018-07-05:
Yep. Sync to github was incomplete. Please next try. Sorry for inconvenience

closed 2018-07-05 by B1G1

globalcommerce, 2018-07-17:
Hi, stefane!

I have setoption81 1 and I cannot run shutter commands