stefanbode/Sonoff-Tasmota #46: Momentary mode

B1G1, 2018-07-05:
Hi Stefan,
In my case I have momentary buttons, one opens and one closes the shutter. If I press the open button the shutter opens totally, If I want to open 50% I push open button and the push again when the shutter is 50% open to stop it. So I set pulsetime1 and pulsetime2 to 5, shutteropenduration 24 and shuttercloseduration 21, when I send shutterposition 50 command, relay 1 goes on and off correctly, but nothing else happens and the shutter opens totally.

Is it possible to make it work as needed for me?

stefanbode, 2018-07-07:
Can you give some more information about how it behaves is some situations. 

1. you stop in the middle and pulse again. does it revert direction or continue to close?
2. you let it run at the end. I assume it stops automatically. what happens if you pulse then. does it revert and open?

In all this is a completely different use case for the shutter and I have to think about if there is a smart way to integrate this.

B1G1, 2018-07-07:
This is the console log with some commands sent

> 13:40:10 HTP: Console
> 13:40:19 CMD: shutterposition
> 13:40:19 SRC: WebConsole from 192.168.1.23
> 13:40:19 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
> 13:40:19 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
> 13:40:19 MQT: stat/sonoff/RESULT = {"shutterposition":49}
> 13:40:25 CMD: shutterclose
> 13:40:25 SRC: WebConsole from 192.168.1.23
> 13:40:25 RSL: Group 0, Index 1, Command SHUTTERCLOSE, Data 
> 13:40:25 Shutter: Command Code: 'SHUTTERCLOSE', Payload: -99
> 13:40:25 SRC: 
> 13:40:25 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
> 13:40:25 MQT: stat/sonoff/POWER2 = ON
> 13:40:25 MQT: stat/sonoff/RESULT = {"shutterclose":1}
> 13:40:25 SRC: PulseTimer
> 13:40:25 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
> 13:40:25 MQT: stat/sonoff/POWER2 = OFF
> 13:40:35 NEW Shutter: Stopped Position 36
> 13:40:35 SRC: 
> 13:40:35 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
> 13:40:35 MQT: stat/sonoff/POWER2 = OFF
> 13:40:35 CFG: Saved to flash at F9, Count 68, Bytes 3812
> 13:40:46 CMD: shutterposition
> 13:40:46 SRC: WebConsole from 192.168.1.23
> 13:40:46 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
> 13:40:46 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
> 13:40:46 MQT: stat/sonoff/RESULT = {"shutterposition":0}
> 13:41:05 CMD: shutteropen
> 13:41:05 SRC: WebConsole from 192.168.1.23
> 13:41:05 RSL: Group 0, Index 1, Command SHUTTEROPEN, Data 
> 13:41:05 Shutter: Command Code: 'SHUTTEROPEN', Payload: -99
> 13:41:05 SRC: 
> 13:41:05 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 13:41:05 MQT: stat/sonoff/POWER1 = ON
> 13:41:05 MQT: stat/sonoff/RESULT = {"shutteropen":1}
> 13:41:06 SRC: PulseTimer
> 13:41:06 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 13:41:06 MQT: stat/sonoff/POWER1 = OFF
> 13:41:21 CMD: shutterposition
> 13:41:21 SRC: WebConsole from 192.168.1.23
> 13:41:21 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
> 13:41:21 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
> 13:41:21 MQT: stat/sonoff/RESULT = {"shutterposition":0}
> 13:41:29 NEW Shutter: Stopped Position 48036
> 13:41:29 SRC: 
> 13:41:29 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 13:41:29 MQT: stat/sonoff/POWER1 = OFF
> 13:41:29 CFG: Saved to flash at F8, Count 69, Bytes 3812
> 13:41:35 CMD: shutteropen
> 13:41:35 SRC: WebConsole from 192.168.1.23
> 13:41:35 RSL: Group 0, Index 1, Command SHUTTEROPEN, Data 
> 13:41:35 Shutter: Command Code: 'SHUTTEROPEN', Payload: -99
> 13:41:35 SRC: 
> 13:41:35 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 13:41:35 MQT: stat/sonoff/POWER1 = ON
> 13:41:35 MQT: stat/sonoff/RESULT = {"shutteropen":1}
> 13:41:35 NEW Shutter: Stopped Position 48136
> 13:41:35 SRC: 
> 13:41:35 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 13:41:35 MQT: stat/sonoff/POWER1 = OFF
> 13:42:05 CMD: shutterposition
> 13:42:05 SRC: WebConsole from 192.168.1.23
> 13:42:05 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
> 13:42:05 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
> 13:42:05 MQT: stat/sonoff/RESULT = {"shutterposition":100}
> 13:43:08 CMD: shutterposition 50
> 13:43:08 SRC: WebConsole from 192.168.1.23
> 13:43:08 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 50
[... 56 lines cut ...]
> 13:44:48 SRC: WebConsole from 192.168.1.23
> 13:44:48 RSL: Group 0, Index 1, Command SHUTTEROPEN, Data 
> 13:44:48 Shutter: Command Code: 'SHUTTEROPEN', Payload: -99
> 13:44:48 SRC: 
> 13:44:48 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
> 13:44:48 MQT: stat/sonoff/POWER1 = ON
> 13:44:48 MQT: stat/sonoff/RESULT = {"shutteropen":1}
> 13:44:49 SRC: PulseTimer
> 13:44:49 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 13:44:49 MQT: stat/sonoff/POWER1 = OFF
> 13:45:00 NEW Shutter: Stopped Position 48028
> 13:45:00 SRC: 
> 13:45:00 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
> 13:45:00 MQT: stat/sonoff/POWER1 = OFF
> 13:45:01 CFG: Saved to flash at F4, Count 73, Bytes 3812
> 13:45:01 CMD: shutterposition
> 13:45:01 SRC: WebConsole from 192.168.1.23
> 13:45:01 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
> 13:45:01 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
> 13:45:01 MQT: stat/sonoff/RESULT = {"shutterposition":100}

If you mean what it should do in that situations for my case then, since I have 2 momentary buttons:

1.  if I stop in the middle i have two choice, push open button so it opens or push close button so it closes. 
2. Yes, if I let it run at the end it stops automatically. Then it's the same as above.

stefanbode, 2018-07-07:
Ahh understand. You still have two relay one for open one for closing. But instead of keeping the relay on you need a pulse at start and one at stop. Ok that makes it a bit easier. Let me check.

B1G1, 2018-07-07:
Yes exactly

stefanbode, 2018-07-08:
Hi Bigi, please check if the newest firmware fulfills your needs. At the end, it was really easy to implement. Just one IF. 

To change the behavior from the normal shutter to the momentary shutter you must define a pulsetimer on BOTH relays that you want to use. Now the relay switches shortly on (PULSETIME) and then waits for the position, after the position is reached the same relay get an additional ON to stop everything.

If pulsetime is 0 it behaves normally with on off

Give it a try. Nice weekend.

B1G1, 2018-07-09:
Hi @stefanbode thank you very much for the great work.
I'm doing some tests and initially the shutter command made mistakes. for example shutterposition 60 from completely closed has completely opened the shutter. But repeating the test I could not reproduce the error.
I do not know why.

I don't need idle time calculation, and since the motor automatically stops when the shutter is fully open or closed, for shutteropen and shutterclose commands only the first pulse it would be enough. 
however it's not a problem

B1G1, 2018-07-09:
to see the current position I have to use shutterposition command with no arguments right?
if I move the shutter using the wall buttons the position is not updated

stefanbode, 2018-07-10:
Regarding your first error: This might be an initialization issue. Could happen if the digital position does not match physical position.
The time calculation is the default. Maybe not all have endstops. Should not make a problem, because using the same relay to stop.
you have two options to get the current digital shutter position. one is to send the command without a value. Correct guess. The second one is to read SHUTTER from the sensor reading that comes with TELEPERIOD. Setting TELEPERIOD=10 will give you the position every 10 seconds.

If you manually move the shutter everything is out of sync because there is no way currently to recognize this movement. 

SHUTTERSTOP is no problem. Will add it with the next upload.

B1G1, 2018-07-10:
ok, i already tried the two option to get the shutter position, but i found that SHUTTER from teleperiod is not always in sync with the response of shutterposition command. Is it possible?

About moving the shutter manually, i've connected my buttons to GPIO0 and GPIO9 as indicated in the Tasmota wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual-and-Dual-R2
so i think it should be possible to manage external button too.

I'll try shutterstop asap thanks ;)

stefanbode, 2018-07-10:
Ok, this makes sense with the buttons and should be possible to update. SHUTTERPOSITION and SHUTTER should report nearly the same value. At least if it is not moving anymore. 

I was expecting that you want to update the position with external switches. This is nearly impossible. but with switches connected to the ESP...

B1G1, 2018-07-10:
SHUTTERSTOP command already takes in consideration that in momentary mode there is a second pulse? Because otherwise after sending that command the second pulse moves again the shutter

stefanbode, 2018-07-10:
I tested it and for me the SHUTTERSTOP during movement only creates ONE pulse

stefanbode, 2018-07-10:
For the NON-momentary switches, my build is working with the up and down switches. With the momentary switches, I have the problem, that I could not differentiate if the relays are still ON because of the defined pulsetime, or if the relay is again ON because you hit the switch. It does not make sense to go to the switch as an input. This needs additional research. Currently you can start the shutter in the desired direction if it is in STOP mode, but you cannot stop it until it reaches the final position (fully open/close)

B1G1, 2018-07-10:
Ok, shutter stop works correctly.

Now since I use home assistant I'm trying to integrate the shutter on it.
With this configuration:
cover:

>  - platform: template
>    covers:
>      tapparella_studio:
>        friendly_name: "Tapparella Studio"
>        position_template: "{{ states.sensor.posizione_tapparella_studio.state | int }}"
>        open_cover:
>          service: mqtt.publish
>          data:
>            topic: 'cmnd/tapparellaStudio/Backlog'
>            payload: 'shutteropen'
>        close_cover:
>          service: mqtt.publish
>          data:
>            topic: 'cmnd/tapparellaStudio/Backlog'
>            payload: 'shutterclose'
>        stop_cover:
>          service: mqtt.publish
>          data:
>            topic: 'cmnd/tapparellaStudio/Backlog'
>            payload: 'shutterstop'
>        set_cover_position:
>          service: mqtt.publish
>          data_template:
>            topic: 'cmnd/tapparellaStudio/Backlog'
>            payload: 'shutterposition {{position}}' 

> - platform: mqtt
>   name: "Posizione tapparella studio"
>   state_topic: "tele/tapparellaStudio/SENSOR"
>   value_template: "{{ value_json.SHUTTER }}"
>   json_attributes:
>     - Time
>     - Switch1
>     - Switch2
>     - SHUTTER

all commands works but using the position slider the shutter stops not in the correct position.
and also on the console to open it 50% I have to send command shutter position 70 for example

stefanbode, 2018-07-11:
Hi Bigi,

new update to support switches 1 & 2 to control the relays manually. Without the momentary switches, this was easy, but the momentary switches are really a pain. A lot of new code required and quite much debug code currently. Please give it a try and give feedback. If something doeas not work please provide log files in debug mode.

Regarding your 50% 70% problem. I assume like you this is because of the holes that needs to be closed, too. This is operating time for the motors. There is no easy fix, because all homeautomation systems only support 0-100 for shutters. I have no idea, how to overcome the issue currently.

stefanbode, 2018-07-11:
From the home assistant definition I would make it simpler to avoid the Backlog translation in the firmware. This is not required. 

```
cover:
 - platform: template
   covers:
     tapparella_studio:
       friendly_name: "Tapparella Studio"
       position_template: "{{ states.sensor.posizione_tapparella_studio.state | int }}"
       open_cover:
         service: mqtt.publish
         data:
           topic: 'cmnd/tapparellaStudio/shutteropen'
       close_cover:
         service: mqtt.publish
         data:
           topic: 'cmnd/tapparellaStudio/shutterclose'
       stop_cover:
         service: mqtt.publish
         data:
           topic: 'cmnd/tapparellaStudio/shutterstop'
       set_cover_position:
         service: mqtt.publish
         data:
           topic: 'cmnd/tapparellaStudio/shutterposition'
           payload: {{'position}}'
```

B1G1, 2018-07-11:
thank you very much Stefan, i'll try all as soon as i go back home.

B1G1, 2018-07-11:
I flashed the latest version and using wall switches the position is updated, but not correctly.
Closing from position 100 to about half open, result in this
`14:52:16 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-11T14:52:16","Switch1":"ON","Switch2":"ON","SHUTTER":97} `

stefanbode, 2018-07-11:
Your switches must be pulse switches. you work with momentary motors. therefore two switches ON is not a good starting point. This might cause the issue. I uses the TOGGLE button for the relay on the web interface for testing and this works reliably. Also check, that the paired interlock is ON and PULETIME on both relays is e.g. 5. It does not work with 1 for sure. I tested it with 5 and it might work with 3

stefanbode, 2018-07-11:
Please see the updated WIKI section for shutters

B1G1, 2018-07-11:
uhm if I set 18 Button 2 on GPIO0 and 19 Button3 on GPIO9 my wall switches don't work, but yes using the web interface works

stefanbode, 2018-07-11:
ok then we have to define the switchmode. this is TASMOTA standard functionality and described there in the wiki. It depends if your shutter switches are push buttons, are ON/OFF switches then, if the Relay should toogle or follow or inverse follow. You have to define this. The ONLY limitation for my shutters is NEVER both switches ON. On my shutter in the kitchen this position is physically blocked. both OFF work but not both on. You have to switch one off before.

stefanbode, 2018-07-11:
If your switched where PUSHBUTTONS like the TOGGLE in the webinterface this would be the easiest

B1G1, 2018-07-11:
well... my switches are pushbuttons like this https://www.gewiss.com/content/dam/gwdam/product/GW/40/00/15/25/GW40001525/_jcr_content/renditions/original.svg
I think they are like yours... only one can be on

B1G1, 2018-07-11:
I found that 18 button2 on GPIO9 works for down, but nothing works on GPIO0.... I don't know why

stefanbode, 2018-07-11:
I found one "bug". If you have switch1 ON and then just switch over without any delay to switch2 on it does also not work on my side. it works, if you switch OFF first and then the other ON. Can you test, if you have the same problem

B1G1, 2018-07-11:
I can't do it on wall switch, if I release up button then it's off

stefanbode, 2018-07-11:
Ok, i wonder why you have GPIO9 but anyway. OK this is not the error. Do you get any log in the console if you hit the GPIO0 button. Please put logfile here or call me on skype stefan.bode

B1G1, 2018-07-11:
sorry, I had to leave.

I just configured my sonoff dual r2 like this https://github.com/arendst/arendst.github.io/raw/master/media/dualr2d.jpg

but now I have 18 button2 on GPIO9 and 19 button 3

log using the the webgui
```
18:40:57 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
18:40:57 MQT: stat/tapparellaStudio/POWER1 = ON
18:40:57 Start moving shutter...
18:40:57 Shutter is moving 3, power: 1, join: 1, direction: 1
18:40:57 Shutter: Real Position 41992, Target 48000, source: Shutter
18:40:57 SRC: PulseTimer
18:40:57 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
18:40:57 MQT: stat/tapparellaStudio/POWER1 = OFF
18:40:58 Shutter: Real Position 43992, Target 48000, source: PulseTimer
18:40:59 Shutter: Real Position 45992, Target 48000, source: PulseTimer
18:40:59 SRC: WebGui from 192.168.1.23
18:40:59 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
18:40:59 MQT: stat/tapparellaStudio/POWER1 = ON
18:40:59 Switch OFF motors. Target: 46292, source: 0
18:40:59 NEW Shutter: Stopped Position 46292, relay: 0, pulsetimer: 5
18:40:59 CFG: Saved to flash at F6, Count 104, Bytes 3812
18:41:00 SRC: PulseTimer
18:41:00 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
18:41:00 MQT: stat/tapparellaStudio/POWER1 = OFF
18:41:04 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-11T18:41:04","SHUTTER":96}
18:41:16 SRC: WebGui from 192.168.1.23
18:41:16 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
18:41:16 MQT: stat/tapparellaStudio/POWER2 = ON
18:41:16 Start moving shutter...
18:41:16 Shutter is moving 3, power: 2, join: 2, direction: -1
18:41:16 Shutter: Real Position 45950, Target 0, source: Shutter
18:41:16 SRC: PulseTimer
18:41:16 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
18:41:16 MQT: stat/tapparellaStudio/POWER2 = OFF
18:41:17 Shutter: Real Position 43670, Target 0, source: PulseTimer
18:41:18 Shutter: Real Position 41390, Target 0, source: PulseTimer
18:41:19 SRC: WebGui from 192.168.1.23
18:41:19 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
18:41:19 MQT: stat/tapparellaStudio/POWER2 = ON
18:41:19 Switch OFF motors. Target: 40022, source: 0
18:41:19 NEW Shutter: Stopped Position 40022, relay: 1, pulsetimer: 5
18:41:19 SRC: PulseTimer
18:41:19 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
18:41:19 MQT: stat/tapparellaStudio/POWER2 = OFF
18:41:19 CFG: Saved to flash at F5, Count 105, Bytes 3812
```

from the wall buttons no logs if I push up, while this if I push down

```
18:43:44 APP: Button2 immediate
18:43:44 SRC: Button
18:43:44 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
18:43:44 MQT: stat/tapparellaStudio/POWER2 = ON
18:43:44 Start moving shutter...
18:43:44 Shutter is moving 3, power: 2, join: 2, direction: -1
18:43:45 SRC: PulseTimer
18:43:45 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
18:43:45 MQT: stat/tapparellaStudio/POWER2 = OFF
18:43:45 Shutter: Real Position 38426, Target 0, source: PulseTimer
18:43:46 Shutter: Real Position 36146, Target 0, source: PulseTimer
18:43:47 APP: Button2 immediate
18:43:47 SRC: Button
18:43:47 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
18:43:47 MQT: stat/tapparellaStudio/POWER2 = ON
18:43:47 Switch OFF motors. Target: 35234, source: 3
18:43:47 NEW Shutter: Stopped Position 35234, relay: 1, pulsetimer: 5
18:43:47 CFG: Saved to flash at F4, Count 106, Bytes 3812
18:43:47 SRC: PulseTimer
18:43:47 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
18:43:47 MQT: stat/tapparellaStudio/POWER2 = OFF
```

B1G1, 2018-07-11:
Solved it!
I had to set SwitchMode1 and SwitchMode2 to 4, that is inverted PUSHBUTTON (Normally 0, 1 = toggle) leaving this configuration https://github.com/arendst/arendst.github.io/raw/master/media/dualr2d.jpg

Now the position is updated correctly.
That's fantastic!

stefanbode, 2018-07-12:
Hi Bigi, one question. How long after the stop the move happens. Less than one second? For our 50% 70% movement problem I have a simple and great idea. Please be patient. The log is to short. I need the information above to see why the power ON was initiated by the shutter. Especially which of the ‚if‘ where responsible.

B1G1, 2018-07-12:
Hi, the movement happens after some seconds, for sure not less than one.

Wow can’t wait to know your idea! :-D

stefanbode, 2018-07-12:
Ok, this is really strange behavior haven't seen on my testbench. Please provide a detailed log from the moment you pushed the stop bottom. Maybe there is something useful visible that causes the software to start the shutter again.

stefanbode, 2018-07-12:
Hi Bigi, please try with the new firmware following process.:

- Move manually to exactly 50% open of the shutter. 
- Then type in the console shutterposition and read the actual value. In your case eg. 60
- Close the shutter completely (shutterposition=0). Important!!
- use new command: shutter50percent=60 to define a new 50% opening.  
- restart = 1

Now the following should be: 0-5 % is used to open close the holes in the shutter. 5-100 percent should represent the real opening % of the shutter. The 5% is hard coded. To reset the complete magic you can set either shutter50percent to 50 or 0. Both revert back to default behavior.

B1G1, 2018-07-12:
hi Stefan
I'm testing the latest version with this cool feature.
after restart I send the command shutterposition 5 and the shutter opened totally

```
17:50:28 MQT: tele/tapparellaStudio/INFO3 = {"RestartReason":"Software/System restart"}
17:50:28 SRC: MQTT
17:50:28 RSL: Group 0, Index 1, Command BACKLOG, Data 33
17:50:28 SRC: MQTT
17:50:28 RSL: Group 0, Index 1, Command BACKLOG SHUTTERPOSITION, Data 52
17:50:28 Shutter: Command Code: 'BACKLOG SHUTTERPOSITION', Payload: 52
17:50:28 MQT: stat/tapparellaStudio/RESULT = {"Command":"Unknown"}
17:50:28 SRC: Backlog
17:50:28 RSL: Group 0, Index 33, Command , Data 
17:50:28 Shutter: Command Code: '', Payload: -99
17:50:28 MQT: stat/tapparellaStudio/RESULT = {"Command":"Unknown"}
17:50:29 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
17:50:29 MQT: stat/tapparellaStudio/POWER1 = OFF
17:50:29 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
17:50:29 MQT: stat/tapparellaStudio/POWER2 = OFF
17:50:36 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-12T17:50:36","Uptime":"0T00:00:15","Vcc":3.438,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14784}
17:50:36 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-12T17:50:36","Switch1":"OFF","Switch2":"OFF","SHUTTER":0}
17:50:42 CMD: shutterposition 5
17:50:42 SRC: WebConsole from 192.168.1.23
17:50:42 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 5
17:50:42 Shutter: Command Code: 'SHUTTERPOSITION', Payload: 5
17:50:42 SRC: Shutter
17:50:42 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
17:50:42 MQT: stat/tapparellaStudio/POWER1 = ON
17:50:42 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":5}
17:50:42 NEW Shutter: Stopped Position 100, relay: 0, pulsetimer: 5
17:50:42 SRC: Shutter
17:50:42 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
17:50:42 MQT: stat/tapparellaStudio/POWER1 = ON
17:50:42 SRC: PulseTimer
17:50:42 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
17:50:42 MQT: stat/tapparellaStudio/POWER1 = OFF
17:51:36 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-12T17:51:36","Uptime":"0T00:01:15","Vcc":3.439,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14760}
17:51:36 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-12T17:51:36","Switch1":"OFF","Switch2":"OFF","SHUTTER":0}
```

then after command shutteropen since the position was 0, I sended the shutterposition 50 and I can say the it works, the shutter stopped at 49 that is ok.
In the previous log there is at the begin a backlog command that I have always when the sonoff restarts, how can I delete that?

then this is a log where I was trying to stop at 50% to use the shutter50percet command, when I stopped at 50% after some seconds the shutter opened totally 

```
17:45:03 HTP: Main Menu
17:45:04 HTP: Configuration
17:45:05 HTP: Configure Logging
17:45:09 HTP: Configuration
17:45:11 HTP: Main Menu
17:45:12 HTP: Console
17:45:13 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-12T17:45:13","Uptime":"0T00:08:46","Vcc":3.439,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14480}
17:45:13 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-12T17:45:13","Switch1":"OFF","Switch2":"OFF","SHUTTER":100}
17:45:17 SRC: Switch
17:45:17 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
17:45:17 MQT: stat/tapparellaStudio/POWER2 = ON
17:45:17 Start moving shutter...
17:45:17 Shutter is moving 3, power: 2, join: 2, direction: -1
17:45:18 SRC: PulseTimer
17:45:18 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
17:45:18 MQT: stat/tapparellaStudio/POWER2 = OFF
17:45:18 Shutter: Real Position 46786, Target 0, source: PulseTimer
17:45:19 Shutter: Real Position 44506, Target 0, source: PulseTimer
17:45:20 Shutter: Real Position 42226, Target 0, source: PulseTimer
17:45:21 Shutter: Real Position 39946, Target 0, source: PulseTimer
17:45:22 Shutter: Real Position 37666, Target 0, source: PulseTimer
17:45:23 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-12T17:45:23","Uptime":"0T00:08:56","Vcc":3.439,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14480}
17:45:23 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-12T17:45:23","Switch1":"OFF","Switch2":"OFF","SHUTTER":73}
[... 132 lines cut ...]
17:46:26 RSL: Group 0, Index 1, Command TELEPERIOD, Data 60
17:46:26 MQT: stat/tapparellaStudio/RESULT = {"TelePeriod":"60"}
17:46:26 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-12T17:46:26","Uptime":"0T00:09:59","Vcc":3.438,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14408}
17:46:26 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-12T17:46:26","Switch1":"OFF","Switch2":"OFF","SHUTTER":57}
17:46:26 Shutter: Real Position 27600, Target 48000, source: PulseTimer
17:46:26 CFG: Saved to flash at F9, Count 155, Bytes 3812
17:46:27 Shutter: Real Position 29600, Target 48000, source: PulseTimer
17:46:28 Shutter: Real Position 31600, Target 48000, source: PulseTimer
17:46:29 Shutter: Real Position 33600, Target 48000, source: PulseTimer
17:46:30 Shutter: Real Position 35600, Target 48000, source: PulseTimer
17:46:31 Shutter: Real Position 37600, Target 48000, source: PulseTimer
[... cut ...]
17:46:35 Shutter: Real Position 45600, Target 48000, source: PulseTimer
17:46:36 Shutter: Real Position 47600, Target 48000, source: PulseTimer
17:46:36 NEW Shutter: Stopped Position 48000, relay: 0, pulsetimer: 5
17:46:36 SRC: Shutter
17:46:36 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
17:46:36 MQT: stat/tapparellaStudio/POWER1 = ON
17:46:37 CFG: Saved to flash at F8, Count 156, Bytes 3812
17:46:37 SRC: PulseTimer
17:46:37 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
17:46:37 MQT: stat/tapparellaStudio/POWER1 = OFF
```

B1G1, 2018-07-12:
SHUTTER from teleperiod is not synched with shutterposition

```
20:53:28 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-12T20:53:28","Switch1":"OFF","Switch2":"OFF","SHUTTER":100}
20:53:41 CMD: shutterposition 50
20:53:41 SRC: WebConsole from 192.168.1.23
20:53:41 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 50
20:53:41 Shutter: Command Code: 'SHUTTERPOSITION', Payload: 50
20:53:41 SRC: Shutter
20:53:41 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
20:53:41 MQT: stat/tapparellaStudio/POWER2 = ON
20:53:41 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":50}
20:53:41 SRC: PulseTimer
20:53:41 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
20:53:41 MQT: stat/tapparellaStudio/POWER2 = OFF
20:53:42 Shutter: Real Position 46555, Target 30250, source: PulseTimer
20:53:43 Shutter: Real Position 44275, Target 30250, source: PulseTimer
20:53:44 Shutter: Real Position 41995, Target 30250, source: PulseTimer
20:53:45 Shutter: Real Position 39715, Target 30250, source: PulseTimer
20:53:46 Shutter: Real Position 37435, Target 30250, source: PulseTimer
20:53:47 Shutter: Real Position 35155, Target 30250, source: PulseTimer
20:53:48 Shutter: Real Position 32875, Target 30250, source: PulseTimer
20:53:49 Shutter: Real Position 30595, Target 30250, source: PulseTimer
20:53:49 NEW Shutter: Stopped Position 30139, relay: 1, pulsetimer: 5
20:53:49 SRC: Shutter
20:53:49 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
20:53:49 MQT: stat/tapparellaStudio/POWER2 = ON
20:53:49 CFG: Saved to flash at F6, Count 174, Bytes 3812
20:53:49 SRC: PulseTimer
20:53:49 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
20:53:49 MQT: stat/tapparellaStudio/POWER2 = OFF
20:54:28 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-12T20:54:28","Uptime":"0T02:26:22","Vcc":3.438,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14376}
20:54:28 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-12T20:54:28","Switch1":"OFF","Switch2":"OFF","SHUTTER":62}
20:54:42 CMD: shutterposition
20:54:42 SRC: WebConsole from 192.168.1.23
20:54:42 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
20:54:42 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
20:54:42 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":49}
```

stefanbode, 2018-07-13:
Regarding the unexpected "continue". I can see in the logs that all Start/Stop are initiated by the SRC: Switch. This is a pulse coming from the Switch. I cannot do anything against it. It looks like the is some bouncing in your switches. at 17:46:10 and the next 3 seconds there are 4!! pressed at the switches.

stefanbode, 2018-07-13:
Regarding the wrong position in teleperiod: Yes this was a bug. Fixed and committed to the code.

B1G1, 2018-07-13:
the 4 switch presses are correct, it was me trying to stop exactly at 50% :)

I do not understand what you mean with mixed OPEN and CLOSE.

B1G1, 2018-07-13:
the pulse that opened totally after the 4 switch presses starts here 17:46:36 SRC: Shutter

B1G1, 2018-07-13:
stefan can I use your firmware with a sonoff 4ch that I use to control 2 shutters?

stefanbode, 2018-07-13:
I'm pretty sure that this should work. There is no special stuff inside, that should prevent the usage.

stefanbode, 2018-07-13:
Regarding mixed relays. You wrote "command shutterposition 5 and the shutter opened totally". This is more than strange. 0 = CLOSE, 100 = OPEN

stefanbode, 2018-07-13:
17:46:13 SRC: Switch
You tapped Switch1 to fulle OPEN the shutter. Current position is nearly closed
17:46:13 Shutter: Real Position 1600, Target 48000, source: Shutter
17:46:26 CMD: teleperiod 60
The shutter is still moving up. you change the teleperiod
17:46:36 NEW Shutter: Stopped Position 48000, relay: 0, pulsetimer: 5
Shutter reached final position fully open and set the STOP command

All fine from my side.

B1G1, 2018-07-13:
yes it's very strange, since i sent shutterposition 5 immediately after sonoff restart i think something else opened the shutter... that backlog i have at the beginning of the log for example is strange.

Ok so it seems there is a pulse from the switch that fully opened the shutter after i stop it at 50%... other thing very strange....

stefanbode, 2018-07-13:
I have added a new calibration command: SHUTTERSETCLOSE. Please execute this, if the shutter is completely closed. This will ensure all variables have the right init values.

stefanbode, 2018-07-13:
any MQTT messages that are saved persistently can be deleted just by sending it again WITHOUT payload, but with the persistent flag. I use MQTTSPY for my MQTT hardcore changes.

B1G1, 2018-07-13:
error for shuttersetclose
```
13:32:30 CMD: shuttersetclose
13:32:30 SRC: WebConsole from 192.168.1.23
13:32:30 RSL: Group 0, Index 1, Command SHUTTERSETCLOSE, Data 
13:32:30 Shutter: Command Code: 'SHUTTERSETCLOSE', Payload: -99
13:32:30 MQT: stat/tapparellaStudio/RESULT = {"Command":"Error"}
```

stefanbode, 2018-07-13:
Ok, it’s working anyway. Shutterposition will show 0 afterwards

B1G1, 2018-07-13:
uhm shutterposition 50 opened the shutter totally

```
13:32:44 CMD: SHUTTERSETCLOSE
13:32:44 SRC: WebConsole from 192.168.1.23
13:32:44 RSL: Group 0, Index 1, Command SHUTTERSETCLOSE, Data 
13:32:44 Shutter: Command Code: 'SHUTTERSETCLOSE', Payload: -99
13:32:44 MQT: stat/tapparellaStudio/RESULT = {"Command":"Error"}
13:33:18 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-13T13:33:18","Uptime":"0T00:02:15","Vcc":3.440,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14648}
13:33:18 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T13:33:18","Switch1":"OFF","Switch2":"OFF","SHUTTER":0}
13:34:18 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-13T13:34:18","Uptime":"0T00:03:15","Vcc":3.440,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14648}
13:34:18 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T13:34:18","Switch1":"OFF","Switch2":"OFF","SHUTTER":0}
13:34:25 CMD: shutterposition
13:34:25 SRC: WebConsole from 192.168.1.23
13:34:25 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
13:34:25 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
13:34:25 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":0}
13:34:43 CMD: shutterposition 50
13:34:43 SRC: WebConsole from 192.168.1.23
13:34:43 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 50
13:34:43 Shutter: Command Code: 'SHUTTERPOSITION', Payload: 50
13:34:43 SRC: Shutter
13:34:43 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
13:34:43 MQT: stat/tapparellaStudio/POWER1 = ON
13:34:43 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":50}
13:34:43 NEW Shutter: Stopped Position 100, relay: 0, pulsetimer: 5
13:34:43 SRC: Shutter
13:34:43 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
13:34:43 MQT: stat/tapparellaStudio/POWER1 = ON
13:34:44 SRC: PulseTimer
13:34:44 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
13:34:44 MQT: stat/tapparellaStudio/POWER1 = OFF
13:35:18 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-13T13:35:18","Uptime":"0T00:04:15","Vcc":3.441,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14480}
13:35:18 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T13:35:18","Switch1":"OFF","Switch2":"OFF","SHUTTER":0}
```

stefanbode, 2018-07-13:
What is your openduration and what your closeduration? Looks like it Zink immediately it is open

B1G1, 2018-07-13:
```
13:45:22 CMD: shuttercloseduration
13:45:22 SRC: WebConsole from 192.168.1.23
13:45:22 RSL: Group 0, Index 1, Command SHUTTERCLOSEDURATION, Data 
13:45:22 Shutter: Command Code: 'SHUTTERCLOSEDURATION', Payload: -99
13:45:22 MQT: stat/tapparellaStudio/RESULT = {"shuttercloseduration":21}
13:45:28 CMD: shutteropenduration
13:45:28 SRC: WebConsole from 192.168.1.23
13:45:28 RSL: Group 0, Index 1, Command SHUTTEROPENDURATION, Data 
13:45:28 Shutter: Command Code: 'SHUTTEROPENDURATION', Payload: -99
13:45:28 MQT: stat/tapparellaStudio/RESULT = {"shutteropenduration":24}
```

stefanbode, 2018-07-13:
This line is very strange
13:34:43 NEW Shutter: Stopped Position 100, relay: 0, pulsetimer: 5

The stop position must be much higher. Here is like the log should look like.
```
14:41:26 CFG: Saved to flash at 3F7, Count 1467, Bytes 3812
14:41:32 CMD: shutterposition
14:41:32 SRC: Serial
14:41:32 RSL: Received Topic /shutterposition, Data Size 0, Data
14:41:32 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data
14:41:32 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
14:41:32 MQT: hm/status/sonoff/RESULT = {"shutterposition":0}
14:41:38 CMD: shutterposition 50
14:41:38 SRC: Serial
14:41:38 RSL: Received Topic /shutterposition, Data Size 2, Data 50
14:41:38 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 50
14:41:38 Shutter: Command Code: 'SHUTTERPOSITION', Payload: 50

14:41:38 SRC: Shutter
14:41:38 MQT: hm/status/sonoff/RESULT = {"POWER1":"ON"}
14:41:38 MQT: hm/status/sonoff/POWER1 = ON (retained)
14:41:38 MQT: hm/status/sonoff/RESULT = {"shutterposition":50}
14:41:38 SRC: PulseTimer
14:41:38 MQT: hm/status/sonoff/RESULT = {"POWER1":"OFF"}
14:41:38 MQT: hm/status/sonoff/POWER1 = OFF (retained)
14:41:39 Shutter: Real Position 1900, Target 54000, source: PulseTimer
14:41:40 Shutter: Real Position 3900, Target 54000, source: PulseTimer
14:41:40 WIF: Checking connection...
14:41:40 WIF: Connected
14:41:41 Shutter: Real Position 5900, Target 54000, source: PulseTimer
14:41:42 Shutter: Real Position 7900, Target 54000, source: PulseTimer
14:41:43 Shutter: Real Position 9900, Target 54000, source: PulseTimer
14:41:44 Shutter: Real Position 11900, Target 54000, source: PulseTimer
14:41:45 Shutter: Real Position 13900, Target 54000, source: PulseTimer
14:41:46 Shutter: Real Position 15900, Target 54000, source: PulseTimer
14:41:47 Shutter: Real Position 17900, Target 54000, source: PulseTimer

14:46:53 Shutter: Real Position 50552, Target 54000, source: PulseTimer
14:46:54 Shutter: Real Position 52552, Target 54000, source: PulseTimer
14:46:55 NEW Shutter: Stopped Position 54052, relay: 0, pulsetimer: 5
14:46:55 SRC: Shutter
14:46:55 MQT: hm/status/sonoff/RESULT = {"POWER1":"ON"}
14:46:55 MQT: hm/status/sonoff/POWER1 = ON (retained)
14:46:56 SRC: PulseTimer
14:46:56 MQT: hm/status/sonoff/RESULT = {"POWER1":"OFF"}
14:46:56 MQT: hm/status/sonoff/POWER1 = OFF (retained)
14:46:56 CFG: Saved to flash at 3FB, Count 1471, Bytes 3812
```

stefanbode, 2018-07-13:
In the beginning of the log there must be the init parameters of the shutter. changed to your parameters and still is working on my site. Here is how it should look like:

```
14:49:36 Shutter Pos Calculation graph: m1: 384, b1 9600, m2: 2304
14:49:36 ShutterInit: Position 28800 [50 %], Open Velocity: 100 Close Velocity: 114 , MAx Way: 48000, Opentime 24 [s], Closetime 21 [s]
```

B1G1, 2018-07-13:
I never see on my logs lines like those...

I don't know what is happening, I sent shutteropen, then shutterclose, and shutterposition 50 opened totally again but shutterposition is still 0 

```
13:48:15 CMD: shutteropen
13:48:15 SRC: WebConsole from 192.168.1.23
13:48:15 RSL: Group 0, Index 1, Command SHUTTEROPEN, Data 
13:48:15 Shutter: Command Code: 'SHUTTEROPEN', Payload: -99
13:48:15 SRC: Shutter
13:48:15 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
13:48:15 MQT: stat/tapparellaStudio/POWER1 = ON
13:48:15 MQT: stat/tapparellaStudio/RESULT = {"shutteropen":1}
13:48:15 SRC: PulseTimer
13:48:15 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
13:48:15 MQT: stat/tapparellaStudio/POWER1 = OFF
13:48:15 Shutter: Real Position 1100, Target 48000, source: PulseTimer
13:48:16 Shutter: Real Position 3100, Target 48000, source: PulseTimer
13:48:17 Shutter: Real Position 5100, Target 48000, source: PulseTimer
13:48:18 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-13T13:48:18","Uptime":"0T00:17:15","Vcc":3.441,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":94, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14480}
13:48:18 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T13:48:18","Switch1":"OFF","Switch2":"OFF","SHUTTER":2}
13:48:18 Shutter: Real Position 7100, Target 48000, source: PulseTimer
13:48:19 Shutter: Real Position 9100, Target 48000, source: PulseTimer
13:48:20 Shutter: Real Position 11100, Target 48000, source: PulseTimer
13:48:21 Shutter: Real Position 13100, Target 48000, source: PulseTimer
13:48:22 Shutter: Real Position 15100, Target 48000, source: PulseTimer
13:48:23 Shutter: Real Position 17100, Target 48000, source: PulseTimer
13:48:24 Shutter: Real Position 19100, Target 48000, source: PulseTimer
13:48:25 Shutter: Real Position 21100, Target 48000, source: PulseTimer
13:48:26 Shutter: Real Position 23100, Target 48000, source: PulseTimer
13:48:27 Shutter: Real Position 25100, Target 48000, source: PulseTimer
13:48:28 Shutter: Real Position 27100, Target 48000, source: PulseTimer
13:48:29 Shutter: Real Position 29100, Target 48000, source: PulseTimer
13:48:30 Shutter: Real Position 31100, Target 48000, source: PulseTimer
13:48:31 Shutter: Real Position 33100, Target 48000, source: PulseTimer
13:48:32 Shutter: Real Position 35100, Target 48000, source: PulseTimer
13:48:33 Shutter: Real Position 37100, Target 48000, source: PulseTimer
13:48:34 Shutter: Real Position 39100, Target 48000, source: PulseTimer
13:48:35 Shutter: Real Position 41100, Target 48000, source: PulseTimer
13:48:36 Shutter: Real Position 43100, Target 48000, source: PulseTimer
13:48:37 Shutter: Real Position 45100, Target 48000, source: PulseTimer
13:48:38 Shutter: Real Position 47100, Target 48000, source: PulseTimer
13:48:39 NEW Shutter: Stopped Position 48000, relay: 0, pulsetimer: 5
13:48:39 SRC: Shutter
13:48:39 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
13:48:39 MQT: stat/tapparellaStudio/POWER1 = ON
13:48:39 CFG: Saved to flash at F7, Count 183, Bytes 3812
13:48:39 SRC: PulseTimer
13:48:39 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
13:48:39 MQT: stat/tapparellaStudio/POWER1 = OFF
13:48:58 CMD: shutterposition
13:48:58 SRC: WebConsole from 192.168.1.23
13:48:58 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
13:48:58 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
13:48:58 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":100}
13:49:03 CMD: shutterclose
13:49:03 SRC: WebConsole from 192.168.1.23
13:49:03 RSL: Group 0, Index 1, Command SHUTTERCLOSE, Data 
13:49:03 Shutter: Command Code: 'SHUTTERCLOSE', Payload: -99
13:49:03 SRC: Shutter
13:49:03 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
13:49:03 MQT: stat/tapparellaStudio/POWER2 = ON
13:49:03 MQT: stat/tapparellaStudio/RESULT = {"shutterclose":1}
13:49:03 Shutter: Real Position 47430, Target 0, source: Shutter
13:49:04 SRC: PulseTimer
13:49:04 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
13:49:04 MQT: stat/tapparellaStudio/POWER2 = OFF
13:49:04 Shutter: Real Position 45150, Target 0, source: PulseTimer
13:49:05 Shutter: Real Position 42870, Target 0, source: PulseTimer
13:49:06 Shutter: Real Position 40590, Target 0, source: PulseTimer
[... 25 lines cut ...]
13:49:25 MQT: stat/tapparellaStudio/POWER2 = OFF
13:49:25 CFG: Saved to flash at F6, Count 184, Bytes 3812
13:49:30 CMD: shutterposition
13:49:30 SRC: WebConsole from 192.168.1.23
13:49:30 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
13:49:30 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
13:49:30 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":0}
13:49:36 CMD: shutterposition 50
13:49:36 SRC: WebConsole from 192.168.1.23
13:49:36 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 50
13:49:36 Shutter: Command Code: 'SHUTTERPOSITION', Payload: 50
13:49:36 SRC: Shutter
13:49:36 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
13:49:36 MQT: stat/tapparellaStudio/POWER1 = ON
13:49:36 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":50}
13:49:36 NEW Shutter: Stopped Position 106, relay: 0, pulsetimer: 5
13:49:36 SRC: Shutter
13:49:36 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"ON"}
13:49:36 MQT: stat/tapparellaStudio/POWER1 = ON
13:49:37 SRC: PulseTimer
13:49:37 MQT: stat/tapparellaStudio/RESULT = {"POWER1":"OFF"}
13:49:37 MQT: stat/tapparellaStudio/POWER1 = OFF
13:50:18 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-13T13:50:18","Uptime":"0T00:19:15","Vcc":3.442,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14480}
13:50:18 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T13:50:18","Switch1":"OFF","Switch2":"OFF","SHUTTER":0}
```

stefanbode, 2018-07-13:
Ok most is fine . What is the value of shutterset50percent. Should be 60 or something

B1G1, 2018-07-13:
```
13:58:30 CMD: SHUTTER50PERCENT
13:58:30 SRC: WebConsole from 192.168.1.23
13:58:30 RSL: Group 0, Index 1, Command SHUTTER50PERCENT, Data 
13:58:30 Shutter: Command Code: 'SHUTTER50PERCENT', Payload: -99
13:58:30 MQT: stat/tapparellaStudio/RESULT = {"shutter50percent":63}
```

B1G1, 2018-07-13:
after a new restart shutterposition 50 it seems to work, but I found other strange behaviors

1- shutterposition 5 is very close to shutterposition 10 and not with only holes open
2- shutterposition 50 from 50 closes the shutter, though the position is still 50 or 49 in this case.

```
14:32:11 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T14:32:11","Switch1":"OFF","Switch2":"OFF","SHUTTER":50}
14:32:39 CMD: shutterposition 50
14:32:39 SRC: WebConsole from 192.168.1.23
14:32:39 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 50
14:32:39 Shutter: Command Code: 'SHUTTERPOSITION', Payload: 50
14:32:39 SRC: Shutter
14:32:39 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
14:32:39 MQT: stat/tapparellaStudio/POWER2 = ON
14:32:39 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":50}
14:32:39 NEW Shutter: Stopped Position 30160, relay: 1, pulsetimer: 5
14:32:39 SRC: Shutter
14:32:39 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
14:32:39 MQT: stat/tapparellaStudio/POWER2 = ON
14:32:40 SRC: PulseTimer
14:32:40 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
14:32:40 MQT: stat/tapparellaStudio/POWER2 = OFF
14:32:40 CFG: Saved to flash at F5, Count 201, Bytes 3812
14:33:11 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-13T14:33:11","Uptime":"0T00:08:16","Vcc":3.440,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14592}
14:33:11 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T14:33:11","Switch1":"OFF","Switch2":"OFF","SHUTTER":49}
```

B1G1, 2018-07-13:
and from that state, I had to send 2 shutterclose commands to close totally with the resulting position of 12098464

```
14:41:36 CMD: shutterposition
14:41:36 SRC: WebConsole from 192.168.1.23
14:41:36 RSL: Group 0, Index 1, Command SHUTTERPOSITION, Data 
14:41:36 Shutter: Command Code: 'SHUTTERPOSITION', Payload: -99
14:41:36 MQT: stat/tapparellaStudio/RESULT = {"shutterposition":50}
14:41:41 CMD: shutterclose
14:41:41 SRC: WebConsole from 192.168.1.23
14:41:41 RSL: Group 0, Index 1, Command SHUTTERCLOSE, Data 
14:41:41 Shutter: Command Code: 'SHUTTERCLOSE', Payload: -99
14:41:41 SRC: Shutter
14:41:41 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
14:41:41 MQT: stat/tapparellaStudio/POWER2 = ON
14:41:41 MQT: stat/tapparellaStudio/RESULT = {"shutterclose":1}
14:41:42 SRC: PulseTimer
14:41:42 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
14:41:42 MQT: stat/tapparellaStudio/POWER2 = OFF
14:41:42 Shutter: Real Position 29120, Target 0, source: PulseTimer
14:41:43 Shutter: Real Position 26840, Target 0, source: PulseTimer
14:41:44 Shutter: Real Position 24560, Target 0, source: PulseTimer
14:41:45 Shutter: Real Position 22280, Target 0, source: PulseTimer
14:41:46 Shutter: Real Position 20000, Target 0, source: PulseTimer
14:41:47 Shutter: Real Position 17720, Target 0, source: PulseTimer
14:41:48 Shutter: Real Position 15440, Target 0, source: PulseTimer
14:41:49 Shutter: Real Position 13160, Target 0, source: PulseTimer
14:41:50 Shutter: Real Position 10880, Target 0, source: PulseTimer
14:41:51 Shutter: Real Position 8600, Target 0, source: PulseTimer
14:41:52 Shutter: Real Position 6320, Target 0, source: PulseTimer
14:41:53 Shutter: Real Position 4040, Target 0, source: PulseTimer
14:41:54 Shutter: Real Position 1760, Target 0, source: PulseTimer
14:41:55 NEW Shutter: Stopped Position 50, relay: 1, pulsetimer: 5
14:41:55 SRC: Shutter
14:41:55 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
14:41:55 MQT: stat/tapparellaStudio/POWER2 = ON
14:41:55 SRC: PulseTimer
14:41:55 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
14:41:55 MQT: stat/tapparellaStudio/POWER2 = OFF
14:41:55 CFG: Saved to flash at FB, Count 203, Bytes 3812
14:42:02 CMD: shutterclose
14:42:02 SRC: WebConsole from 192.168.1.23
14:42:02 RSL: Group 0, Index 1, Command SHUTTERCLOSE, Data 
14:42:02 Shutter: Command Code: 'SHUTTERCLOSE', Payload: -99
14:42:02 SRC: Shutter
14:42:02 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
14:42:02 MQT: stat/tapparellaStudio/POWER2 = ON
14:42:02 MQT: stat/tapparellaStudio/RESULT = {"shutterclose":1}
14:42:02 Shutter: Real Position 50, Target 0, source: Shutter
14:42:02 NEW Shutter: Stopped Position -64, relay: 1, pulsetimer: 5
14:42:02 SRC: Shutter
14:42:02 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"ON"}
14:42:02 MQT: stat/tapparellaStudio/POWER2 = ON
14:42:02 CFG: Saved to flash at FA, Count 204, Bytes 3812
14:42:02 SRC: PulseTimer
14:42:02 MQT: stat/tapparellaStudio/RESULT = {"POWER2":"OFF"}
14:42:02 MQT: stat/tapparellaStudio/POWER2 = OFF
14:42:11 MQT: tele/tapparellaStudio/STATE = {"Time":"2018-07-13T14:42:11","Uptime":"0T00:17:16","Vcc":3.438,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"FRITZ2.4", "RSSI":100, "APMac":"C8:0E:14:1E:C4:9C"}, "DeepSleep":0, "Heap":14592}
14:42:11 MQT: tele/tapparellaStudio/SENSOR = {"Time":"2018-07-13T14:42:11","Switch1":"OFF","Switch2":"OFF","SHUTTER":12098464}
```

stefanbode, 2018-07-13:
yes, there is still side effect with this F*** momentary switches. If there are additional commands inside the PULSETIME you're in trouble. If you set 50 and then 49 this will not work with momentary switches. I can make a stop for 50 50 to avoid starting the move, but it will not solve the general problem, that you make changes during PULSETIME is still running. Also, there is a minimum timeframe of 0.1 seconds that you need to respect for any action.

[68 more replies]

closed 2018-09-20 by stefanbode