stefanbode/Sonoff-Tasmota #47: Support for multible Shutters

ITTV-tools, 2018-07-15:
Hi

could you please add support for multible Shutters on one ESP ?
So that it's possible to use Sonoff 4ch odera NodeMCU Dev Board ?

Thanks

stefanbode, 2018-07-16:
Hi, yes it will work with all kind of ESP8266 like all the other functions. Regarding multiple shutters I will give it a try. I already know that there are sideeffects because not all user variables can be indexed and assigned to the correct shutter without making troumendous changes in the mastercode. Anyhow I assume that shutter with motors ON/OFF will work fine. Only the ones like Bigi uses with momentary motors that get their command through pulses are more than difficult.

stefanbode, 2018-07-16:
Updated up to 4 shutter support. Crazy stuff. You can give it a try. Please see the wiki how to enable it. Default is ONE shutter.

ITTV-tools, 2018-07-17:
Maybe I do something wrong, I dont get it compiled 

Error:
sketch\user_config.h:363:34: fatal error: user_config_override.h: No such file or directory
 #include <user_config_override.h>

But I see this files in the Arduino IDE

Do you have an idea ?
[image]

stefanbode, 2018-07-17:
You can comment this out if you don’t have your own configuration for compile

ITTV-tools, 2018-07-17:
Now Tasmota is working :) But when I enter the Command: SETOPTION80=1
I get:
00:02:33 CMD: SETOPTION80=1
00:02:33 MQT: stat/RolloESP01/RESULT = {"Command":"Unknown"}

stefanbode, 2018-07-17:
This is more than strange. Please check the information page if the update was really successful. Version and the Compiler time must be today evening

B1G1, 2018-07-17:
Setoption80 1
Not
Setoption80=1

ITTV-tools, 2018-07-17:
OMG 👍  Its working. Thanks ! I have set the SHUTTEROPENDURATION  and SHUTTERCLOSEDURATION  to 20 sec. But with a stopwatch I get 21 and something. Is that normal ?

stefanbode, 2018-07-18:
Thank you very much for this interesting observation. The Every50ms function from TASMOTA is based on the millis() function and this one gets halted during normal operation several times. E.g. to communicate with MQTT and other stuff. 

[millis](https://github.com/stefanbode/stefanbode.github.io/blob/master/images/nonlinear_millis.png)

This effect depends on several factors and is neither stable nor predictable. I started with a calibration, but because the process itself is not stable 2-3 sec per minute this does not solve the problem. Now there is a new implementation based on interrupts. This looks for me very accurate on all the time definitions.

ITTV-tools, 2018-07-18:
Now its on point. Something else, I noticed that when you press a button (like they are on a Sonoff 4CH) and its not the one that is on at this moment, the other one truns off (thats great) but the the one you pressed truns on for half a sec. I think thats not very healthy for the motor.

stefanbode, 2018-07-19:
Which mode did you use? With pulsetimer? The expected behavior is that the motor stop and go to the direction you indicated with the Switch. Please check if you have set SWITCHMODE to 4 or 3. The switch must be OFF not pressed and ON if pressed. If it operates the other way around what you see is likely. You have to set switchmode to all switches for the shutter.

ITTV-tools, 2018-07-19:
I used this:
Normal Operation (relay1 OFF/DOWN, relay2 OFF/UP), SETOPTION14 1, SETOPTION81 1, No PULSETIME
So without Pulsetimer. (I have not set one)

00:06:18 CMD: pulsetime
00:06:18 MQT: stat/RolloESP01/RESULT = {"PulseTime1":"0 (Active 0)"}

I set Switchmode 4. But it does still not change the direction. It just turns on for half a sec. I noticed the same if the shutter is closed for example. If you press the close button, it turns on for half a sec.

stefanbode, 2018-07-20:
you have to look at the STATUS messages. There you can see the status of the SWITCHES. These must be OFF in normal mode and be ON if you press them. As mentioned before, the behavior you describe match to a switch being ON and going OFF during a press. Please provide a detailed log file if you cannot fix it. I will have a look at what and why happens.

ITTV-tools, 2018-07-20:
22:49:36 MQT: tele/RolloESP01/STATE = {"Time":"2018-07-20T22:49:36","Uptime":"0T01:15:16","Vcc":3.556,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF", "Wifi":{"AP":1, "SSId":"Home", "RSSI":46, "APMac":"82:2A:A8:D4:79:6D"}, "DeepSleep":0, "Heap":15280}
22:49:36 MQT: tele/RolloESP01/SENSOR = {"Time":"2018-07-20T22:49:36","SHUTTER-0":32,"SHUTTER-1":0}
22:53:39 MQT: stat/RolloESP01/RESULT = {"POWER1":"ON"}
22:53:39 MQT: stat/RolloESP01/POWER1 = ON
22:53:40 Shutter 0: Real Pos: 13800, Target 40000, source: Shutter, pos %: 32, direction: 1, rtcshutter: 8
22:53:41 Shutter 0: Real Pos: 15900, Target 40000, source: Shutter, pos %: 32, direction: 1, rtcshutter: 29
22:53:42 Shutter 0: Real Pos: 17900, Target 40000, source: Shutter, pos %: 32, direction: 1, rtcshutter: 49
22:53:43 Shutter 0: Real Pos: 19900, Target 40000, source: Shutter, pos %: 32, direction: 1, rtcshutter: 69
22:53:44 Shutter 0: Real Pos: 21900, Target 40000, source: Shutter, pos %: 32, direction: 1, rtcshutter: 89
22:53:45 Shutter 0: Real Pos: 23900, Target 40000, source: Shutter, pos %: 32, direction: 1, rtcshutter: 109
22:53:46 Shutter 0: Real Pos: 25900, Target 40000, source: Shutter, pos %: 32, direction: 1, rtcshutter: 129
22:53:46 MQT: stat/RolloESP01/RESULT = {"POWER1":"OFF"}
22:53:46 MQT: stat/RolloESP01/POWER1 = OFF
22:53:46 MQT: stat/RolloESP01/RESULT = {"POWER2":"ON"}
22:53:46 MQT: stat/RolloESP01/POWER2 = ON
22:53:46 MQT: stat/RolloESP01/RESULT = {"POWER2":"OFF"}
22:53:46 MQT: stat/RolloESP01/POWER2 = OFF
22:53:47 MQT: stat/RolloESP01/RESULT = {"POWER1":"OFF"}
22:53:47 MQT: stat/RolloESP01/POWER1 = OFF

If you need more detail, I can send a video.

stefanbode, 2018-07-21:
In the beginning I see in the status line no information for switches. I expect here switch1 and switch2. Also there is nothing visible why at the end the relay goes on. Seriallog 5 may help to get more info. Please verify that in Configuration page you have assigned switch1 and 2 to a GPIO

stefanbode, 2018-07-21:
One addition that might explain something. Is the shutter is closed alread and you hit the close switch for sure the motor will start because this is Tasmota Standard functionality I cannot jump in. The buttons are always first executed before my code come into action. The only thing I can do is switching off the relay immediately. As long as it is not fully closed or fully open everything should be fine. If you send the commands through MQTT or something there should be no side effect. This should work 100%. Maybe a swichtoption that decouple the switch from the relay will solve this issue. But this is something i need to test very intensive.

ITTV-tools, 2018-07-28:
Is there a way to command the shutters without the seperate toptics ? Something like: cmnd/homeassistant/action Message: UP/DOWN/STOP

stefanbode, 2018-07-29:
You can use the standard command “backlog” and then send the different messages. How to use this is described in the Wiki of the original firmware

ITTV-tools, 2018-07-29:
is'nt the backlog commend for sending multible commands at once ? Maybe I did not express myself correctly.
In your example for Homeassistatant you're using 
cmnd/tapparellaStudio/shutteropen
cmnd/tapparellaStudio/shutterclose
cmnd/tapparellaStudio/shutterstop

But there is a MQTT Cover component in Home Assistant. Where you can only use one "command" topic for Up Down Stop. 
Like cmnd/sonoff/shutter1 Message: UP

Maybe im just blind, but what do I need to enter for shutter1

stefanbode, 2018-07-30:
cmnd/sonoff/shutter1/backlog Message: shutteropen

closed 2018-09-17 by stefanbode