stefanbode/Sonoff-Tasmota #48: Shutters - hardware switches and state ON/OFF.

pszafer, 2018-07-16:
Hello,
Great extension for sonoff!

I have question for you. Maybe nothing bad, but anyway I'd like to ask.

I have standard doorbell switches = momentary switches.
```
GPIO0 - Switch1
GPIO9 - Switch2
```
```
switchmode1 0
switchmode2 0
```
Everything is working very well most of the time.

But normal status is:
```
18:51:42 MQT: tele/roleta_balkon1/STATE = {"Time":"2018-07-16T18:51:42","Uptime":"0T00:01:38","Vcc":3.186,"POWER1":"OFF","POWER2":"OFF", "Wifi":{"AP":1, "SSId":"hass", "RSSI":82, "APMac":"02:9F:C2:74:3E:8C"}, "DeepSleep":0, "Heap":18208}
18:51:42 MQT: tele/roleta_balkon1/SENSOR = {"Time":"2018-07-16T18:51:42","Switch1":"ON","Switch2":"ON","SHUTTER-0":54}
```

when I click switch then switch1 or 2 is Off.

Is it suppose to be like that?

stefanbode, 2018-07-17:
yes, if you define a shutter also the switches are assigned to the shutter to control the shutter in parallel with the switches. If you want to use the switches with your doorbell you need to do two things. First assign Switch3 and 4 to GPIO0 and 9. Normally this should help because there is no relay 3 and 4. If you still have the behavior, please assign INTERLOCKMASK=1100. This will remove 3 and 4 from the interlock mode that normally is used to prevent short circuit on the motor.

stefanbode, 2018-07-17:
Just joking: If you leave it as it is, you should be able to control the shutter with the doorbell. Great fun

pszafer, 2018-07-17:
:+1: yes, it is working, I know. just thinking if it is proper state ON/OFF.

Thanks for this software!

closed 2018-07-17 by pszafer