stefanbode/Sonoff-Tasmota #50: Cannot load Shutter mode

globalcommerce, 2018-07-17:
Hi stefane!

I flashed my sonoff dual R2 setoption14 and 81 on but I cannot perform shutteropen and close commands.

Did I missed something?

stefanbode, 2018-07-18:
Yes, these are the people who do not read the Wiki, that takes hours to create and maintain. Until now you only enabled the interlock mode and changed it to paired interlock. Now setoption 80 will enable the shutter mode. There a quite a lot configuration aspects. wiki is essential.

globalcommerce, 2018-07-18:
Hi,

Yes I also set the option 80 to 1 at the same time, only here I think I had restart the sonoff.
I get back my office to do and feedback to you.

Thanks for your support

stefanbode, 2018-07-18:
As soon as option 80 is set you should be able to see the shutter slider in the webpage as shown in the wiki. Good luck.