stefanbode/Sonoff-Tasmota #52: SR04 and deepsleep dosn`t work

Tobbi35, 2018-07-26:
I have connected an SR04 distance meter to the GPIO TX and RX, if I activate the deepsleep it will also be executed, but the esp will not be activated anymore. Where can that be?

stefanbode, 2018-07-26:
TX and RY do nothing t work with SR04. This needs two GPIO ports. Deepsleep send Wakeup through gpio0 and must be physical connected to the rst pin. There is a lot of documentation at google How to do that.

Tobbi35, 2018-07-26:
maybe I've mispronounced something wrong, I have the SR04 on the GIO pins 01 and 03. The SR04 works too, but I give a deepsleep time so the ESP goes into sleep mode but does not wake up anymore. the RST pin is connected to GPIO0

closed 2018-08-19 by stefanbode