stefanbode/Sonoff-Tasmota #56: MCP230XX used the same way as PCF8574

MadDoct, 2018-08-07:
Could you add support for mcp23008 and mcp23017, the same way as PCF8574. I mean allowing them to be set as relays and not as sensors as in the current implementation...

stefanbode, 2018-08-19:
I just received some modules to test. Also have to see what’s already in the master branch. Anyhow quite busy these days. Time is gold .

welbo97, 2018-12-10:
Master added support and driver for mcp230xx series and implemented a sensor29 command to control it. I suggest you close this issue.

stefanbode, 2018-12-10:
Thanx!

closed 2018-12-10 by stefanbode

MadDoct, 2018-12-10:
This shouldn't be closed! It wasn't about implementing support for mcp23017. That has been done for a long time and I even helped implementing it... It was about implementing it seamlessly like with pcf8574, being able to configure it as relay or switch from tasmota...