stefanbode/Sonoff-Tasmota #58: NFC Reader and mqtt

Tobbi35, 2018-08-12:
Hello I would like to include an NFC reader via tasmota in mqtt. the assignment SDA and SDC are already present in the firmware, but unfortunately no data is transmitted in mqtt

stefanbode, 2018-08-13:
Hi Tobi, what kind of hardware did you use. To enable this we have to create a driver for the NFC that read and write the right bits on the I2C Bus. Then it is possible to transfer this through  MQTT. But I assume this will be very hardware specific. Additionally, I would not recommend using the tasmota for security purpose. There might be more problems than we already know.

stefanbode, 2018-08-13:
Hi Tobi. I ordered one and give him a try. In the meantime I will take a look how and what espeasy is doing. Copy good things is not a shame. Typically it takes 3weeks until the China electronic is here.

stefanbode, 2019-08-31:
Hi Tobi, received the device and will give him a try

stefanbode, 2019-09-01:
Already included in master firmware

closed 2019-09-01 by stefanbode