stefanbode/Sonoff-Tasmota #60: Shutterinvert needs additional adjustment

udo1toni, 2018-08-28:
I've setup a Sonoff T1 2CH as a rollershutter.

shutterposition works perfectly.
using `shutterinvert 1` results in upper position = 0, lower position = 100, perfect.

But now, there is something weird (`shutterinvert=1`):
When sending `shutteropen`, the shutter closes to position `0`
When sending `shutterclose`, the shutter opens ans stops immediately (returning `1` as position)
When sending `shutterclose` from a position other than fully open or fully closed, it moves the shutter up
When sending `shutterstop`, there is no reaction when the shutter is moving.
When sending `shutterstop` when the shutter is already stopped, the shutter moves to position `65`.
When sending `shutterstop` when the shutter is already stopped at position `65`, the shutter moves to position `35`.
When sending a `shutterinvert` without any parameter, tasmota reacts in sending a
`stat RESULT {"shutterinvert1":-99}` regardless, if `shutterinvert` is set to 0 or to 1.

stefanbode, 2018-08-28:
the last message hints at the root of the evil. I have seen this -99 stuff also but then it went away without any reason. In the log, in the beginning, you can see the definition if the shutter with all parameters. Shutterinvert must be 1. Otherwise, it will not work as requested. I will check where the -99 came from

stefanbode, 2018-08-28:
shutterinvert1 1
should do the job and also give a correct answer in the result. See code below.

```
  else if (CMND_SHUTTERINVERT == command_code && (index > 0) && (index <= shutters_present)) {
    if ( (XdrvMailbox.payload >= 0) && (XdrvMailbox.payload <= 1)) {
      Settings.shutter_invert[index-1] = XdrvMailbox.payload;
    }
    snprintf_P(mqtt_data, sizeof(mqtt_data), S_JSON_COMMAND_INDEX_NVALUE, command, index, XdrvMailbox.payload);
  }
```

udo1toni, 2018-08-28:
13:36:53 CMD: shutterinvert1 1
13:36:53 MQT: sonoff_t9/stat/RESULT = {"shutterinvert1":1}
13:37:09 CMD: shutterinvert1
13:37:09 MQT: sonoff_t9/stat/RESULT = {"shutterinvert1":-99}

stefanbode, 2018-08-28:
Ahh, no worries about this. The second one is a bug I just fixed right now. the shutterinvert is 1. Now back to your problem. Please post a logfile and tell we what you do. In the logfile open is 0 and close is 100. This will not change. Only in the UI and from the MQTT messages you send and receive just the opposite position. With shutterinvert not the internal calculation will change. Only the reporting changes.

stefanbode, 2018-08-28:
ok, let's work on getting it fixed. First, you HAVE to switch your relays. If on open close the direction is wrong you need to switch it.
shutteropen and shutterstop have no options like 50 or 75. They just open/close/stop. No matter where they are. shutterposition is the only command that is aware of invert and allows a value to define the position from 0-100. Let me know if more questions.

stefanbode, 2018-08-28:
Additionally if the position is 50 you are not able to set it to 49 or 51 because in most cases the operating time is to low. therefore the firmware just do not move. If you want to test change it at least 5 or better 10% up or down to have a minimum movement

stefanbode, 2018-08-28:
Hmm, shutterstop really seems to start the relay. also if it is already stopped. This seems to be a bug.

stefanbode, 2018-08-28:
Yep on the shutterstop there was an IF missing that takes care about the inverted functionality. I checked in a new version. Maybe you can give it a try.

udo1toni, 2018-08-28:
as far as I understood the wiki, I have to use channel 1 for UP and channel 2 for DOWN, so if switching POWER1 to ON, the shutter has to move up, and if switching the POWER2 to ON, the shutter has to move down. And this is what I see here. But when sending a shutteropen command, POWER2 is set to ON, so the Shutter moves down.

If changing shutterinvert to 0, the commands work as expected. So, the shutterinvert does a **complete** invert. If changing the "polarity" of the motor, this would result in shutter opening if switching POWER2 to ON, but also in a 100 if open and a 0 if closed... that's the same problem as without the shutterinvert option at all, in fact, I could handle the problem by using the wrong commands, but I think this is not intended... ;)

In question of shutterstop, this command is ok now :)

udo1toni, 2018-08-28:
AH. I think i found the issue... in `xdrv_97_shutter.ino` the code for open and close is
```
XdrvMailbox.payload = CMND_OPEN == command_code ? 100 : 0;
command_code = CMND_POSITION;
```
So, if sending a shutteropen, this will always result in a command `position 100`

stefanbode, 2018-08-28:
Ok let me do another deep dive. I assume you are right. It was not my idea to implement the invert. Anyhow should be fixable and I assume the error came in with the change of reporting shutterposition as a result.

stefanbode, 2018-08-28:
Yep. And another bug fixed. I reverted the last change and applied the current fix to it too. If I set the SOURCE to WEBGUI then the inverted will have no effect. This I'm using now for open/close/stop. Then these commands are indepedend from the invert and should work all the time current version checked in. Also upgraded to the latest TASMOTA version.

udo1toni, 2018-08-28:
Cool! Will have a try...

stefanbode, 2018-09-11:
Hi, I do some cleanup and fixed some side-effects when using a wall switch. now the shutter should stop and move in the opposite direction if you hit the switch in the other direction

udo1toni, 2018-09-11:
Did not see any problems with wall switches (I'm using a Sonoff T 2-gang), in fact, the last version works pretty good here, since last week... sorry for forgetting to close the issue!

closed 2018-09-11 by udo1toni