stefanbode/Sonoff-Tasmota #63: Mqtt and compiled *. Bin files

abaszczok, 2018-09-05:
I uploaded your bin file but despite many diffferent settings versions I Could not connect to my mqtt broker. Is there any Secret option enabled which is providing to connection failure? (info from sonoff console). I have many sonoffs working together with HA and default mqtt broker but yours is not.. Thx

abaszczok, 2018-09-10:
here is my log from PI

sonoff ip: is 192.168.8.111
any idea how to fix it??

Sep 05 04:43:59 raspberrypi hass[264]: 2018-09-05 04:43:59 INFO (MainThread) [hbmqtt.broker] Listener 'default': 2 connections acquired
Sep 05 04:43:59 raspberrypi hass[264]: 2018-09-05 04:43:59 INFO (MainThread) [hbmqtt.broker] Connection from 192.168.8.111:30225 on listener 'default'
Sep 05 04:44:03 raspberrypi hass[264]: 2018-09-05 04:44:03 ERROR (MainThread) [homeassistant.core] Timer got out of sync. Resetting
Sep 05 04:44:03 raspberrypi hass[264]: 2018-09-05 04:44:03 INFO (MainThread) [transitions.core] Exited state new
Sep 05 04:44:03 raspberrypi hass[264]: 2018-09-05 04:44:03 INFO (MainThread) [transitions.core] Entered state connected
Sep 05 04:44:03 raspberrypi hass[264]: 2018-09-05 04:44:03 WARNING (MainThread) [hbmqtt.mqtt.protocol.handler] BrokerProtocolHandler Unhandled exception in reader coro: IncompleteReadError('0 bytes read on a total of 1 expected bytes',)
Sep 05 04:44:03 raspberrypi hass[264]: 2018-09-05 04:44:03 INFO (MainThread) [transitions.core] Exited state connected
Sep 05 04:44:03 raspberrypi hass[264]: 2018-09-05 04:44:03 INFO (MainThread) [transitions.core] Entered state disconnected

abaszczok, 2018-09-10:
after aprox 200 retries (sonof is retrying every 10 seconds) device is successfully connected to MQTT brooker nad works normal

stefanbode, 2018-09-11:
Hi Adam,

I assume there is something wrong with your MQTT broker. This does not look like a firmware issue. Anything related to the firmware must be submitted with the log of the firmware. I'm sorry I cannot support all of the different home automation software out in the field like fhem, openHAB, domotircz, homeassistant and so on.

abaszczok, 2018-09-11:
Hi Stefan..
Ok understand.. Strange is that all other sonoff with tasmota on it are working without this issue. I will try to reflash it again.. Thx

stefanbode, 2018-09-11:
Maybe you have a timeout problem. I have a quite aggressive MQTT timeout of 1 second. this is hardcoded in lib/PubSubClient-2.6.09/src/PubSubClient.h Line 38:

`#define MQTT_SOCKET_TIMEOUT 1`

Maybe this causes your issue. My mqtt is fast and local. therefore no issues here. Anyhow the log you send was not from the device.

abaszczok, 2018-09-11:
I compared with tasmota 6.1.0.. In fact there is time out of 15seconds. So I will change it and compile again.. We will see

abaszczok, 2018-09-12:
Stefan.. 
Following your advise I installed Platfom.io-IDE. First i tried with ATOM layer but already on first step I have many different errors when downloading Platformio IDE..
i.e. 
"unable to get local issuer certificate"
So I couldnt test it...
Therfore I tried with VSCode layer...and installed it successfully..but.. when compiling your project ( I selected sononff by uncommenting line in platfomio.ini). 
I got following errors when building:
(part of log)
Compiling .pioenvs\sonoff\src\core_esp8266_timer.c.o
Compiling .pioenvs\sonoff\src\core_esp8266_wiring_digital.c.o
Compiling .pioenvs\sonoff\src\core_esp8266_wiring_pwm.c.o
Compiling .pioenvs\sonoff\src\sonoff.ino.cpp.o
xtensa-lx106-elf-g++: error: sonoff\sonoff.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.

when selecting [wemos] env
Compiling .pioenvs\wemos\lib113\NeoPixelBus-2.2.9\internal\HsbColor.cpp.o
In file included from C:\users\cztab\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include/osapi.h:29:0,
from C:\users\cztab\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h:11,
from C:\users\cztab\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:256,
from c:\users\cztab\appdata\local\temp\tmpifv4jb:1:
sonoff/user_config.h:407:34: fatal error: user_config_override.h: No such file or directory

Any ideas how to move one step forward?? This is why i asked you first time for *.bin file:(..
Just one more questinon - my PC is administrated by our IT dept + Symantec End point + proxy .. It seems that I have administrative rights but somehow limited because I can install software on my PC but I cant disable Symantec and other security options. I'm wondering if this issue is somehow related to administrative rights to create temporary files in some forbidden directories..

stefanbode, 2018-09-12:
Yes. I assume you have not your own user_config_override.h. Therefore create an empty file with the name in the sonoff directory where all the other sources are. Then compile wemos should work. If you want to put in your WIFI settings and other compiler options this can be done by #define and #ifdef in this file. There is a brief explanation about it in the Tasmota main Wiki

stefanbode, 2018-09-12:
This is looking like a virus scan issue. "Someone" is sitting on the file and block it. It will be temporarily created during compilation.

```
Compiling .pioenvs\sonoff\src\sonoff.ino.cpp.o
xtensa-lx106-elf-g++: error: sonoff\sonoff.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
```

closed 2018-09-17 by stefanbode