stefanbode/Sonoff-Tasmota #64: Introduce new interlock mode for testing

stefanbode, 2018-09-08:
I made the interlock more flexible. In the past it was interlock with all (setoption 14) or paired with the interlockmask to exclude relays. After some brain fuck of binary operation I introduced a new variable called BUCKETSIZE. This defines the amount of relays belong to the interlock. Default is 2.

B1G1, 2018-09-10:
Hi stefan,
this new interlock mode affects in some way the shutter mode?

stefanbode, 2018-09-10:
Hi Bigi, hopefully not. The shutter should use the BUCKTESIZE=2 and should behave as before. I tested quite a lot with the change. Please let me know, if you find a bug.

stefanbode, 2018-09-17:
I fixed a bug that uses the TASMOTO full interlock if switching OFF a relay that was already OFF

closed 2018-10-04 by stefanbode