stefanbode/Sonoff-Tasmota #65: big problem with sonoff 4ch pro

madmicio, 2018-09-14:
Hi!
I found a new bug in 4ch pro son mode.
you are active the relay 1 and before turning it off active the 3 or the 4 by slider. the two relays remain permanently on

stefanbode, 2018-09-14:
Yes already found this one too this afternoon after crashing my shutter a bit. The recognition through Alexa was not correct. But it should work without the Hue emulation. Anyhow maybe the same error somewhere else. R1 and r2 are normal and r3-r5 belong to the shutter?

madmicio, 2018-09-14:
sorry, I think I did not understand the question.
however, relay1 shutteropen1, relay2 shutterclose1, relay3 shutteropen2, relay4 shutterclose2

stefanbode, 2018-09-16:
Ok now I got it and this is quite serious. I will check this today and get it fixed soon. I thought all variables for the schütter are already separated, but I have an idea which one is missing. The real-time clock. Could get complicated.

closed 2018-10-04 by stefanbode