stefanbode/Sonoff-Tasmota #66: problem with shutterstop

turin400, 2018-09-17:
hello, I have a problem with a 4ch pro when I run via mqtt shutterstop1 or shutterstop2, power1 or power3 goes to off and power2 or power 4 make a blink. Because of this, the roller changes direction and stops.

08:32:22 MQT: stat/techo/RESULT = {"POWER1":"ON"}
08:32:22 MQT: stat/techo/POWER1 = ON
08:32:22 MQT: stat/techo/RESULT = {"shutteropen1":100}
08:32:23 Shutter 0: Real Pos: 15900, Target 90000, source: Shutter, pos %: 16, direction: 1, rtcshutter: 11
08:32:24 Shutter 0: Real Pos: 17900, Target 90000, source: Shutter, pos %: 16, direction: 1, rtcshutter: 31
08:32:25 Shutter 0: Real Pos: 19900, Target 90000, source: Shutter, pos %: 16, direction: 1, rtcshutter: 51
08:32:26 Shutter 0: Real Pos: 21900, Target 90000, source: Shutter, pos %: 16, direction: 1, rtcshutter: 71
08:32:27 Shutter 0: Real Pos: 23900, Target 90000, source: Shutter, pos %: 16, direction: 1, rtcshutter: 91
08:32:28 Shutter 0: Real Pos: 25900, Target 90000, source: Shutter, pos %: 16, direction: 1, rtcshutter: 111
08:32:29 Shutter 0: Real Pos: 27900, Target 90000, source: Shutter, pos %: 16, direction: 1, rtcshutter: 131
08:32:29 MQT: stat/techo/RESULT = {"POWER1":"ON"}
08:32:29 MQT: stat/techo/POWER1 = ON
08:32:30 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
08:32:30 MQT: stat/techo/POWER1 = OFF
08:32:30 MQT: stat/techo/RESULT = {"POWER2":"ON"}
08:32:30 MQT: stat/techo/POWER2 = ON
08:32:30 MQT: stat/techo/RESULT = {"shutterstop1":32}
08:32:30 MQT: stat/techo/SHUTTER1 = 31
08:32:30 MQT: stat/techo/RESULT = {"POWER2":"OFF"}
08:32:30 MQT: stat/techo/POWER2 = OFF

Thanks for everything and good work

turin400, 2018-09-17:
I forgot to tell you the configuration I have:
SETOPTION80 1, SETOPTION14 1, SETOPTION81 1, SHUTTERRELAY2 3
SHUTTEROPENDURATION2 45 , SHUTTEROPENDURATION2 45
SHUTTERCLOSEDURATION2 45,  SHUTTERCLOSEDURATION2 45

stefanbode, 2018-09-17:
Looks like you have defined a pulsetime on one of these relays. This indicates that you have momentary Schutters that need pulses to start and stop. Please ensure pulsetime on your shutterrelays are 0.

stefanbode, 2018-09-17:
Anyhow I will check if I can rsend produce. The if then else are quite complex.

stefanbode, 2018-09-17:
Hi @turin400. I just uploaded a sync to the newest TASMOTAmain branch 6.2.1.4. I checked my current version and cannot reproduce the issue. Can you check with the current version, please? THX,

turin400, 2018-09-17:
I have compiled the new source and the problem continues and it is the same in the two shutter,

12:26:33 MQT: stat/techo/RESULT = {"POWER1":"ON"}
12:26:33 MQT: stat/techo/POWER1 = ON
12:26:33 MQT: stat/techo/RESULT = {"shutteropen1":100}
12:26:34 Shutter 0: Real Pos: 10400, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 18
12:26:35 Shutter 0: Real Pos: 12400, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 38
12:26:36 Shutter 0: Real Pos: 14300, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 57
12:26:37 Shutter 0: Real Pos: 16400, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 78
12:26:38 Shutter 0: Real Pos: 18300, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 97
12:26:38 MQT: stat/techo/RESULT = {"POWER1":"ON"}
12:26:38 MQT: stat/techo/POWER1 = ON
12:26:38 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
12:26:38 MQT: stat/techo/POWER1 = OFF
12:26:39 MQT: stat/techo/RESULT = {"POWER2":"ON"}
12:26:39 MQT: stat/techo/POWER2 = ON
12:26:39 MQT: stat/techo/RESULT = {"shutterstop1":20}
12:26:39 MQT: stat/techo/SHUTTER1 = 19
12:26:39 MQT: stat/techo/RESULT = {"POWER2":"OFF"}
12:26:39 MQT: stat/techo/POWER2 = OFF
12:26:42 MQT: stat/techo/RESULT = {"POWER3":"ON"}
12:26:42 MQT: stat/techo/POWER3 = ON
12:26:42 MQT: stat/techo/RESULT = {"shutteropen2":100}
12:26:42 Shutter 1: Real Pos: 1100, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 11
12:26:43 Shutter 1: Real Pos: 3000, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 30
12:26:44 Shutter 1: Real Pos: 5000, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 50
12:26:45 Shutter 1: Real Pos: 7100, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 71
12:26:46 MQT: stat/techo/RESULT = {"POWER3":"ON"}
12:26:46 MQT: stat/techo/POWER3 = ON
12:26:46 MQT: stat/techo/RESULT = {"POWER3":"OFF"}
12:26:46 MQT: stat/techo/POWER3 = OFF
12:26:46 MQT: stat/techo/RESULT = {"POWER4":"ON"}
12:26:46 MQT: stat/techo/POWER4 = ON
12:26:46 MQT: stat/techo/RESULT = {"shutterstop2":8}
12:26:46 MQT: stat/techo/SHUTTER2 = 7
12:26:46 MQT: stat/techo/RESULT = {"POWER4":"OFF"}
12:26:46 MQT: stat/techo/POWER4 = OFF

turin400, 2018-09-17:
--now what happens is that first shutteropen1 after shutteropen2 when I do shutterstop1 stops shutter 1 and shuter 2.

23:31:12 MQT: stat/techo/RESULT = {"POWER1":"ON"}
23:31:12 MQT: stat/techo/POWER1 = ON
23:31:12 MQT: stat/techo/RESULT = {"shutteropen1":100}
23:31:12 Shutter 0: Real Pos: 100, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 1
23:31:13 MQT: stat/techo/RESULT = {"POWER3":"ON"}
23:31:13 MQT: stat/techo/POWER3 = ON
23:31:13 MQT: stat/techo/RESULT = {"shutteropen2":100}
23:31:13 Shutter 0: Real Pos: 2200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 22
23:31:13 Shutter 1: Real Pos: 300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 3
23:31:14 Shutter 0: Real Pos: 4200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 42
23:31:14 Shutter 1: Real Pos: 2300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 23
23:31:15 Shutter 0: Real Pos: 6200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 62
23:31:15 Shutter 1: Real Pos: 4300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 43
23:31:16 Shutter 0: Real Pos: 8100, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 81
23:31:16 Shutter 1: Real Pos: 6200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 62
23:31:17 Shutter 0: Real Pos: 10200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 102
23:31:17 Shutter 1: Real Pos: 8300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 83
23:31:18 Shutter 0: Real Pos: 12100, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 121
23:31:18 Shutter 1: Real Pos: 10200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 102
23:31:19 Shutter 0: Real Pos: 14200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 142
23:31:19 Shutter 1: Real Pos: 12300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 123
23:31:19 MQT: stat/techo/RESULT = {"shutterstop1":17}
23:31:19 MQT: stat/techo/SHUTTER2 = 14
23:31:19 MQT: stat/techo/RESULT = {"POWER3":"OFF"}
23:31:19 MQT: stat/techo/POWER3 = OFF
23:31:19 MQT: stat/techo/SHUTTER1 = 17
23:31:19 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
23:31:19 MQT: stat/techo/POWER1 = OFF

--also when closing

23:37:10 MQT: stat/techo/RESULT = {"POWER2":"ON"}
23:37:10 MQT: stat/techo/POWER2 = ON
23:37:10 MQT: stat/techo/RESULT = {"shutterclose1":0}
23:37:11 Shutter 0: Real Pos: 59900, Target 0, source: Shutter, pos %: 67, direction: -1, rtcshutter: 4
23:37:11 MQT: stat/techo/RESULT = {"POWER4":"ON"}
23:37:11 MQT: stat/techo/POWER4 = ON
23:37:11 MQT: stat/techo/RESULT = {"shutterclose2":0}
23:37:12 Shutter 0: Real Pos: 57900, Target 0, source: Shutter, pos %: 67, direction: -1, rtcshutter: 24
23:37:12 Shutter 1: Real Pos: 42400, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 11
23:37:13 Shutter 0: Real Pos: 55900, Target 0, source: Shutter, pos %: 67, direction: -1, rtcshutter: 44
23:37:13 Shutter 1: Real Pos: 40400, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 31
23:37:14 Shutter 0: Real Pos: 53900, Target 0, source: Shutter, pos %: 67, direction: -1, rtcshutter: 64
23:37:14 Shutter 1: Real Pos: 38400, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 51
23:37:15 Shutter 0: Real Pos: 51900, Target 0, source: Shutter, pos %: 67, direction: -1, rtcshutter: 84
23:37:15 Shutter 1: Real Pos: 36400, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 71
23:37:16 Shutter 0: Real Pos: 49900, Target 0, source: Shutter, pos %: 67, direction: -1, rtcshutter: 104
23:37:16 Shutter 1: Real Pos: 34400, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 91
23:37:16 MQT: stat/techo/RESULT = {"shutterstop1":53}
23:37:16 MQT: stat/techo/SHUTTER2 = 37
23:37:16 MQT: stat/techo/RESULT = {"POWER4":"OFF"}
23:37:16 MQT: stat/techo/POWER4 = OFF
23:37:17 Shutter 0: Real Pos: 47900, Target 47700, source: Shutter, pos %: 67, direction: -1, rtcshutter: 124
23:37:17 MQT: stat/techo/SHUTTER1 = 53
23:37:17 MQT: stat/techo/RESULT = {"POWER2":"OFF"}
23:37:17 MQT: stat/techo/POWER2 = OFF

stefanbode, 2018-09-18:
Thanks for the intensive testing. Especially the side effects when several shutter work simultaneously. Your last bug was really hard to find because there was absolutely no reason why shutter2 stop on a shutter1 stop. I got the hint when accidentally started shutter 2 before shutter1. In this configuration, it worked. There was one little variable, where I save the bitmap of the last operated relays to react on manual changes and I forgot to =0 this at the end. Any additional bugs are welcome.

turin400, 2018-09-18:
--Thanks to you for your great work.
I have tried the new code and I have found another problem.
Starting from shutterposition1 0, command by mqtt, shutteropen1 then shuttestop1 then power1 off. Then shutteropen1 then shutterstop1 then power1 off and quickly 1 time more shutterstop1 and then power2 on. The same thing happens is the shutter2.

09:04:11 Switched relay: 1
09:04:11 MQT: stat/techo/RESULT = {"POWER1":"ON"}
09:04:11 MQT: stat/techo/POWER1 = ON
09:04:11 MQT: stat/techo/RESULT = {"shutteropen1":100}
09:04:11 Shutter 0: Real Pos: 300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 3
09:04:12 Shutter 0: Real Pos: 2300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 23
09:04:13 Shutter 0: Real Pos: 4300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 43
09:04:14 Shutter 0: Real Pos: 6300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 63
09:04:15 Shutter 0: Real Pos: 8300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 83
09:04:16 Shutter 0: Real Pos: 10300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 103
09:04:17 Shutter 0: Real Pos: 12300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 123
09:04:18 Shutter 0: Real Pos: 14300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 143
09:04:19 Shutter 0: Real Pos: 16300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 163
09:04:20 Shutter 0: Real Pos: 18300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 183
09:04:21 Shutter 0: Real Pos: 20300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 203
09:04:21 MQT: stat/techo/RESULT = {"shutterstop1":23}
09:04:21 MQT: stat/techo/SHUTTER1 = 23
09:04:21 Switched relay: 1
09:04:21 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
09:04:21 MQT: stat/techo/POWER1 = OFF
09:04:25 Switched relay: 1
09:04:25 MQT: stat/techo/RESULT = {"POWER1":"ON"}
09:04:25 MQT: stat/techo/POWER1 = ON
09:04:25 MQT: stat/techo/RESULT = {"shutteropen1":100}
09:04:25 Shutter 0: Real Pos: 21600, Target 90000, source: Shutter, pos %: 23, direction: 1, rtcshutter: 9
09:04:26 Shutter 0: Real Pos: 23600, Target 90000, source: Shutter, pos %: 23, direction: 1, rtcshutter: 29
09:04:27 Shutter 0: Real Pos: 25600, Target 90000, source: Shutter, pos %: 23, direction: 1, rtcshutter: 49
09:04:28 Shutter 0: Real Pos: 27600, Target 90000, source: Shutter, pos %: 23, direction: 1, rtcshutter: 69
09:04:29 MQT: stat/techo/RESULT = {"shutterstop1":33}
09:04:29 Shutter 0: Real Pos: 29600, Target 29700, source: WebGui, pos %: 23, direction: 1, rtcshutter: 89
09:04:29 MQT: stat/techo/SHUTTER1 = 33
09:04:29 Switched relay: 1
09:04:29 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
09:04:29 MQT: stat/techo/POWER1 = OFF
09:04:32 Switched relay: 2
09:04:32 MQT: stat/techo/RESULT = {"POWER2":"ON"}
09:04:32 MQT: stat/techo/POWER2 = ON
09:04:32 MQT: stat/techo/RESULT = {"shutterstop1":15}
09:04:32 Shutter 0: Real Pos: 29300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 4
09:04:33 Shutter 0: Real Pos: 27300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 24
09:04:34 Shutter 0: Real Pos: 25300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 44
09:04:35 Shutter 0: Real Pos: 23300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 64
09:04:36 Shutter 0: Real Pos: 21300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 84
09:04:37 Shutter 0: Real Pos: 19300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 104
09:04:38 Shutter 0: Real Pos: 17300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 124
09:04:39 Shutter 0: Real Pos: 15300, Target 13500, source: Shutter, pos %: 33, direction: -1, rtcshutter: 144
09:04:40 MQT: stat/techo/SHUTTER1 = 15
09:04:40 Switched relay: 2
09:04:40 MQT: stat/techo/RESULT = {"POWER2":"OFF"}
09:04:40 MQT: stat/techo/POWER2 = OFF

stefanbode, 2018-09-18:
Yes, very interesting behavior if sending command just multiple times and without a context like stop a shutter that is already stopped. But I will fix them all :-) Good luck