stefanbode/Sonoff-Tasmota #68: problem with web button 4ch pro

turin400, 2018-09-19:
I have 2 shutter configured as follows:
SETOPTION80 1, SETOPTION14 1, SETOPTION81 1, SHUTTERRELAY2 3
SHUTTEROPENDURATION1 45 , SHUTTEROPENDURATION1 45
SHUTTERCLOSEDURATION2 45, SHUTTERCLOSEDURATION2 45

the problem is in the control from the buttons of the web, I pulse the first to open and start, I pulse the second to change direction and change direction but it never ends closing.

09:53:00 MQT: stat/techo/RESULT = {"POWER1":"ON"}
09:53:00 MQT: stat/techo/POWER1 = ON
09:53:00 Shutter 0: Real Pos: 1200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 12
09:53:01 Shutter 0: Real Pos: 3200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 32
09:53:02 Shutter 0: Real Pos: 5300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 53
09:53:03 Shutter 0: Real Pos: 7300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 73
09:53:04 Shutter 0: Real Pos: 9200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 92
09:53:05 Shutter 0: Real Pos: 11300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 113
09:53:06 Shutter 0: Real Pos: 13200, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 132
09:53:07 Shutter 0: Real Pos: 15300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 153
09:53:08 Shutter 0: Real Pos: 17300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 173
09:53:09 Shutter 0: Real Pos: 19300, Target 90000, source: Shutter, pos %: 0, direction: 1, rtcshutter: 193
09:53:10 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
09:53:10 MQT: stat/techo/POWER1 = OFF
09:53:11 MQT: stat/techo/RESULT = {"POWER2":"ON"}
09:53:11 MQT: stat/techo/POWER2 = ON
09:53:11 MQT: stat/techo/SHUTTER1 = 24

The same thing happens in the second shutter and also if we first close and then open.

Thank you

stefanbode, 2018-09-19:
yes the fix from yesterday broke this up. let me have a look how to ensure this also will work.

turin400, 2018-09-21:
Ok, thanks again.
Greetings, Sergio

closed 2018-09-21 by turin400