stefanbode/Sonoff-Tasmota #69: Short Circuit Safe Configuration (help needed)

Michel-IOT, 2018-09-20:
Hello and Greetings from Belgium.

I am trying to configure a nodemcu with your firmware (latest rev as of 20/09/2018).

Am using a PCF8574 to drive 4 relays according to the Short Circuit safe wire configuration.

As far as parameters are concerned, I use SETOPTION14 0 / SETOPTION80 1 / SETOPTION81 0 / PULSETIME 0

What I observe: / From Web interface:

On Shutter open .. Relay1 triggers .. as expected

On Shutter close .. Relay1 triggers ..... problem ... Relay2 does not trigger at same time .. which is a problem for short circuit safe hardware configuration.

With the buttons
I am not sure how to configure them so button1 will do Open (Trigger relay1)  and button2 will do a Close (trigger relay1+relay2).

Thank you for your time and help.

stefanbode, 2018-09-21:
Relay 2 must be a switch relay with 3 ports switching the main from relay 1 to wire 1 or wire 2. on on/off relay does not work as a relay 2.

turin400, 2018-09-21:
I'm trying the latest version in Short Circuit Safe mode and I think it's not working properly,
I'm running from webbutton1 and when I press webbutton2 it turns off power1 and it quickly turns on and off power2.

12:42:48 MQT: stat/techo/RESULT = {"POWER1":"ON"}
12:42:48 MQT: stat/techo/POWER1 = ON
12:42:49 Shutter 0: Real Pos: 10000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 11
12:42:50 Shutter 0: Real Pos: 12000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 31
12:42:51 Shutter 0: Real Pos: 14000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 51
12:42:52 Shutter 0: Real Pos: 16000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 71
12:42:53 Shutter 0: Real Pos: 18000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 91
12:42:54 Shutter 0: Real Pos: 20000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 111
12:42:55 Shutter 0: Real Pos: 22000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 131
12:42:56 Shutter 0: Real Pos: 24000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 151
12:42:57 Shutter 0: Real Pos: 26000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 171
12:42:58 Shutter 0: Real Pos: 28000, Target 90000, source: Shutter, pos %: 9, direction: 1, rtcshutter: 191
12:42:58 MQT: stat/techo/RESULT = {"POWER2":"ON"}
12:42:58 MQT: stat/techo/POWER2 = ON
12:42:58 MQT: stat/techo/SHUTTER1 = 31
12:42:58 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
12:42:58 MQT: stat/techo/POWER1 = OFF
12:42:58 MQT: stat/techo/RESULT = {"POWER2":"OFF"}
12:42:58 MQT: stat/techo/POWER2 = OFF

via mqtt it happens that only the two relays are activated if the shutter is in operation and we change from shutteropen to shutterclose for example. If we start from the stopped shutter, it never activates the two relays.

12:52:04 MQT: stat/techo/RESULT = {"POWER1":"ON"}
12:52:04 MQT: stat/techo/POWER1 = ON
12:52:04 MQT: stat/techo/RESULT = {"shutteropen1":100}
12:52:05 Shutter 0: Real Pos: 33500, Target 90000, source: Shutter, pos %: 37, direction: 1, rtcshutter: 2
12:52:06 Shutter 0: Real Pos: 35500, Target 90000, source: Shutter, pos %: 37, direction: 1, rtcshutter: 22
12:52:07 Shutter 0: Real Pos: 37500, Target 90000, source: Shutter, pos %: 37, direction: 1, rtcshutter: 42
12:52:08 Shutter 0: Real Pos: 39500, Target 90000, source: Shutter, pos %: 37, direction: 1, rtcshutter: 62
12:52:09 Shutter 0: Real Pos: 41500, Target 90000, source: Shutter, pos %: 37, direction: 1, rtcshutter: 82
12:52:09 Stop moving shutter 1: direction:1
12:52:09 MQT: stat/techo/RESULT = {"shutterstop1":48}
12:52:09 MQT: stat/techo/SHUTTER1 = 48
12:52:09 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
12:52:09 MQT: stat/techo/POWER1 = OFF
12:52:14 MQT: stat/techo/RESULT = {"POWER1":"ON"}
12:52:14 MQT: stat/techo/POWER1 = ON
12:52:14 MQT: stat/techo/RESULT = {"shutterclose1":0}
12:52:15 Shutter 0: Real Pos: 42600, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 6
12:52:16 Shutter 0: Real Pos: 40600, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 26
12:52:17 Shutter 0: Real Pos: 38600, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 46
12:52:18 Shutter 0: Real Pos: 36600, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 66
12:52:19 Shutter 0: Real Pos: 34600, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 86
12:52:20 Shutter 0: Real Pos: 32600, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 106
12:52:21 Shutter 0: Real Pos: 30600, Target 0, source: Shutter, pos %: 48, direction: -1, rtcshutter: 126
12:52:21 Stop moving shutter 1: direction:-1
12:52:21 MQT: stat/techo/RESULT = {"shutterstop1":30}
12:52:22 Shutter 0: Real Pos: 28600, Target 27000, source: WebGui, pos %: 48, direction: -1, rtcshutter: 146
12:52:22 MQT: stat/techo/SHUTTER1 = 30
12:52:22 MQT: stat/techo/RESULT = {"POWER1":"OFF"}
12:52:22 MQT: stat/techo/POWER1 = OFF

web slider tiene el mismo problema que desde mqtt.

a greeting.

stefanbode, 2018-09-21:
I‘m so sorry. You’re totally right. Tha last changes have broken the functionality. I will provide a fix during the weekend.

turin400, 2018-09-28:
Goodnight,
Have you been able to implement the solution for these problems?
Greetings,

stefanbode, 2018-09-29:
Hi just after the last comment I had an accident an now I’m laying i the hospital.  I had a look today but was not able to fix it. Maybe tomorrow. Now I have plenty of time.....

turin400, 2018-09-29:
I am so sorry. The priority is health.

A hug and come home soon.

Michel-IOT, 2018-09-30:
Hi,
Sorry to read about accident.  Indeed health is priority.  Get well !

Michel-IOT, 2018-09-30:
Sorry, .. pushed the wrong button ... I did not mean to close the issue.

stefanbode, 2018-09-30:
Hi Michael, I cleaned up several things and tested now all three configurations. I would assume it will work now.

Michel-IOT, 2018-10-02:
Hi, I rebuilt everything + tested the 2-relay config in short circuit safe mode, and am happy to confirm it is working as expected.  Thanks for your help and support.
Cheers, Michel.

closed 2018-10-04 by stefanbode

AdrielGH, 2018-10-28:
Hello all, Greetings from Brazil.

@stefanbode I hope you are totally recovered from your accident!!

Somebody could help me?

I have a Sonoff 4CH Pro with Short Circuit Safe Configuration. I'm also using SETOPTION14 0 / SETOPTION80 1 / SETOPTION81 0 / PULSETIME 0.

When I'm using web slider it works well. 
Sliding to open... Relay1 triggers .. as expected
Sliding to close... Relay1 and Relay2 triggers .. also as expected

When I'm using webbutton or physical button I have problem just to close.
pressing button 1 to open... Relay1 triggers... as expected
pressing button 2 to close... Relay2 triggers... but relay 1(on/off)... don't trigger...

16:36:41 MQT: hm/status/sonoff/SENSOR = {"Time":"2018-10-28T16:36:41","SHUTTER-1":100}
16:36:43 MQT: hm/status/sonoff/RESULT = {"POWER2":"ON"}
16:36:43 MQT: hm/status/sonoff/POWER2 = ON (retained)
16:36:43 Shutter 0: Real Pos: 8200, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 18
16:36:44 Shutter 0: Real Pos: 6200, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 38
16:36:45 Shutter 0: Real Pos: 4200, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 58
16:36:46 Shutter 0: Real Pos: 2200, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 78
16:36:47 Shutter 0: Real Pos: 200, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 98
16:36:48 MQT: hm/status/sonoff/SHUTTER1 = 0 (retained)
16:36:48 MQT: hm/status/sonoff/RESULT = {"POWER1":"OFF"}
16:36:48 MQT: hm/status/sonoff/POWER1 = OFF (retained)
16:36:48 MQT: hm/status/sonoff/RESULT = {"POWER2":"OFF"}
16:36:48 MQT: hm/status/sonoff/POWER2 = OFF (retained)

Look below that when I do the same thing using slider it trigger Relay 1 and 2.

16:39:05 MQT: hm/status/sonoff/RESULT = {"POWER2":"ON"}
16:39:05 MQT: hm/status/sonoff/POWER2 = ON (retained)
16:39:05 MQT: hm/status/sonoff/RESULT = {"POWER1":"ON"}
16:39:05 MQT: hm/status/sonoff/POWER1 = ON (retained)
16:39:05 MQT: hm/status/sonoff/RESULT = {"shutterposition1":0}
16:39:06 Shutter 0: Real Pos: 8700, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 13
16:39:07 Shutter 0: Real Pos: 6700, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 33
16:39:08 Shutter 0: Real Pos: 4700, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 53
16:39:09 Shutter 0: Real Pos: 2700, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 73
16:39:10 Shutter 0: Real Pos: 700, Target 0, source: , pos %: 100, direction: -1, rtcshutter: 93
16:39:10 MQT: hm/status/sonoff/SHUTTER1 = 0 (retained)
16:39:10 MQT: hm/status/sonoff/RESULT = {"POWER1":"OFF"}
16:39:10 MQT: hm/status/sonoff/POWER1 = OFF (retained)
16:39:10 MQT: hm/status/sonoff/RESULT = {"POWER2":"OFF"}
16:39:10 MQT: hm/status/sonoff/POWER2 = OFF (retained)

Thanks!!

stefanbode, 2018-10-28:
Yes this is intended behavior. With this configuration button 1 is on off and the other the direction. If you change direction it will alway go off first. As fa4 as I know there is no way with buttons to make one up and one down.

AdrielGH, 2018-10-28:
Thank you for clarifying!!