stefanbode/Sonoff-Tasmota #71: DeepSleep feature missing

Kallenator, 2018-09-25:
Hello and greetings,

I was very happy to see a fork of Tasmota that would enable battery-driven projects though the implementation of the deepsleep feature.

But now I am looking at the current version (6.2.1.4) of the stefanbode/Sonoff-Tasmota (meaning that I downloaded the _master_, made my settings and flashed it via the Arduino IDE onto a ESP-01) and cannot find any hint of the deepsleep feature. I always get _unknown command_ as feedback. Then I checked the code and found nothing under the keyword 'deepsleep' either.

Could this have been overwritten by syncing with the arendst-Tasmota or am I doing something wrong here?

Thank you very much for your help. Best regards,
Kalle

stefanbode, 2018-09-25:
Hi Kalle, as Long as you checkout the master branch everything should be ok. Please see the excellent Wiki I try to create with this fork. To ensure wake up as always with the deep sleep please ensure RST AND gio0 are connected to get the wake up all. I also support a gpio Pin like espeasy to simple switch on/off the deep sleep. All in the wiki.

Kallenator, 2018-09-25:
Hi Stefan,

thank you for your quick reply. This is realy strange: I could have sworn that there was not once an occurrence of 'deepsleep' in the code. Now I've downloaded the master once more and now I found that the code pieces are there. I'll flash it asap.

Maybe it was because I had the arendst-Tasmota with the folder 'sonoff' in the Arduino IDE sketchbook and the IDE wanted to use the same folder for your version.
Hardware shouldn't be a problem.

Thanx for your support and sharing your good work!
Regards!

closed 2018-09-25 by Kallenator