stefanbode/Sonoff-Tasmota #72: Sonoff 4CH Pro R2 firmware and scheme

achillealb, 2018-09-28:
I have Sonoff 4CH Pro R2
1- Does your firmware work with Sonoff 4CH Pro R2?
2-Which of these patterns can you use with your firmware?
3- do you have the firmware in the bin format for Sonoff 4CH Pro R2?
[image]
[image]

stefanbode, 2018-10-01:
Yes fore sure it iss supporter, and the shown 1st layout will work. This is the normal configuration. The second layout will not wok because it his case you change the shutterposition without giving this information to the firmware. This is causing a messs if it is out of sync. But you can use the switches of the esp to trigger up and down from a switch. This will work. The only thing you have to check that there are no two parallel pathes to change position.

closed 2018-10-04 by stefanbode

Enrico75s, 2018-11-25:
Hi, what is the configuration in Home Assistant for a second shutter? From the web ui I configured it's ok, I tried some configurations but I can not find the exact commands. Thank's

stefanbode, 2018-11-25:
Taking the config for the last release I suppose the config for shutter2 should look like this
```

 platform: mqtt
    name: "Balcony Blinds"
    availability_topic: "tele/sonoff/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    position_topic: stat/sonoff/SHUTTER2
    position_open: 100
    position_closed: 0
    set_position_topic: "cmnd/sonoff/shutterposition2"
    value_template: "{{ value | int }}"
    command_topic: "cmnd/sonoff/backlog"
    payload_open: "SHUTTEROPEN2"
    payload_close: "SHUTTERCLOSE2"
    payload_stop: "SHUTTERSTOP2"
    retain: false
    optimistic: false
    qos: 1
```