stefanbode/Sonoff-Tasmota #74: Shutterstop problem when 50% corrected

junior013, 2018-10-03:
I'm trying your code on a 4ch and it works well, but if I correct the 50% position, the shutter under 50% don't stop immediately. In closing direction it runs over, in opening direction it stops and runs back a bit. I think, the code correcting the position value, when stop command sent and sets the shutter to the corrected value.

stefanbode, 2018-10-04:
Thanks for bringing this up. This was not tested after the recent changes. Will try to reproduce .

stefanbode, 2018-10-04:
Hi the shutter50percent must be between 50..100 and defines the position without invert. If the shutter is completely closed this is 0%. to open it 50% you normally need more time because during the first movements it only open the gaps. Therefore 60 or 65% is a typical correction value. I can't see a problem currently. Please provide the init parameters in the beginning. should look like this:

```
10:48:21 Shutter 0 (Relay:1): Init. Pos: 45180 [17 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 90000, opentime 45 [s], Closetime 45 [s], Calc: m1: 540, b1 36000, m2: 7740, binmask 15, is inverted 0
10:48:21 Shutter 1 (Relay:3): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 90000, Opentime 45 [s], Closetime 45 [s], Calc: m1: 900, b1 0, m2: 900, binmask 15, is inverted 0
```

junior013, 2018-10-04:
Here are my init params:
```
00:00:00 Shutter 0 (Relay:1): Init. Pos: 7770 [1 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 74000, Opentime 37 [s], Closetime 37 [s], Calc: m1: 370, b1 37000, m2: 7770, binmask 3, is inverted 0
00:00:00 Shutter 1 (Relay:3): Init. Pos: 5270 [1 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 52000, Opentime 26 [s], Closetime 26 [s], Calc: m1: 270, b1 25000, m2: 5270, binmask 15, is inverted 0
00:00:00 Shutter 2 (Relay:255): Init. Pos: 12739514 [255 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 510000, Opentime 255 [s], Closetime 255 [s], Calc: m1: 49726, b1 59384, m2: 61602, binmask -1073741809, is inverted 255
00:00:00 Shutter 3 (Relay:255): Init. Pos: 12739514 [255 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 510000, Opentime 255 [s], Closetime 255 [s], Calc: m1: 49726, b1 59384, m2: 61602, binmask -1073741809, is inverted 255
```
the SHUTTER50PERCENT settings are 75 and 74 for the 2 shutters. Can you tell me how do you calculate with it? Only add/divide the time differenc at/to 0% position?

stefanbode, 2018-10-05:
The shutter50percent is not calculated. You manually open it 50 percent and then read the value reported. During this calibration shutter50percent must be 50. The value you see, you set as the parameter. Done. I will copy your parameter and see how my test shutter behave.

stefanbode, 2018-10-05:
Please set shutterrelay3 to 0. The you do not get the crazy shutter 3 and 4

junior013, 2018-10-05:
> The shutter50percent is not calculated.
Yes, it's clear. But how your code modifies the runtimes with this value? Because, after it set, the 50% is 50%, but touch position is position 2 or 3 (depends which shutter) and position 5 looks like ~10-15% open.
I know, it differs on shutter type. If speed differs on position (by the weight of the shutter) then all runtimes must be corrected with the ratio between 50 and real 50% position value - I can't imagine, how complex algorythm :( But, if some time needed only to open the gaps then the difference only be added from/to 0 position. So, is it possible somhow (maybe with another setoption value, or differentiate with SHUTTER50PERCENT + or - value), to change between the algorythms? Or do you use any other calculation method?

And, another little comment to the runtimes: If shutter used for a while without full open/close, the position may slip a bit. It's OK, can be "reset" at the ends, but because of it, the open/close don't runs really to the ends :( May need some extra runtime, if target position is 0 or 100 - not 10% as in @arendst fork, just a second or a half.

stefanbode, 2018-10-05:
The last comment make sense. And a second on top is quite a good idea, even if it is not enough. Maybe I will calculate something based on the operation time without hitting the endstop. Good idea. 
On this wooden/Alu shutters the movement is without the gaps a polynomial function of degree 2. Nothing really complicated. Is the actual position of 30% and 70% so far away of where it should be? My assumption was that the error is small enough to ignore. My problem is: I don’t want to make the usage complicated. With the gaps and the nonlinear stuff you need at least 4 calibration points to calculate the coefficients.

stefanbode, 2018-10-05:
Opening the gaps should be 5%. But then for sure 10% could not be 10% because it was still close at 5%. If I switch it to 1% the you could not open the gaps half. It’s just 0 and 1. that’s the reason why the first 5% are reserved for this action.

stefanbode, 2018-10-06:
Ok I will change or let me name it to enhance the calibration process. One to make it simple and one to make it accurate. With the current update there is a 250ms add on for each movement that will not hit the 0 endstop (not implemented for the max endstop). Therefore if you move it 4 times and then go to 0 you will get 1sec additional runtime. If you like i want to enhance the calibration. For me it looks like your motor has some delay to start. can you please measure the actual opening, if you go from fully open=0 to the following values. I need the position in cm of the shutter from ground.
shutterposition 80, shutterposition 60, shutterposition 40, shutterposition 80, shutterposition 40, shutterposition 60, shutterposition 80.

Please do it exactly this way. There is no value missing. There is a plan behind.

My idea is to do a calibration run with the following positions: fully closed, open, the gap, 30% open, 60% open 100%open. The gaps will be a linear function and the rest will be a polynomial with order 2 or 3. This will allow me to calculate the thickness of the shutter, the actual diameter on the axle and so on. Anyhow, I need first the information if there are shutters that need a startup time before they start actual moving.

junior013, 2018-10-11:
I updated all of my devices (4ch, dual, T1 2ch) to the last version, and now all works well, with correct calibration. (except gap open percentage, that's still 2-3% for me - depending on the length of the actual shutter)
Thank you for the super-fast help!

junior013, 2018-10-12:
Sorry, I missed the calibration data, you asked

```
- Window height is 207.5cm
- Shutter full runtime is 39sec (both direction, with some 0.1 sec difference)
- SHUTTER50PERCENT set to 67

Shutter 0 (Relay:1): Init. Pos: 11668 [2 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 78000, Opentime 39 [s], Closetime 39 [s], Calc: m1: 514, b1 26600, m2: 5834, binmask -1073741809, is inverted 0
Measured values are:
- 100 to 80%, stops at: 160.5cm (77.34%)
- 80 to 60%: 119cm (57.34%)
- 60 to 40%: 81.5cm (39.27%)
- 40 to 80%: 159.5cm (76.86%)
- 80 to 40%: 80.5cm (38.79%)
- 40 to 60%: 117.5cm (56.62%)
- 60 to 80%: 158.5cm (72.87%)
- 80 to 100%: 204cm (98.31%)
So, there's 3,5cm slip down during this 8 movements, but it's in 2% of the runtime, around 0.1sec/movement - I don't think it can be corrected.
```

stefanbode, 2018-10-12:
Thank you very much. Let me check. For sure there is value in this data. Will let you know as soon as I have an update

electrontube, 2018-10-24:
@stefanbode 
when shutteropenduration and shuttercloseduration are set quite close to the measured time values to fully open / close the shutter, there still is the problem that when a shutter is moved up/down several times without hitting the endstop, the calibration becomes more and more inaccurate and "shutterposition 100" will not fully open the shutter as the Software believes it has reached the end position.

Is there a chance to add a variable that allows the user to add some extra "runtime" when shutterposition 0 or 100 is executed.

I often set shutters to 5 or 10% rather than 0% and calibration becomes inaccurate quite quickly.

stefanbode, 2018-10-24:
Hi, there is already some runtime depending on the number of moves without hitting the end stop. Anyhow I can see in your measured numbers that the UP and DOWN time correlation is not exact. This causes the shutter to shift continuously. I already started to implement a new correlation matrix to overcome this issue. Also a problem is that the open and closetime is only in seconds. This makes it impossible to set the relation exactly. My idea is to get opentime and then the 6 measures you provided and calculate everything with these measures. This will be accurate as hell, but is for sure a big effort to get it operating. My current idea is to have both methods. As mentioned already changed the code in that way. Next step is to implement that 0.1seconds step problem and the correlation matrix.

stefanbode, 2018-10-24:
I changed the code to finer set the openspeed compared to closedspeed. Resolution changes from 1 sec to 0.1 sec.

B1G1, 2018-10-25:
> I changed the code to finer set the openspeed compared to closedspeed. Resolution changes from 1 sec to 0.1 sec.

so we can set shutteropenduration and shuttercloseduration with something like 20.1 or what?

stefanbode, 2018-10-25:
Yes, this is exactly what it means. This is part of a major calibration project that ensures Shutter at exact positions also for a longer time, if not hitting endstop. Release planned for the weekend. But currently I got feedback that there is a problem again with shutteropen and sutterclose. Let me fix this first.

electrontube, 2018-10-25:
does not compile:

xdrv_97_shutter:27:214: error: expected ',' or ';' before 'D_CMND_CALIBRATIONMATIX'
   D_CMND_OPEN "|" D_CMND_CLOSE "|" D_CMND_STOP "|" D_CMND_POSITION  "|" D_CMND_OPENTIME "|" D_CMND_CLOSETIME "|" D_CMND_SHUTTERRELAY "|" D_CMND_SET50PERCENT "|" D_CMND_SHUTTERSETCLOSE "|" D_CMND_SHUTTERINVERT "|" D_CMND_CALIBRATIONMATIX;
                                                                                                                                                                                                                      ^
exit status 1
expected ',' or ';' before 'D_CMND_CALIBRATIONMATIX'

stefanbode, 2018-10-26:
Oh yes, you do not compile in EN. Let me update the other language files..

electrontube, 2018-10-27:
@stefanbode 
Thx for the update,
I have done some testing...

I have moved the shutter from 0% to 4%
(4% is the Position where just all gaps are open, so I don't need to measure anything as I can directly see if there is any slip with further movements)
Then I have moved several times back and forth between 4% and 30% and with every move the physical blind position slips further down towards 0%.
The problem is, once there is a slip towards the close position, the shutter will not fully open anymore unless it is set to 0% where it seems to recalibrate.
As mentioned earlier, I think, some (configurable) extra runtime when position is set to 100% and a recalibration at the fully open position would be great.
(recalibration at 0% appears to be working fine)

stefanbode, 2018-10-27:
Have you Seen, that you can change opentime and closetime now with fractions of a second? E.g. 33.4. please check if this minimize your slip problem by increasing or decreasing the closeduration.

electrontube, 2018-10-27:
I have seen that, but this is not really what we need, as open-/closedurations may vary anyway i.e. depending on temperature.
If the fractions help internally to minimize rounding errors, it's fine, but I doubt, someone would try to calibrate open duration in fractions on seconds.

The idea is: 
calibrate real open-/closeduration as close as possible to the next full second.
whenever the logical position is set to either 0 or 100, give some extra runtime to allow it to reach it's physical endstop and then recalibrate the internal counters (as well at position 0 as at position 100).
There might still be a small inaccuracy, but this can be tolerated as both the endstops can be reached and recalibration is done at both the end stops.

I quite often set shutters between say 10% and 100% without setting it to 0%.
Doing so results in a situation, where you can no longer fully open the shutter without fully closing it first and that is very annoying.
Try to explain to your wife, why that shutter does not fully open when she wants it to be open. . .

stefanbode, 2018-10-28:
If your clseduration is longer than needed, you already have this behavior, because the shutter still operates and the shutter is physically at the endstop. I do operate all my shutters in this way. There is  no difference between extra runtime or just make the time longer

electrontube, 2018-10-28:
I had already given 1-2 extra seconds for open/close duration, but the „slip“ problem is still there.
Probably I will have to add even more for open and see what happens.

Is there e recalibration as well at position 0 as at position 100?

electrontube, 2018-10-31:
Some more test results:

one of my smaller shutters takes 16.5 sec for opening / closing (physical endstop to endstop)
open-/ closeduration are both set to 18, so there should be plenty of extra time for a shutter of that small size.

Now I set positions in a sequence als follows (starting from the 100% physical endstop position)
50%
10%
30%
10%
30%
100%
and guess what, the shutter does not reach the open endstop position.

This behaviour can be reproduced on any of my shutters, regardless of the size (i.e. duration 35)
So 1/10 secs for the duration do not really help as several full seconds are to be added to the real runtime to ensure the shutter can be fully opened without fully closing it first.

When you position shutters depending on presence of sunlight and position of sun, you will have several shutter adjustments throughout the day without ever reaching the close endstop.
It is very annoying when the shutter does not fully open at the end without fully closing it first.

B1G1, 2018-10-31:
> When you position shutters depending on presence of sunlight and position of sun, you will have several shutter adjustments throughout the day without ever reaching the close endstop.
> It is very annoying when the shutter does not fully open at the end without fully closing it first.

Hi electrontube, sorry for the off topic but i would like to know how you control your shutters position depending of the sun light and position.

stefanbode, 2018-10-31:
Hi Bigi, you can use the altitude of the sun and map these to positions. Or you have a light sensor and close the shutter until the light is back into "the dark".  Now regarding the endstop discussion. Sorry to ask so many questions and not just add 10 seconds at the end. You have set open and closeduration to the same value and for me, it looks like the shutter is not at the correct same position at 10% when you go there a second time. It is essential for the mechanism, that the relation between open and close match. the 0.1 second resolution of the timing was mainly implemented to ensure that 50%->20%->50% end at the same position. My shutter move 10x up and down and have less than 1cm missfit. So measure opentime and add 1 second (should be enough to hit the endstop securely). Move to 50%. mark position. Move 20% and 50%. If you are LOWER than before DECREASE closeduration. Again 20% , 50% and adjust again until you reach the position as before. If you move up and down in the 50% startposition between the different measurements this is no problem. you just have a new starting point.

electrontube, 2018-10-31:
@B1G1 
As Stefan already mentioned, given your geographical position, and the time, the position of the sun and the angle of incidence can be calculated.
This combined with a sunlight sensor (simply brightness i.e.LDR, better a combination of brightness and temperature (temp difference between a bright and a dark surface) will lead to good results.

stefanbode, 2018-10-31:
hi @electrontube . This has nothing to do with heavy calibration stuff. shutters behave differently on the load and on the motor attached. some are heavily oversized. therefore open an cluseduration is the same and the shutter reaches up and down always the correct position. But if the shutter is very heavy and or the motor is selected to support this shutter weight, then the opentime will be significantly longer than the closetime. This is nothing you guess upfront. This must be measured. If you do not compensate this the position will shift like in your case and you will not reach the end stop anymore. Except you put heavy additional runtime to the shutter. But this is fixing the symptoms and not the cause. You are the only one that does not care about any values because you open/close with light settings. All other users rely on more or less correct % opening.

electrontube, 2018-10-31:
@stefanbode 
I'm totally aware, there are two values that need to be set correctly, that is open- and closeduration.
I'm also aware, these values are dependent on mechanical setup, motor power etc.
(All this has been mentioned several times in this discussion)

like other users I would also like to get exact (in terms of reproducible) % opening values, but I do not really care wether 48% physical is 46 or 50% logical, as long as shutter will hit that position anytime exactly.
All I have to do is to remember my wifes favourite shutter position is say 57%, that's it. 
probably, due to the calculations (runtime, % opening, 50% correction, etc.) rounding errors accumulate and the accuracy is lost after just a few moves.

As for my test shutter:
It is just a small shutter, open duration only 18s, and the motor certainly is oversized, there is really no difference in open- / closeduration.
This makes it perfect for testing, since we do not have to deal with other impacts.
When I set the durations precisely to the real measured values, positioning inaccuracy starts after 2-3 moves.

Again, little inaccuracy does not really matter, the main problem is, after a few moves errors have accumulated and the shutter will not fully open anymore since it believes, it is already at 100%.
Now imagine, we are talking about a large shutter that takes 40sec to fully open / close.
Due to the mentioned inaccuracies it now is at logically 100% but physically only 90%.
The only way to fully open it now is set it to 0%, wait for about 40sec until it's fully closed, set it to 100% to reopen and wait for another 40sec (in total almost 1.5min)

I'm sure, if users had to explain that to their wives, they would immediately be forced to reinstall the purely mechanical shutter switches.

Last try...
what we need is a way to open the shutter even if it (due to inaccuracies) logically already is at 100%.
And yes, I'm aware, that in this case the physical OPEN Endstop must be working properly.

mabraun, 2018-12-31:
@stefanbode

I have the same Problem like electrontube.

After a few stops the shutter does not close anymore without a full run in opposite direction.

My idea is, that we have to add a „Motor starting time“ in the calculation for all movments. 
Actually it considered in the „SHUTTERxxxxDURATION“ time for one run. It should be considered in all movements and have to be added each time.

What do u think? 

Marcel

electrontube, 2018-12-31:
Since there appears to be no furhter progress in development, I switched back to Pulsetime.
That way I can at least set Shutter Positions precisely anytime.

HAPPY NEW YEAR

danhae, 2019-05-29:
Hi,
same problem!
@stefanbode 
I think @mabraun idea to add a "motor starting time" will fix this issue.
Can i help by giving you any more information?
Maybe a video of my shutter or anything else?

Best regards from Düsseldorf, Germany.
Daniel

stefanbode, 2019-05-30:
Hi danhae, I have this in production on many shutter and also use the endstop function of the shutter. If you increase open and closeduration you will 100% hit the endstop. If the shutter shift during several movements just change the relation between open and closeduration. I am 100% sûre current functionality is sufficient

stefanbode, 2019-05-30:
Hi Daniel, just double checked. If shifting positions is your problem you should change the relation between openduration and closeduration. To do this in fraction of seconds should allow you to get it sync quite smoothly. The issue is that under load the motor wind slower than without.

electrontube, 2019-06-03:
@stefanbode
The issue is not motor load, and fiddeling around with open-/closeduration to eventually get „close“ does not solve the problem.

stefanbode, 2019-06-03:
Gentlemen, if you want support or even enhancements please be as accurate to describe the issue as possible. Maybe you can make a video and upload to YouTube if this helps. Motor start time is only an issue if you get different levels when doing 90-80-50-30-20 compared to 90-20. The first approach must end somewhere below 20. Please make a video and make some cm infos on the end position.

stefanbode, 2019-09-07:
Calibration Video uploaded for everybody who has problems with shutter shifting away from position.

https://youtu.be/Z-grrvnu2bU

zwinsch, 2019-09-19:
Hi to all,
I recognized similiar problems.
I think, they are also related to the following observation:
The moving speed is not only different for up/down direction, but also the difference or relation of up down speed changes with position, is different in the upper and the lower half... so I simply cannot calibrate the 50% value in a way, that the same position is reached coming from 0% or from 100%.
Would be nice, to have a feature, were the 50% position can be calibrated for comming from 0% and for comming from 100% independently...
Also the metioned add on time, in case of the command to 0 or 100%, to make a safe sync is a good idea - could maybe realized by own rules (temporarly overwrite the duration setting)?

stefanbode, 2019-09-19:
There is already some addition run time if the shutter did not close for several movements. You can use rules and define a backlog. For example Shutterrelay1 0;power1 on; wait for some seconds; power1 off; shutterrelay1 1 might do what you want. By undef the shutterrelay you get back control to switch relays on off like you want to do.

stefanbode, 2019-10-09:
There is a new calibrationmatrix committed to the code. Now it should be possible to get all positions correctly. Again open and closetime relation must match and if much longer than real open and closetime the 50% opening from top is different than from Button. If you not believe make it smaller than required und you see it will operate perfectly.

closed 2019-10-09 by stefanbode