stefanbode/Sonoff-Tasmota #77: Configuration of D0/GPIO16 - Missing 71 Deep Sleep in GUI

twinkler2, 2018-10-07:
Hi Stefan,
I am on Version 6.2.1.11 stb-1.1-2_4_0
I want to configure GPIO16 for wake up from Deep Sleep. In the Web GUI I cannot find "71 Deep Sleep" as mentioned in the Wiki Page for Deep Sleep. In the Web GUI seems to be only listed "71 SerBr Tx". I this the right configuration for this Pin for wake up from deep sleep?

Regards

Torsten

stefanbode, 2018-10-07:
Hi Thorsten, thanks for bringing this up. No, this is a bug initiated by the latest enhancements of the original firmware. Let me check where the change got lost.

twinkler2, 2018-10-07:
`Just one addition. You cannot set the wakeup pin this is by default the gpio0 as far as I remember. This switch is to easily enable/disable the complete deepsleep with a Switch for testing or a firmware update.`

ok, thanks a lot for the clarification. I was already surprised why DeepSleep seems to work even though the right config for the pin is missing in the software. I did not try to disable DeepSleep vial a pin.

From the wiki I have to say it is not perfectly clear to me what the difference btw. using "ANY pin and connect it to Ground" or using GPIO16 is.

stefanbode, 2018-10-08:
Anyhow the function is broking with some recent changes in the master firmware. The procedure how to build the drop down list has changed and I have to figure out how to get my value again into the list.

closed 2018-10-09 by stefanbode