stefanbode/Sonoff-Tasmota #78: CHIRP moisture sensor config

madmicio, 2018-10-11:
Hi!
I bought 2 CHIRP moisture sensor from the wiki link. I'm trying to make it work on a  sonoff basic. I disabled the seriallog (seriallog 0). I connected vcc to vcc, gnd to gnd, sda to gpio1, scl to gpio3, .... but nothing happens.

what is the right configuration?

madmicio, 2018-10-12:
v.2.7.6

madmicio, 2018-10-16:
no news?

madmicio, 2018-10-21:
tasmota find the sensor

19:58:12 CMD: i2cscan
19:58:12 MQT: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x20"}

but no information on the main page

madmicio, 2018-10-23:
this code works. can you implement it on your firmware?

#include <Wire.h>

void setup() {
  Wire.begin();
  Serial.begin(9600);
}

void writeI2CRegister8bit(int addr, int value) {
  Wire.beginTransmission(addr);
  Wire.write(value);
  Wire.endTransmission();
}

unsigned int readI2CRegister16bit(int addr, int reg) {
  Wire.beginTransmission(addr);
  Wire.write(reg);
  Wire.endTransmission();
  delay(1100);
  Wire.requestFrom(addr, 2);
  unsigned int t = Wire.read() << 8;
  t = t | Wire.read();
  return t;
}

void loop() {
  Serial.print(readI2CRegister16bit(0x20, 0)); //read capacitance register
  writeI2CRegister8bit(0x20, 3); //request light measurement 
  delay(9000);                   //this can take a while
  Serial.print(", ");
  Serial.println(readI2CRegister16bit(0x20, 4)); //read light register
  delay(500);
}

stefanbode, 2018-10-23:
Yes for sure. Will update tomorrow and let you know.

stefanbode, 2018-10-24:
Hi, the code you provided is exactly the code that is already in the firmware. Take a look at line 25ff you will see the definition on the register. For example to measure light you wait 9 seconds. line 53ff you see I wait for a busy flag and then read the value. That is the method that the manufacturer gave.

Have you set the compiler options:
#define USE_I2C
#define USE_CHIRP

Because there must be after the I2CScan a message that say CHIRP found. Without this message I assume this is not enabled during compilation.

stefanbode, 2018-10-24:
Please send the log file from the beginning after reboot. the CHIRP message must be quite early during the boot process after scanning the bus.

madmicio, 2018-10-24:
> Have you set the compiler options:
#define USE_I2C
#define USE_CHIRP
Where?

here?
// -- I2C sensors ---------------------------------
#define USE_I2C                                  // I2C using library wire (+10k code, 0k2 mem, 124 iram)

#ifdef USE_I2C
  #define USE_CHIRP
  #define USE_SHT                                // Enable SHT1X sensor (+1k4 code)
  #define USE_HTU                                // Enable HTU21/SI7013/SI7020/SI7021 sensor (I2C address 0x40) (+1k5 code)
  #define USE_BMP                                // Enable BMP085/BMP180/BMP280/BME280 sensor (I2C address 0x76 or 0x77) (+4k code)
//    #define USE_BME680                           // Enable support for BME680 sensor using Bosch BME680 library (+4k code)
  #define USE_BH1750                             // Enable BH1750 sensor (I2C address 0x23 or 0x5C) (+0k5 code)
//  #define USE_VEML6070                           // Enable VEML6070 sensor (I2C addresses 0x38 and 0x39) (+1k5 code)
    #define USE_VEML6070_RSET    270000          // VEML6070, Rset in Ohm used on PCB board, default 270K = 270000ohm, range for this sensor: 220K ... 1Meg
    #define USE_VEML6070_SHOW_RAW                // VEML6070, shows the raw value of UV-A
//  #define USE_ADS1115                            // Enable ADS1115 16 bit A/D converter (I2C address 0x48, 0x49, 0x4A or 0x4B) based on Adafruit ADS1x15 library (no library needed) (+0k7 code)
//  #define USE_ADS1115_I2CDEV                     // Enable ADS1115 16 bit A/D converter (I2C address 0x48, 0x49, 0x4A or 0x4B) using library i2cdevlib-Core and i2cdevlib-ADS1115 (+2k code)
//  #define USE_INA219                             // Enable INA219 (I2C address 0x40, 0x41 0x44 or 0x45) Low voltage and current sensor (+1k code)
  #define USE_SHT3X                              // Enable SHT3x (I2C address 0x44 or 0x45) or SHTC3 (I2C address 0x70) sensor (+0k7 code)
//  #define USE_TSL2561                            // Enable TSL2561 sensor (I2C address 0x29, 0x39 or 0x49) using library Joba_Tsl2561 (+2k3 code)
//  #define USE_MGS                                // Enable Xadow and Grove Mutichannel Gas sensor using library Multichannel_Gas_Sensor (+10k code)
    #define MGS_SENSOR_ADDR    0x04              // Default Mutichannel Gas sensor i2c address
  #define USE_SGP30                              // Enable SGP30 sensor (I2C address 0x58) (+1k1 code)
//  #define USE_SI1145                             // Enable SI1145/46/47 sensor (I2C address 0x60) (+1k code)
  #define USE_LM75AD                             // Enable LM75AD sensor (I2C addresses 0x48 - 0x4F) (+0k5 code)
//  #define USE_APDS9960                           // Enable APDS9960 Proximity Sensor (I2C address 0x39). Disables SHT and VEML6070 (+4k7 code)
//  #define USE_MCP230xx                           // Enable MCP23008/MCP23017 - Must define I2C Address in #define USE_MCP230xx_ADDR below - range 0x20 - 0x27 (+4k7 code)
//    #define USE_MCP230xx_ADDR 0x20               // Enable MCP23008/MCP23017 I2C Address to use (Must be within range 0x20 through 0x27 - set according to your wired setup)
//    #define USE_MCP230xx_OUTPUT                  // Enable MCP23008/MCP23017 OUTPUT support through sensor29 commands (+1k5 code)
//    #define USE_MCP230xx_DISPLAYOUTPUT           // Enable MCP23008/MCP23017 to display state of OUTPUT pins on Web UI (+0k2 code)
//  #define USE_PCA9685                            // Enable PCA9685 I2C HW PWM Driver - Must define I2C Address in #define USE_PCA9685_ADDR below - range 0x40 - 0x47 (+1k4 code)
//    #define USE_PCA9685_ADDR 0x40                // Enable PCA9685 I2C Address to use (Must be within range 0x40 through 0x47 - set according to your wired setup)
//    #define USE_PCA9685_FREQ 50                  // Define default PWM frequency in Hz to be used (must be within 24 to 1526) - If other value is used, it will rever to 50Hz
//  #define USE_MPR121                             // Enable MPR121 controller (I2C addresses 0x5A, 0x5B, 0x5C and 0x5D) in input mode for touch buttons (+1k3 code)
//  #define USE_CCS811                             // Enable CCS811 sensor (I2C address 0x5A) (+2k2 code)
//  #define USE_MPU6050                            // Enable MPU6050 sensor (I2C address 0x68 AD0 low or 0x69 AD0 high) (+2k6 code)

//  #define USE_DISPLAY                            // Add I2C Display Support (+2k code)
    #define USE_DISPLAY_MODES1TO5                // Enable display mode 1 to 5 in addition to mode 0
    #define USE_DISPLAY_LCD                      // [DisplayModel 1] Enable Lcd display (I2C addresses 0x27 and 0x3F) (+6k code)
    #define USE_DISPLAY_SSD1306                  // [DisplayModel 2] Enable SSD1306 Oled 128x64 display (I2C addresses 0x3C and 0x3D) (+16k code)
    #define USE_DISPLAY_MATRIX                   // [DisplayModel 3] Enable 8x8 Matrix display (I2C adresseses see below) (+11k code)
      #define MTX_ADDRESS1     0x71              // [DisplayAddress1] I2C address of first 8x8 matrix module
      #define MTX_ADDRESS2     0x74              // [DisplayAddress2] I2C address of second 8x8 matrix module
      #define MTX_ADDRESS3     0x75              // [DisplayAddress3] I2C address of third 8x8 matrix module
      #define MTX_ADDRESS4     0x72              // [DisplayAddress4] I2C address of fourth 8x8 matrix module
      #define MTX_ADDRESS5     0x73              // [DisplayAddress5] I2C address of fifth 8x8 matrix module
[... cut ...]
16:06:48 MQT: tele/sonoff/STATE = {"Time":"2018-10-24T16:06:48","Uptime":"0T00:00:16","Vcc":2.993, "Wifi":{"AP ":1, "SSId ":"Fast-Arci", "BSSId ":"10:13:31:F8:D3:43", "Channel":6, "RSSI":56}, "DeepSleep":0, "Heap":21328}
16:06:58 WIF: Controllo connessione...
16:06:58 WIF: Connesso
16:07:06 CMD: i2cscan
16:07:06 SRC: WebConsole from 192.168.1.165
16:07:06 RSL: Topic Ricevuto /i2cscan, Dimensione Dati 0, Dati 
16:07:06 RSL: Gruppo 0, Indice 1, Comando I2CSCAN, Dati 
16:07:06 MQT: stat/sonoff/RESULT = {"I2CScan":"No devices found"}

madmicio, 2018-10-27:
WHERE???????

> Have you set the compiler options:
#define USE_I2C
#define USE_CHIRP

stefanbode, 2018-10-27:
If you have a user_config_override.h file there. If not you can also set this in the user_config.h

This is a typical work, because not all sensors are activated by default. There is not enough memory to have them ON all.

stefanbode, 2018-10-27:
On top of this you have to define the SCL and SDA Pins for the data lines that are connected with the i2c bus.

madmicio, 2018-10-27:
It works! It was very complicated. unfortunately with your firmware I always had the same result: device non fuond
with the original tasmota: {"I2CScan": "Device (s) found at 0x20"}
but I had no information from the chirp.
in the end I found another file for the chirp on the internet. but the same no information.
So I joined some of your chirp code and part of the online chirp code. everything runs, however, only on the original tasmota but not on your firmware. I link the file that seems to work

madmicio, 2018-10-27:
[xsns_20_chirp_mois.zip](https://github.com/stefanbode/Sonoff-Tasmota/files/2521970/xsns_20_chirp_mois.zip)

stefanbode, 2018-10-27:
Thanks, but the issue is the missing #define. If you define gpio correct it will work for sure Ootb.

madmicio, 2018-10-27:
Where should I define the gpio?
in the web interface or in the code before compilation?

stefanbode, 2018-10-27:
In the webinterface you have to define SCL and SDA.

madmicio, 2018-10-27:
obviously I did it! but with you firmware: device not fuond

stefanbode, 2019-08-31:
Chirp will be include in the master branch with the. Next release. I will remove my code from the repository.

closed 2019-09-19 by stefanbode