stefanbode/Sonoff-Tasmota #79: Shelly2 support

B1G1, 2018-10-12:
Hi Stefan, 
I would like to buy two shelly2 to control my rolling shutters from 2 buttons i have near the entrance, do you think your firmware could work?

stefanbode, 2018-10-12:
Yes it will. I also have 2 Shelly just beside here on the desk to assemble. Looking great

stefanbode, 2018-10-12:
But as far as inknownthe shelly only has one relay. Therefore using it for a shutter might not work. Or am I wrong?

B1G1, 2018-10-12:
The shelly2 has 2 relays 
https://shelly.cloud/shelly2/

stefanbode, 2018-10-12:
Ah, this is what I was expecting worst case. No, it will not work because you have in this situation TWO independent ESP8266 chips that do not know each other. This will never work. The firmware required that both relays can be controlled with ONE chip. Maybe you find a fency configuration to do this with ONE shelly2, but I would say: Definitly not.

stefanbode, 2018-10-12:
First read, then understand, then answer. I was too quick with my reply. Yes, indeed it looks like the shelly2 operates 2 relays. In this case, the shutter use case will work. Worst case we have to adapt the configuration a bit. On my desk are only the shelly version2 with one relay. The other order is still open and I wait for delivery. With the single relay for sure, it will not work.